UIPEthernet/UIPEthernet #4: Crashes esp module when i call ethernet.begin

littleoz2, 2017-02-03:
I have an Esp8266 -12e which crashes and resets on Ethernet.begin, I am using it with a w5100 Ethernet module

cassy73, 2017-02-06:
Hi!
Please change "#define ACTLOGLEVEL     LOG_NONE" to "#define ACTLOGLEVEL LOG_DEBUG" in utility/logging.h file. With this modification You can take log messages on serial port.
Please send me the log results.
This library support only ENC28j60 ethernet chip, but I want correct this crash.

Best Regards

littleoz2, 2017-02-07:
I have plugged the ENC28J60 ethernet chip into the ESP module now
I have it attached:
SO  - GPIO 12
SI    - GPIO 13
SCK - GPIO 14
CS - GPIO 15

The log output when i set the loglevel to debug is as follows:

áStarting up...
ENC28J60::init DEBUG:csPin = 15
ENC28J60::init DEBUG:miso = 12
ENC28J60::init DEBUG:mosi = 13
ENC28J60::init DEBUG:sck = 14
ENC28J60::init DEBUG:Use SPI lib SPI.begin()
ENC28J60::init INFO: Chip erevid=0 initialization completed.
UIPEthernetClass::tick() ERROR:EREVID=0 -> Not found ENC28j60 device !!! Function ended !!!
UIPUDP::beginPacket DEBUG:udp beginPacket, rip: 255.255.255.255, port: 67
uipudp_appcall(void) DEBUG:udp, uip_poll preparing packet to send: 1, size: 0
UIPEthernetClass::upper_layer_chksum(uint8_t proto) DEBUG:uip_buf[34-42]: CE02
UIPEthernetClass::upper_layer_chksum(uint8_t proto) DEBUG:uip_packet(1)[42-317]: CF01
UIPUDP::_send() DEBUG:udp, uip_packet to send: 1
UIPEthernetClass::network_send() DEBUG:uip_packet: 1, hdrlen: 42
Enc28J60Network::sendPacket(memhandle handle) DEBUG:sendPacket(1) [800-93D]: FF 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 

Soft WDT reset

ctx: cont 
sp: 3ffef900 end: 3ffefcc0 offset: 01b0

>>>stack>>>
3ffefab0:  00000007 00000040 3ffeeaec 40205ad4  
3ffefac0:  0000001f 000000a0 3ffeeaec 40205ad4  
3ffefad0:  000003e8 0000093e 0000001c 0000000a  
3ffefae0:  000003e8 0000093e 000001b4 4020614a  
3ffefaf0:  00000000 00000800 0000002a 40205f75  
3ffefb00:  3ffe87b4 00000009 4023401e 3ffee915  
3ffefb10:  3ffee914 3ffeebd0 3ffee916 40204e2a  
3ffefb20:  3ffee9aa 0000002e 402341bf 3ffee8a4  
3ffefb30:  3ffee8fc 3ffee916 3ffeebd0 40205997  
3ffefb40:  00000001 3ffee8e8 3ffee8e8 402059ea  
3ffefb50:  00000001 3ffee8e8 3ffee89c 40203b74  
3ffefb60:  03010637 3b3a0f06 697001ff 00302d69  
3ffefb70:  4e450c0c 4a383243 30334432 00003030  
3ffefb80:  1e030000 ffffffff 3ffe87b0 ffffffff  
3ffefb90:  feef0080 feefeffe feefeffe feefeffe  
3ffefba0:  00000001 00000001 3ffe87b5 40207874  
3ffefbb0:  00000000 00000080 3ffee8e8 3ffee89c  
3ffefbc0:  00000000 0000014d 00000000 40203f55  
3ffefbd0:  3ffe87b4 ea000000 3ffeeaec 3ffeebb0  
3ffefbe0:  3ffee8e8 00000019 3ffeebd0 40206d80  
3ffefbf0:  0000000e 3ffee99e 3ffeebd0 3ffee8a4  
3ffefc00:  00000006 00000000 3ffee89c 402040ac  
3ffefc10:  3ffe8508 00000001 3ffe87b5 3ffeec8c  
3ffefc20:  3fffdad0 3ffee898 3ffe8508 3ffeec8c  
3ffefc30:  3fffdad0 3ffee898 3ffee908 40205134  
3ffefc40:  3ffe87b4 00000000 3ffeebd0 40206d55  
3ffefc50:  3ffe8484 feefeffe 3ffeebd0 40206d80  
3ffefc60:  3ffe8508 feefeffe 3ffeebd0 40206dcc  
3ffefc70:  3fffdad0 3ffee898 3ffeebd0 402033fd  
3ffefc80:  feefeffe feefeffe feefeffe feefeffe  
3ffefc90:  feefeffe feefeffe feefeffe 3ffeec8c  
3ffefca0:  3fffdad0 00000000 3ffeec85 402073f4  
3ffefcb0:  feefeffe feefeffe 3ffeeca0 40100718  
<<<stack<<<

 ets Jan  8 2013,rst cause:2, boot mode:(3,6)

load 0x4010f000, len 1384, room 16 
tail 8
[... 35 lines cut ...]
3ffefb40:  00000001 3ffee8e8 3ffee8e8 402059ea  
3ffefb50:  00000001 3ffee8e8 3ffee89c 40203b74  
3ffefb60:  03010637 3b3a0f06 697001ff 00302d69  
3ffefb70:  4e450c0c 4a383243 30334432 00003030  
3ffefb80:  e3010000 ffffffff 3ffe87b0 ffffffff  
3ffefb90:  feef0080 feefeffe feefeffe feefeffe  
3ffefba0:  00000001 00000001 3ffe87b5 40207874  
3ffefbb0:  00000000 00000080 3ffee8e8 3ffee89c  
3ffefbc0:  00000000 0000014c 00000000 40203f55  
3ffefbd0:  3ffe87b4 ea000000 3ffeeaec 3ffeebb0  
3ffefbe0:  3ffee8e8 00000019 3ffeebd0 40206d80  
3ffefbf0:  0000000e 3ffee99e 3ffeebd0 3ffee8a4  
3ffefc00:  00000006 00000000 3ffee89c 402040ac  
3ffefc10:  3ffe8508 00000001 3ffe87b5 3ffeec8c  
3ffefc20:  3fffdad0 3ffee898 3ffe8508 3ffeec8c  
3ffefc30:  3fffdad0 3ffee898 3ffee908 40205134  
3ffefc40:  3ffe87b4 00000000 3ffeebd0 40206d55  
3ffefc50:  3ffe8484 feefeffe 3ffeebd0 40206d80  
3ffefc60:  3ffe8508 feefeffe 3ffeebd0 40206dcc  
3ffefc70:  3fffdad0 3ffee898 3ffeebd0 402033fd  
3ffefc80:  feefeffe feefeffe feefeffe feefeffe  
3ffefc90:  feefeffe feefeffe feefeffe 3ffeec8c  
3ffefca0:  3fffdad0 00000000 3ffeec85 402073f4  
3ffefcb0:  feefeffe feefeffe 3ffeeca0 40100718  
<<<stack<<<

cassy73, 2017-02-07:
Hi!
This is a SPI communication error.
Can not initialize ENC28j60 device:"UIPEthernetClass::tick() ERROR:EREVID=0 -> Not found ENC28j60 device !!! Function ended !!!"

The EREVID must be greater than 0. (This is the ENC28j60 chip revision number.)

Please check wiring: https://github.com/UIPEthiernet/UIPEthernet/blob/master/hardware/ESP8266_enc28j60_wiring.PNG
This wiring is tested, and working properly.

I will to fix it the wdt_reset in the code.

Best regards

littleoz2, 2017-02-07:
how come there is mosfets used in the circuit?, i am using a NodeMCU dev board, would this make a difference?

cassy73, 2017-02-07:
Hi!

If You use NodeMCU you must use these mosfets or line driver IC between GPIO 15 and ENC28j60 CS.
NodeMCU have 12K pull down resistor on GPIO 15, ENC28j60 have 10K pull up resistor on CS pin.
This is causing the error.
NodeMCU wiring:
https://github.com/UIPEthernet/UIPEthernet/blob/master/hardware/NodeMCU_enc28j60_wiring.PNG

Best Regards

cassy73, 2017-02-14:
Fixed

closed 2017-02-14 by cassy73