UIPEthernet/UIPEthernet #5: setting UIP_CONF_UDP to 0 or commenting it out generates compilation errors

pabloandresm, 2017-02-10:
The utility/uipethernet-conf.h says:
/* for UDP
 * set UIP_CONF_UDP to 0 to disable UDP (saves aprox. 5kb flash) */

Commenting it out, or redefining it to 0 generates compilation errors, like:

libraries\UIPEthernet/Dhcp.h:139: undefined reference to `UIPUDP::UIPUDP()'

cassy73, 2017-02-13:
The error is fixed.
Please try it with actual code.

closed 2017-02-13 by cassy73

consolacion, 2017-08-01:
I am using the latest version as far as I know (2.0.4)
and it doesnt seem fixed:
Arduino: 1.8.3 (Linux), Board: "Arduino/Genuino Uno"

/tmp/cc8rxrWd.ltrans1.ltrans.o: In function `setup':
/home/me/Arduino/enc28j60MQTT_v2/enc28j60MQTT_v2.ino:55: undefined reference to `UIPEthernetClass::begin(unsigned char const*)'