UIPEthernet/UIPEthernet #6: Compile errors when using Eclipse

RalphBacon, 2017-02-14:
First error is that there is no "mbed.h" file (line 23 of print.h in the mbed subfolder).

If I comment that out I get several errors about ambiguous references (apologies for the line wrapping):

call of overloaded 'utoa(unsigned char&, char [9], unsigned char&)' is ambiguous	/Ethernet_Time_NTP/libraries/UIPEthernet/mbed	line 74	WString.cpp

declaration of C function 'char* utoa(long unsigned int, char*, int)' conflicts with	/Ethernet_Time_NTP/libraries/UIPEthernet/mbed	line 34	itoa.h

previous declaration 'char* utoa(unsigned int, char*, int)' here:		line 506, external location: c:\users\ralph\desktop\eclipse\eclipsearduino\arduinoplugin\tools\arduino\avr-gcc\4.8.1-arduino5\avr\include\stdlib.h

recipe for target 'libraries/UIPEthernet/mbed/WString.cpp.o' failed	/Ethernet_Time_NTP/Release/libraries/UIPEthernet/mbed	line 51	subdir.mk

(That last error might be an error about an error).

I'm hoping this (improved) library will work with a UDP NTP call that the original fails on (but the standard Ethernet library works just fine, but not on ENC28J60 hardware of course).

Look forward to any replies / corrections.

cassy73, 2017-02-14:
Hi!

I tested compile "only" with 'online MBED compiler', SMeshStudio (MBED,Eclipse), and Arduino IDE.
The SMeshStudio use Eclipse.
Do you use Eclipse configured by yourself, or what IDE?
Please send me the Your IDE specific -D gcc option like for arduino is "-DARDUINO" or for MBED is "-D__MBED__".

Best Regards

RalphBacon, 2017-02-15:
I can dig out the actual settings tonight but I'm using the Eclipse Arduino add-on if that helps you identify the settings at all?

cassy73, 2017-02-15:
Hi!

The Eclipse Arduino define is:-D__IN_ECLIPSE__=1
You can see in compile console.

The workaround is if you use atmel MCU:
Only delete mbed directory with all content.

Best regards

RalphBacon, 2017-02-16:
I will try this tonight and also report back my finding on the UDP NTP sketch too (fails using standard [ntruchess] UIPEthernet with some sort of UDP_NULL_PORT error). I am hoping this more recent library will work as does the standard Ethernet library (on W5100 module).

RalphBacon, 2017-02-17:
Yesterday evening I removed the "mbed" folder and recompiled using Eclipse. Success!

Then I used the standard Arduino NTP sketch that uses UDP to get the date/time from a server in a pool of NTP server. Success! As previously mentioned this did **NOT** work using the older **ntruchsess** version of this library (UDP_NULL_PORT error) so congratulations in resolving that issue.

This means that I can once again use the UIPEthernet library in full confidence that it is truly a drop-in replacement for the standard Ethernet library and runs boh W5100 and ENC28J60 Ethernet modules.

Thanks again for responding so promptly. I will now close this issue as resolved.

closed 2017-02-17 by RalphBacon