UIPEthernet/UIPEthernet #8: IOS and Android Don't Work With This Example --- DHCP doesn't either

JZ-SmartThings, 2017-03-08:
Hi @cassy73 thanks for looking into this from our SmartThings forum post. I wanted to submit this like you requested. Two issues, my particular way of building a char array doesn't seem to work. Also in this code sample, you'll see that DHCP doesn't work either. I've never been able to get that to work in any version of UIPEthernet including the original/AVR-only library. Here's the sample:

```
#include <UIPEthernet.h>
EthernetServer server = EthernetServer(80);

void setup() {
  Serial.begin(115200);
  // ENC28J60 ETHERNET
  uint8_t mac[6] = {0xFF,0x01,0x02,0x03,0x04,0x05};
  IPAddress myIP(192,168,0,226);
  Ethernet.begin(mac,myIP);
  /* //DHCP NOT WORKING
  if (Ethernet.begin(mac) == 0) {
    while (1) {
      Serial.println("Failed to configure Ethernet using DHCP");
      delay(10000);
    }
  }
  */
  server.begin();
  Serial.print("localIP: "); Serial.println(Ethernet.localIP());
  Serial.print("subnetMask: "); Serial.println(Ethernet.subnetMask());
  Serial.print("gatewayIP: "); Serial.println(Ethernet.gatewayIP());
  Serial.print("dnsServerIP: "); Serial.println(Ethernet.dnsServerIP());
  Serial.println("Setup finished...");
}

void loop() {
  String request, fullrequest;
  EthernetClient client = server.available();  // try to get client
  Serial.print("client="); Serial.println(client);
  Serial.print("client.connected="); Serial.println(client.connected());
  Serial.print("client.read FIRST CHAR=");Serial.println(client.read());

  if (client) {  // got client?
    boolean currentLineIsBlank = true;
    String HTTP_req;          // stores the HTTP request
    while (client.connected()) {
      if (client.available()) {   // client data available to read
        char c = client.read(); // read 1 byte (character) from client
        HTTP_req += c;  // save the HTTP request 1 char at a time
        // last line of client request is blank and ends with \n
        // respond to client only after last line received
        if (c == '\n' && currentLineIsBlank) { //&& currentLineIsBlank --- first line only on low memory like UNO/Nano otherwise get it all for AUTH
          fullrequest=HTTP_req;
          request = HTTP_req.substring(0,HTTP_req.indexOf('\r'));
          //auth = HTTP_req.substring(HTTP_req.indexOf('Authorization: Basic '),HTTP_req.indexOf('\r'));
          HTTP_req = "";    // finished with request, empty string
          request.replace("GET ", ""); request.replace(" HTTP/1.1", ""); request.replace(" HTTP/1.0", "");
          Serial.println(request);
          delay(10); // pause to make sure pin status is updated for response below
          client.println("TEST");
          break;
        }
        // every line of text received from the client ends with \r\n
        if (c == '\n') {
          // last character on line of received text
          // starting new line with next character read
          currentLineIsBlank = true;
        } else if (c != '\r') {
          // a text character was received from client
          currentLineIsBlank = false;
        }
      } // end if (client.available())
    } // end while (client.connected())
    delay(1);      // give the web browser time to receive the data
    client.stop(); // close the connection
  } // end if (client)
}
```

cassy73, 2017-03-21:
Hi!

The MAC ADDRESS (0xFF,0x01,0x02,0x03,0x04,0x05) IS WRONG!!!!!!!!!!!!!!!!!!!!!!
You can test it on:
http://sqa.fyicenter.com/Online_Test_Tools/MAC_Address_Format_Validator.php

https://en.wikipedia.org/wiki/MAC_address
If You set 0. bit of first octet then the mac address is multicast. If the 0. bit is 0 then mac address is unicast.
If You set 0. bit of first octet then the mac address is local. If the 0. bit is 0 then mac address is global unique.

I think: The first octet can not be 0xFF.

Maybe didn't work dhcp for You, and didn't work with android and IOS.
Please change it and test it.

Best Regards

JZ-SmartThings, 2017-03-24:
Awesome, thanks for that! I was able to verify that DHCP is definitely working when a "good" MAC is used. I'm out of town though, so not easily able to test Android & IOS to see if that fixed them as well. I'll let you know in the next couple of days... thank you!!!

JZ-SmartThings, 2017-03-26:
Wow, I feel like a ditz :) you were dead on and I'm glad I asked... BOTH items reported in this issue were fixed when I used a "good" MAC address. Now Android and IOS are easily pulling up the ESP8266 hosted web-page. Interesting that a PC could pull up the "bad" MAC address while IOS and Android both prevent usage in this scenario. Closing the issue and very thankful for your answer and the hard work you've contributed.

closed 2017-03-26 by JZ-SmartThings