UIPEthernet/UIPEthernet #9: DHCP timeout?

alxferraz, 2017-03-10:
How can I set a timeout while trying to connect with DHCP?  I noticed there is one but I would like it shorter. Thanks in advance!

JZ-SmartThings, 2017-03-18:
In the UIP library folder, take a look at DHCP.h, go to the bottom and look 5 lines up. I'm surprised that you have DHCP working at all. This has never worked for me.

cassy73, 2017-03-20:
Hi!

The DHCP code already modified but doesn't tested.
Publication coming soon.

Best Regards

JZ-SmartThings, 2017-03-20:
Happy to test if you have a preview ready somewhere. Thanks for the changes!

cassy73, 2017-03-20:
You can set timeouts in dhcp.h file.
#define DHCP_TIMEOUT            60000
#define DHCP_RESPONSE_TIMEOUT   4000
The values are in milliseconds.

Tested TCPCLIENT.INO example with ESP8266.

Best Regards

alxferraz, 2017-03-22:
Thank You very much guys!

cassy73, 2017-03-24:
Hi!

Please try it, and if working properly then close this issue.

Best Regards

closed 2017-03-24 by alxferraz