UIPEthernet/UIPEthernet #10: error using mysql conector

izzyneko, 2017-03-19:
i had this error could you help me

Arduino:1.8.1 (Windows 7), Tarjeta:"Arduino/Genuino Uno"

Opciones de compilación cambiadas, reconstruyendo todo
In file included from C:\Users\luis\Documents\Arduino\libraries\UIPEthernet/UIPEthernet.h:35:0,

                 from C:\Users\luis\AppData\Local\Temp\arduino_modified_sketch_832608\connect.ino:27:

C:\Users\luis\Documents\Arduino\libraries\UIPEthernet/utility/Enc28J60Network.h:99:8: warning: #warning "Not defined ENC28J60_CONTROL_CS. Use borad default SS pin setting. You can configure in 'utility/Enc28J60Network.h'." [-Wcpp]

       #warning "Not defined ENC28J60_CONTROL_CS. Use borad default SS pin setting. You can configure in 'utility/Enc28J60Network.h'."

        ^

In file included from C:\Users\luis\Documents\Arduino\libraries\UIPEthernet/UIPClient.h:37:0,

                 from C:\Users\luis\Documents\Arduino\libraries\UIPEthernet/UIPEthernet.h:41,

                 from C:\Users\luis\AppData\Local\Temp\arduino_modified_sketch_832608\connect.ino:27:

C:\Users\luis\Documents\Arduino\libraries\UIPEthernet/utility/logging.h:24:2: warning: #warning "You can configure LogObject and ACTLOGLEVEL in 'utility/logging.h'. More verbosity more memory usage." [-Wcpp]

 #warning "You can configure LogObject and ACTLOGLEVEL in 'utility/logging.h'. More verbosity more memory usage."

  ^

In file included from C:\Users\luis\Documents\Arduino\libraries\UIPEthernet/UIPEthernet.h:23:0,

                 from C:\Users\luis\AppData\Local\Temp\arduino_modified_sketch_832608\connect.ino:27:

C:\Users\luis\Documents\Arduino\libraries\UIPEthernet/ethernet_comp.h:5:24: error: redefinition of 'class UIPClient'

 #define EthernetClient UIPClient

                        ^

C:\Program Files\Arduino\libraries\Ethernet\src/EthernetClient.h:8:7: note: in expansion of macro 'EthernetClient'

 class EthernetClient : public Client {

       ^

In file included from C:\Users\luis\Documents\Arduino\libraries\UIPEthernet/UIPEthernet.h:41:0,

                 from C:\Users\luis\AppData\Local\Temp\arduino_modified_sketch_832608\connect.ino:27:

C:\Users\luis\Documents\Arduino\libraries\UIPEthernet/UIPClient.h:75:9: error: previous definition of 'class UIPClient'

   class UIPClient : public Client {

         ^

In file included from C:\Users\luis\Documents\Arduino\libraries\UIPEthernet/UIPEthernet.h:23:0,

                 from C:\Users\luis\AppData\Local\Temp\arduino_modified_sketch_832608\connect.ino:27:

C:\Users\luis\Documents\Arduino\libraries\UIPEthernet/ethernet_comp.h:6:24: error: redefinition of 'class UIPServer'

 #define EthernetServer UIPServer

                        ^

C:\Program Files\Arduino\libraries\Ethernet\src/EthernetServer.h:8:7: note: in expansion of macro 'EthernetServer'

 class EthernetServer : 

       ^

In file included from C:\Users\luis\Documents\Arduino\libraries\UIPEthernet/UIPEthernet.h:42:0,

                 from C:\Users\luis\AppData\Local\Temp\arduino_modified_sketch_832608\connect.ino:27:

C:\Users\luis\Documents\Arduino\libraries\UIPEthernet/UIPServer.h:42:9: error: previous definition of 'class UIPServer'

   class UIPServer : public Server {

         ^

In file included from C:\Users\luis\Documents\Arduino\libraries\UIPEthernet/UIPEthernet.h:23:0,

                 from C:\Users\luis\AppData\Local\Temp\arduino_modified_sketch_832608\connect.ino:27:

C:\Users\luis\Documents\Arduino\libraries\UIPEthernet/ethernet_comp.h:4:18: error: conflicting declaration 'EthernetClass UIPEthernet'

 #define Ethernet UIPEthernet

                  ^

C:\Program Files\Arduino\libraries\Ethernet\src/Ethernet.h:39:22: note: in expansion of macro 'Ethernet'

 extern EthernetClass Ethernet;

                      ^

In file included from C:\Users\luis\AppData\Local\Temp\arduino_modified_sketch_832608\connect.ino:27:0:

C:\Users\luis\Documents\Arduino\libraries\UIPEthernet/UIPEthernet.h:133:25: note: previous declaration as 'UIPEthernetClass UIPEthernet'

 extern UIPEthernetClass UIPEthernet;

                         ^

exit status 1
Error compilación en tarjeta Arduino/Genuino Uno.

Este reporte podría tener más información con
"Mostrar salida detallada durante la compilación"
opción habilitada en Archivo -> Preferencias.

cassy73, 2017-03-20:
Hi!

I found this in Your log: " redefinition of 'class UIPClient'"
Please search in all libraries directories of arduinoIDE another version of UIPEthernet or Ethernet library.
The path of libraries:
c:\arduino\libraries
c:\users\....\Arduino\libraries
c:\users\appdata\\Arduino15\packages\hardware\sam\1.6.5\libraries

If You found another version of UIPEthernet library please remove.

Best Regards

cassy73, 2017-03-28:
Hi!

Please remove "C:\Program Files\Arduino\libraries\Ethernet\" directory with all content.

Best Regrads

cassy73, 2017-05-03:
I think: The problem is with Your framework doesn't with UIPEthernet library.

I close this issue.

closed 2017-05-03 by cassy73