UIPEthernet/UIPEthernet #11: CS pin for Due is not practical

SteveHodge, 2017-03-20:
For the Due, Enc28J60Network.h defines ENC28J60_CONTROL_CS to BOARD_SPI_DEFAULT_SS which is defined as 78. But pin 78 isn't actually available on a header (see https://www.arduino.cc/en/Hacking/PinMappingSAM3X). In order to get the library working with the Due I had to change the definition of ENC28J60_CONTROL_CS to 10. I suggest changing the definition to one of the ones specified here: https://www.arduino.cc/en/Reference/SPI (4, 10, or 52).

cassy73, 2017-03-20:
Hi!

Really: I use the BOARD_SPI_DEFAULT_SS to define ENC28J60_CONTROL_CS.
BOARD_SPI_DEFAULT_SS defined in arduino DUE variant.h.
"c:\........\Application Data\Arduino15\packages\arduino\hardware\sam\1.6.5\variants\arduino_due_x\variant.h"
This file contains this:
"
#define PIN_SPI_SS3          (78u)
#define PIN_SPI_MOSI         (75u)
#define PIN_SPI_MISO         (74u)
#define PIN_SPI_SCK          (76u) 
#define BOARD_SPI_SS0        (10u)
#define BOARD_SPI_SS1        (4u)
#define BOARD_SPI_SS2        (52u)
#define BOARD_SPI_SS3        PIN_SPI_SS3
#define BOARD_SPI_DEFAULT_SS BOARD_SPI_SS3
"
OK. I understand You, but: Which SPI use instead SPI3? 0? 1? 2?

Best Regards

SteveHodge, 2017-03-21:
I suggest using BOARD_SPI_SS0 (pin 10), just because that seems to be the most commonly used default pin (https://www.arduino.cc/en/Reference/SPI). I think that's what is used on the "official" ethernet shield, going by the docs (I haven't tried it).
It's weird that they chose to make SS3 the default for the Due given it's not taken out to a header.

cassy73, 2017-03-21:
Hi!

Chnaged to:
#define ENC28J60_CONTROL_CS     BOARD_SPI_SS0

Best Regards

closed 2017-03-21 by SteveHodge