UIPEthernet/UIPEthernet #13: Arduino Due

clhett01, 2017-05-12:
I see where this library will work with an Arduino Due, and I have spent several hours looking for proper wiring.  Can you please add a wiring diagram for the Due.  Also, are there any nuances to the library to make it work with the Due?

cassy73, 2017-05-12:
The library use all default pins for SPI bus for all device, exclude Arduino Due.
See Issue#11.
On arduino due the SPI default is:
#define PIN_SPI_SS3 (78u)
#define PIN_SPI_MOSI (75u)
#define PIN_SPI_MISO (74u)
#define PIN_SPI_SCK (76u)
#define BOARD_SPI_SS0 (10u)
#define BOARD_SPI_SS3 PIN_SPI_SS3
#define BOARD_SPI_DEFAULT_SS BOARD_SPI_SS3

This library use with Arduino Due pin74:MISO,  pin75:MOSI, pin76:SCK and pin10 for SS.

I will create wiring image.

clhett01, 2017-05-16:
Thank you. I was using the ICSP port instead of the SPI port.  Works like a dream!  Thank you for your continued development on this fantastic library!

Chris

closed 2017-05-16 by clhett01