UIPEthernet/UIPEthernet #14: Changing CS pin causes hang

Electronics, 2017-05-31:
As there is no easy way to change the chip select pin from (on an arduino) pin 10 to say, pin 8, changing the ENC28J60_CONTROL_CS definition manually causes a hang on Ethernet.begin(mac).

On a scope, no transition can be seen until bizarely serial is closed at which point some packets seem to flow but still no DHCP assigned.

(TcpClient Example used for testing)

cassy73, 2017-06-07:
Hi!

For what purpose do you use it pin 10?
I found this:
http://forum.arduino.cc/index.php?topic=129538.0

Best Regards

cassy73, 2018-01-23:
"
The reason to make the SS pin Output, high has to do with the SPI hardware in the Chip.  if the SS pin is currently INPUT, when you start using the SPI functions the chip goes into SLAVE SPI mode instead MASTER SPI mode.  
The Master SPI device control the Clock. If the chip is in SLAVE SPI mode, The SPI.transfer() function will hang forever waiting for the 'Master' to provide the clock signal.

So anytime you use SPI, pinMode(SS,OUTPUT); must be executed.  The digitaWrite(SS,HIGH); is optional.  Personally I use SS(slave Select) as a CS_pin(chip Select) when ever I can.  That is what SS was to do.
Because you are using SPI Master mode the SS pin must be an output while any SPI transfers are executing.
"

closed 2018-01-23 by cassy73

TLS1000, 2018-01-23:
I'm having the same issue with my ENJ chip connected to a MEGA. I want to use pin 20 as CS. The default pin is used for other SPI-slave. Communication with other SPI slave is ok.

Strange thing is that with an older version of the library it has worked. how can I use pin 20 as CS? 
thank you!