UIPEthernet/UIPEthernet #17: Detect dead TCP connection

tenbaht, 2017-06-21:
I am trying to find a way to detect a dead TCP connection where the other side silently went away without properly closing an open connection (by crash or reboot or similar). So far such a connection would be considered open by UIPEthernet indefinitly and would block one entry in the connection table permanently.

Automatic generation of TCP-keep-alive packets within uIP or on the UIPEthernet level would be ideal, but I can't find the right way to do so.

Is there any way to send a valid TCP packet without data? Whenever I try to do so, the packet is never sent - probably because one of the "if (uip_len > 0)" tests discards it somewhere in the process.

Background: I am using this lib for a control application. The device is almost never rebooted and it is common, that a TCP connection is inactive, but still valid and open over night or the weekend. And it happens that the controlling PC is silently disconnected without closing the connection. This happens frequently enough that my device must be able to detect and remove dead connections without affecting valid, but inactive connections.

farifori, 2021-12-30:
Hi there,

I faced a similar problem and I'll be very happy to any receiving help to fix it in a pull request. I send dummy packets after the server silently goes down with `write` API and it returns current sent bytes while I'm pretty sure the client didn't receive any ACKs.

Thanks for any help or suggestions on fixing this issue.

jandrassy, 2021-12-31:
do you use the github master version?

farifori, 2021-12-31:
Yes I use latest master version. I also tried ENCethernet library. They both have the same behavior on silent server disappearing.

jandrassy, 2021-12-31:
my project has 3 servers (web, telnet, ota). I did a long run test with EthernetENC in August. The servers stopped responding after 3 weeks. But clients (modbus, blynk and two others) worked a week longer. Then a watchdog reset happen. But since then I fixed at least one problem.

Now I run a new long run test with the same project. It is in day 11. Servers respond.

farifori, 2021-12-31:
I didn't use server functions of this library, I just use TCP client. I have a android TCP Server which these clients connect to that. When my sever loses its network connection, clients (which uses this library) don't realize that there is no server any more. They send they're data and return no errors. Therefore if the server comes back, clients won't connect the server again because clients think they're still connected. How can I trace this problem? Why receiving no TCP ack throws no error?

jandrassy, 2021-12-31:
it is the long timeout of TCP connection. all TCP/IP stacks have it. on small devices it is a problem because it blocks the limited count of connections.
but if you send something to the server and it doesn't respond you can close the connection.

farifori, 2021-12-31:
Thanks for your time and suggestions. I initially thought if I send something to server in this scenario, the connection will close, but it didn't! It works just like a normal connected socket! So I can't figure out how to reset the connection.

I do appreciate your cooperation on this issue.

jandrassy, 2021-12-31:
doesn't the server send some response or confirmation on the application level which you read? if the read times out, then close the connection and try to reconnect.

farifori, 2021-12-31:
No, for testing purposes I disconnect servers network. So there's no chance to server responding anything. But uip-ethernet client still sends data and throws no error. There seems to be no time out.

Is there any TCP keep alive implemented on this library? This property is in charge for detecting dead connections.
https://tldp.org/HOWTO/TCP-Keepalive-HOWTO/overview.html

jandrassy, 2021-12-31:
TCP keep alive is not part of TCP/IP. it is something the application can implement.
for you it would be enough to get some response from server as confirmation of recieving. if you don't receive a response in some timeout, close the connection and try to reopen it.
[here](https://github.com/jandrassy/Regulator/blob/99f37f6bb335f85f4a9df7a68087935ca8a9a3e8/Regulator/WemoInsight.ino#L80) I have a timeout

TCP re-transmits a packet if it doesn't receive ACK. 8 times in uIP (uipopt.h UIP_MAXRTX). every time the time until next retry is longer.

closed 2022-01-16 by jandrassy