UIPEthernet/UIPEthernet #19: Possible integration of Wiznet W5500 / Wiz550io

asetyde, 2017-07-17:
Hi , I ve many problems wit not updated library of wiznet of 5500 , this board is important because is a few models wit a low price can give a macaddress inside board .

I ask if someone can help me to integrate on this library , this board .
I see that w5100 is compatible , w5500 is not so different but has Mac and more powerful  chip.

Ref of wiznet libraries and wiki 👍 
https://github.com/embeddist/WIZ_Ethernet_Library-IDE1.6.x

zoomx, 2017-07-17:
"Compatible with Wiznet W5100 Ethernet library API" mean that you can use the same functions not that the library works with a Wiznet W5100. This is a libary for ENC28j60.

asetyde, 2017-07-17:
ok many thanks !

closed 2017-07-17 by asetyde