UIPEthernet/UIPEthernet #23: How to connect to Arduino Leonardo?

braian87b, 2017-12-02:
Hi, I have an Arduino Leonardo and I want to know how to connect the board to the Leonardo.
I google it about but there is about 5 or 6 different connection schemes, I have tried the same as for UNO and another one, I do not want to keep trying to avoid damage the board...
Can you point me out how I need to connect and what I need to do to be able to get it working?
Thanks in advance.

cassy73, 2018-01-26:
Hi!

On arduino leonardo the spi bus doesn't wired to the common arduino pins.
On leonardo SPI_SCK, SPI_MISO, SPI_MOSI wired to the ICSP connector.
The SPI_SS wired to the RX LED.
You can find atmega32u4 (arduino leonardo) arduino pin assigment:
https://forum.arduino.cc/index.php?topic=241369.0

I changed the code and added leonardo specific ENC28J60_CONTROL_CS pin setup to PIN_A10.

Here is the wiring:
https://microcontrollerelectronics.com/wp-content/uploads/2015/07/Leonardo_Enc28j60.png

Best Regards

cassy73, 2018-01-29:
Hi!

I made wiring picture:
https://github.com/UIPEthernet/UIPEthernet/blob/master/hardware/arduino_leonardo_enc28j60_wiring.PNG

Best Regards

closed 2018-01-29 by cassy73