UIPEthernet/UIPEthernet #25: This library is not working: WARNING: Don't free NOBLOCK handle

Dravon5X, 2017-12-06:
Hi

I have issue that this library is not working at all. client.connect() is returning -5 and in debug log I see this:
```
ENC28J60::init INFO: Chip erevid=6 initialization completed.
MemoryPool::freeBlock(memhandle handle) WARNING: Don't free NOBLOCK handle
MemoryPool::freeBlock(memhandle handle) WARNING: Don't free NOBLOCK handle
MemoryPool::freeBlock(memhandle handle) WARNING: Don't free NOBLOCK handle
MemoryPool::freeBlock(memhandle handle) WARNING: Don't free NOBLOCK handle
MemoryPool::freeBlock(memhandle handle) WARNING: Don't free NOBLOCK handle
MemoryPool::freeBlock(memhandle handle) WARNING: Don't free NOBLOCK handle
```

cassy73, 2018-01-23:
Hi!

If You want store hostname to flash, please try it with PSTR function.
example:
if (client.connect(PSTR("server.local"),5000))

Best Regards!

jandrassy, 2019-01-23:
the connect function doesn't take progmem string as parameter

closed 2019-01-23 by jandrassy