UIPEthernet/UIPEthernet #27: changing CS causes hang

TLS1000, 2018-01-24:
Hi, I'm using the ENJ chip on an Arduino mega, CS connected to pin 20(default pin is used for another SPI slave). 

I cannot get the example TCP server program to work. I get nothing on the serial monitor, so I guess it hangs at the very beginning.

I tried adding '#define ENC28J60_CONTROL_CS     20' in my program, I also tried changing it in the library file.  My hardware has worked before with the previous truchsess version of the library. But after changing laptop with update IDE and library...nothing. 

please help

TLS1000, 2018-01-24:
based on a previous issue I also tried adding this in the 'loop' section:
pinMode(SS,OUTPUT);
digitalWrite(SS,HIGH);

TLS1000, 2018-01-24:
update: I got it working again with the truchsess library:
`https://github.com/ntruchsess/arduino_uip`

writing #define ENC28J60_CONTROL_CS 20'  in the header doesn't work, you need to change it in the Enc28J60Network.h file.

cassy73, 2018-01-24:
Hi!

I added SS pin OUTPUT configuration to the code.
Please download new version, and try it.

Best Regards

closed 2018-01-31 by cassy73