UIPEthernet/UIPEthernet #29: DHCP on arduino due dont have response

antoniosap, 2018-02-24:
attach partial log

`DhcpClass::request_DHCP_lease(void) DEBUG
_V1:dhcp_state=STATE_DHCP_START -> send_DHCP_MESSAGE DHCP_DISCOVER
DhcpClass::send_DHCP_MESSAGE(uint8_t messageType, uint16_t secondsElapsed) DEBUG_V1:Function started
UIPUDP::beginPacket DEBUG:udp beginPacket, rip: 255.255.255.255, port: 67
Enc28J60Network::writePacket(memhandle handle, memaddress position, uint8_t* buffer, uint16_t len) DEBUG_V2: Write bytes:28 save to block(1) [82B]: 1 1 6 0 0 0 4 A6 0 8 80 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 
Enc28J60Network::writePacket(memhandle handle, memaddress position, uint8_t* buffer, uint16_t len) DEBUG_V2: Write bytes:16 save to block(1) [847]: B5 58 D1 3 3D 26 0 0 0 0 0 0 0 0 0 0 
...
DhcpClass::printByte(char * buf, uint8_t n ) DEBUG_V1:Function started
DhcpClass::printByte(char * buf, uint8_t n ) DEBUG_V1:Function started
Enc28J60Network::writePacket(memhandle handle, memaddress position, uint8_t* buffer, uint16_t len) DEBUG_V2: Write bytes:30 save to block(1) [917]: 63 82 53 63 35 1 1 3D 7 1 B5 58 D1 3 3D 26 C C 45 4E 43 32 38 4A 30 33 33 44 32 36 
Enc28J60Network::writePacket(memhandle handle, memaddress position, uint8_t* buffer, uint16_t len) DEBUG_V2: Write bytes:9 save to block(1) [935]: 37 6 1 3 6 F 3A 3B FF 
uipudp_appcall(void) DEBUG:udp, uip_poll preparing packet to send: 1, size: 317
UIPEthernetClass::upper_layer_chksum(uint8_t proto) DEBUG:uip_buf[34-42]: CE02
UIPEthernetClass::upper_layer_chksum(uint8_t proto) DEBUG:uip_packet(1)[42-317]: 80E3
UIPUDP::_send() DEBUG:udp, uip_packet to send: 1
UIPEthernetClass::network_send() DEBUG:uip_packet: 1, hdrlen: 42
Enc28J60Network::writePacket(memhandle handle, memaddress position, uint8_t* buffer, uint16_t len) DEBUG_V2: Write bytes:42 save to block(1) [801]: FF FF FF FF FF FF B5 58 D1 3 3D 26 8 0 45 0 1 2F 0 3 0 0 40 11 79 BC 0 0 0 0 FF FF FF FF 0 44 0 43 1 1B 1C 7F 
Enc28J60Network::sendPacket(memhandle handle) DEBUG:sendPacket(1) [800-93D]: 0 FF FF FF FF FF FF B5 58 D1 3 3D 26 8 0 45 0 1 2F 0 3 0 0 40 11 79 BC 0 0 0 0 FF FF FF FF 0 44 0 43 1 1B 1C 7F 1 1 6 0 0 0 4 A6 0 8 80 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 B5 58 D1 3 3D 26 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 63 82 53 63 35 1 1 3D 7 1 B5 58 D1 3 3D 26 C C 45 4E 43 32 38 4A 30 33 33 44 32 36 37 6 1 3 6 F 3A 3B FF 
 `

cassy73, 2018-03-01:
I'm sorry to react so late.
I see the mistake.
The good package start with 6 piece 'FF' byte, not '0'.
At this time i don't know that writing to 801 is wrong,
or sending with beginning 800.
Please replace LOG_DEBUG_V3 to  LOG_DEBUG_V2 in the condition of UIPUDP::endPacket(void) function in UIPUdp.cpp file, and compile, and send me new debug log.
UIPUDP::endPacket(void)
{
  #if ACTLOGLEVEL>=LOG_DEBUG_V3
Please chnage to '#if ACTLOGLEVEL>=LOG_DEBUG_V2'!

Best Regards

antoniosap, 2018-03-03:
attached new log file

[new-debug-endpacket-log.zip](https://github.com/UIPEthernet/UIPEthernet/files/1777499/new-debug-endpacket-log.zip)

cassy73, 2018-03-05:
Hi!

Can You send me full log? (With enc28j60 init)

Best regards.

antoniosap, 2018-03-07:
[debug-dhcp.zip](https://github.com/UIPEthernet/UIPEthernet/files/1788050/debug-dhcp.zip)

cassy73, 2018-03-07:
Hi!

Thank You.

Please try change line 
"#define MEMPOOL_STARTADDRESS TXSTART_INIT+1"
to "#define MEMPOOL_STARTADDRESS TXSTART_INIT"
in utility/mempool_conf.h file.

Please inform me about results.
And please send me new full log.

Best Regards

cassy73, 2018-03-07:
Hi!

Sorry.

First I haven't see: Your MAC address:B5 58 D1 03 3D 26 is wrong.
First octet of mac address must even value.

https://en.wikipedia.org/wiki/MAC_address

Please change B5 to B4, and try it.

Best Regards

antoniosap, 2018-03-07:
the MAC address is generated here
https://www.miniwebtool.com/mac-address-generator/

cassy73, 2018-03-07:
Hi!

Can you test it with B4.... mac address?

Best Regards

antoniosap, 2018-03-07:
[logged.zip](https://github.com/UIPEthernet/UIPEthernet/files/1790452/logged.zip)

B4 + change line
"#define MEMPOOL_STARTADDRESS TXSTART_INIT+1"
to "#define MEMPOOL_STARTADDRESS TXSTART_INIT"
in utility/mempool_conf.h file

OK. DHCP assigned

cassy73, 2018-03-08:
Hi!

Please change "#define MEMPOOL_STARTADDRESS TXSTART_INIT" to
original "#define MEMPOOL_STARTADDRESS TXSTART_INIT+1".

The error was only caused by wrong mac address.

Best Regards

closed 2018-03-09 by cassy73