UIPEthernet/UIPEthernet #30: How to use ENC28J60 module with SAMD controler

ethicstechOSS, 2018-03-03:
Hi. I am struggling to use ENC28J60 module with Adafruit Feather M0 board (ATSAMD21G18). Should I change any configuration in enc28j60network.h file for this board?. Any kind of help is appreciated.

TIA,
Prakash.

cassy73, 2018-03-04:
Hi!

What's Your problem?
The library didn't compiled?
The hardware doesn't work?

I tried compile to Arduino Zero, and Arduino MKRZero (SAM21D mcu).
and example compiled properly.
I doesn't have board with SAM21D mcu.

The variants.h of mkrzero contains:
"
// SPI Interfaces
// --------------
#define SPI_INTERFACES_COUNT 2
 
// SPI
#define PIN_SPI_MISO  (10u)
#define PIN_SPI_MOSI  (8u)
#define PIN_SPI_SCK   (9u)
#define PIN_SPI_SS    (4u)
#define PERIPH_SPI    sercom1
#define PAD_SPI_TX    SPI_PAD_0_SCK_1
#define PAD_SPI_RX    SERCOM_RX_PAD_3

static const uint8_t SS   = PIN_SPI_SS;   // SPI Slave SS not used. Set here only for reference.
static const uint8_t MOSI = PIN_SPI_MOSI;
static const uint8_t MISO = PIN_SPI_MISO;
static const uint8_t SCK  = PIN_SPI_SCK;

// SPI1: Connected to SD
#define PIN_SPI1_MISO (29u)
#define PIN_SPI1_MOSI (26u)
#define PIN_SPI1_SCK  (27u)
#define PIN_SPI1_SS   (28u)
#define PERIPH_SPI1   sercom2
#define PAD_SPI1_TX   SPI_PAD_0_SCK_1
#define PAD_SPI1_RX   SERCOM_RX_PAD_3
static const uint8_t SS1   = PIN_SPI1_SS;
static const uint8_t MOSI1 = PIN_SPI1_MOSI;
static const uint8_t MISO1 = PIN_SPI1_MISO;
static const uint8_t SCK1  = PIN_SPI1_SCK; 
"

Best Regards

ethicstechOSS, 2018-03-05:
Hi. Thanks for your reply. Sorry for this mess-up. I would like reframe my words here and explain my issue. 

1. I would like to change default CS/SS pin configuration on the UIPEthernet library. Because, I am not going to use the default Arduino SS pin. Guide me how to change this CS pin settings for ATSAMD on UIPEthernet.

BTW, I can able to compile the library for my board without any issues. But, I did not tried any sample programs yet, coz my CS pin is not properly configured.

TIA.

cassy73, 2018-03-05:
Hi!

Hi!

You can define ENC28J60_CONTROL_CS at the beginning of the file utility/Enc28J60Network.h,
before line "#if !defined(ENC28J60_CONTROL_CS)".

example:
#define ENC28J60_CONTROL_CS PIN_WIRE_SDA

Best regards

ethicstechOSS, 2018-03-06:
Hi. Thanks for your input.
I added a line #define ENC28J60_CONTROL_CS D5 before  "#if !defined(ENC28J60_CONTROL_CS)" in the file  utility/Enc28J60Network.h. But I am getting the following error,

Enc28J60Network.h:25:29: error: 'D5' was not declared in this scope
 #define ENC28J60_CONTROL_CS D5   

Then I tried with #define ENC28J60_CONTROL_CS PIN_WIRE_SDA
I connected SDA pin as CS in ENC28J60 module. Now I can able to compile the sample WebClient program and it gives me the log,

> Failed to configure Ethernet using DHCP
> connecting...
> connected
> disconnecting.
> 

I tried EchoServer and other couple of sample programs from UIPEthernet. But I am not getting any log on the serial monitor. 

TIA.

ethicstechOSS, 2018-03-06:
Another query : Should I required pull-up for CS pin?. 
I disconnected my ENC module and run the same sample program. Looks like the module is not detected by the MCU. Please advise what are the pins I can use it for Adafruit feather M0 board's  schematic here, https://learn.adafruit.com/assets/32914

TIA.

cassy73, 2018-03-06:
Hi!

The chinese ENC28j60 module have pull up resistor on CS wire.
You can change log level in utility/logging.h
Default log level is "LOG_NONE"

I looked at  variant.cpp of Arduino MKRZERO.
 | 08         | MOSI             |  PA16  on feather D11
 | 09         | SCK              |  PA17    on feather D13
 | 10         | MISO             |  PA19  on feather D12

If you #define ENC28J60_CONTROL_CS PIN_WIRE_SDA
the CS pin is 
11         | SDA              |  PA08 on feather D4_RST

Best Regards

ethicstechOSS, 2018-03-10:
Hi. Thanks for the clarification.
- I did not used any external pullup in my connection for CS.
- Now I can able to see the log by enabling LOG_INFO and disable LOG_NONE.
- As per your suggestion, I used D11, D12, D13 & D4_RST pins feather M0.
But I am getting the below, 
> ENC28J60::init ERROR:TIMEOUT !!
> ENC28J60::init INFO: Chip erevid=0 initialization completed.
> UIPEthernetClass::tick() ERROR:EREVID=0 -> Not found ENC28j60 device !! Function ended !!
> Enc28J60Network::sendPacket(memhandle handle) ERROR:ENC28j50 Device not found !! Bypass sendPacket function !!
> UIPEthernetClass::tick() ERROR:EREVID=0 -> Not found ENC28j60 device !! Function ended !!
> UIPEthernetClass::tick() ERROR:EREVID=0 -> Not found ENC28j60 device !! Function ended !!
> 

Looks like the ENC28J60 not found due to the wrong communication b/w the board & ENC module.

- As per the schematic from feather M0 board schematic https://learn.adafruit.com/assets/32914 They have been used SPI PINs 19-MOSI, 20-SCK,21-MISO,CS-11.  They did not used D11, D12, D13 & D4_RST pins for SPI. Please correct me If I am wrong. I am not yet sure what is the right PIN configuration for Feather M0 with ENC28J60.

ethicstechOSS, 2018-03-10:
Greetings! working on Feather M0 board.
- I did not used any external pullups
- I used the same default SS pin. You have to connect A2/PIN8.
- I used the following PIN connections for future reference on Feather M0 board:
ENC28J60 - Adafruit Feather M0 LoRa board:
SI - MOSI (PIN #19)
SO - MISO (PIN #21)
SCK - SCK (PIN #20)
CS - CS/SS (PIN #8)
VCC - +3.3V
GND - GND

My sample MQTT program output for reference : 

> MQTT Arduino Demo on Feather M0 LoRa Board
> 
> ENC28J60::init INFO: Chip erevid=6 initialization completed.
> MemoryPool::freeBlock(memhandle handle) WARNING: Don't free NOBLOCK handle
> MemoryPool::freeBlock(memhandle handle) WARNING: Don't free NOBLOCK handle
> MemoryPool::freeBlock(memhandle handle) WARNING: Don't free NOBLOCK handle
> MemoryPool::freeBlock(memhandle handle) WARNING: Don't free NOBLOCK handle
> MemoryPool::freeBlock(memhandle handle) WARNING: Don't free NOBLOCK handle
> Ethernet configured via DHCP
> IP address: 192.168.0.115
> 
> MQTT client configured
> DHT sensor initialized
> 
> Ready to send data
> Temperature: nan
> 
My sincere thanks to @cassy73

cassy73, 2018-03-11:
I'm really glad you did it.

I would like to ask you some questions:
What IDE did you use? (Arduino IDE?)
Which board did you install and use? (Adafruit SAMD boards by Adafruit?)
Does "pin19" mean "phisical pin 19"/"IDE pin23"/PB10 ?

Best Regards

ethicstechOSS, 2018-03-13:
Hi.
- I tried from Arduino IDE 1.8.5 on Linux platform.
- We have our clone board of Adafruit Feather M0 LoRa, https://www.adafruit.com/product/3178
-  Sorry for this PIN number confusion again. I edited my comment. Please check it.

cassy73, 2018-03-14:
Hi!

THX

Did You install "Adafruit SAMD boards by Adafruit" in arduino board manager?
And You compiled the program to "Adafruit Feather M0"?

Best Regards

ethicstechOSS, 2018-03-15:
Yes. I did.

closed 2018-03-19 by cassy73

rsispal, 2018-04-09:
Hi, 

I am also trying to get an ENC28J60 module connected to my Adafruit Feather M0 with ATWINC1500, with no success so far. 

I wonder if the library is using pins that are potentially being used by the onboard ATWINC1500 SPI connection.

On the Feather M0, I have connected the module as follows, as per @ethicstechOSS comment:

ENC28J60 =>Feather M0
- SI => MISO (PA12 / IDE pin 22)
- SO => MOSI (PB10 / IDE pin 23)
- CS => A2 (PB09 / IDE pin 16)
- SCK => SCK (PB11 / IDE pin 24)

I am only getting the following with `LOGLEVEL_DEBUG`, as if the Feather isn’t even identifying the ENC28J60.

```
UIPEthernetClass::tick() ERROR:EREVID=0 -> Not found ENC28j60 device !! Function ended !!
UIPEthernetClass::tick() DEBUG_V3:Function started
Enc28J60Network::geterevid(void) DEBUG_V3:Function started and return:0
UIPEthernetClass::tick() ERROR:EREVID=0 -> Not found ENC28j60 device !! Function ended !!
UIPClient::operator bool() DEBUG_V3:Function started
UIPEthernetClass::tick() DEBUG_V3:Function started
Enc28J60Network::geterevid(void) DEBUG_V3:Function started and return:0
UIPEthernetClass::tick() ERROR:EREVID=0 -> Not found ENC28j60 device !! Function ended !!
UIPEthernetClass::tick() DEBUG_V3:Function started
Enc28J60Network::geterevid(void) DEBUG_V3:Function started and return:0
UIPEthernetClass::tick() ERROR:EREVID=0 -> Not found ENC28j60 device !! Function ended !!
UIPClient::operator bool() DEBUG_V3:Function started
UIPEthernetClass::tick() DEBUG_V3:Function started
Enc28J60Network::geterevid(void) DEBUG_V3:Function started and return:0
UIPEthernetClass::tick() ERROR:EREVID=0 -> Not found ENC28j60 device !! Function ended !!
UIPEthernetClass::tick() DEBUG_V3:Function started
Enc28J60Network::geterevid(void) DEBUG_V3:Function started and return:0
UIPEthernetClass::tick() ERROR:EREVID=0 -> Not found ENC28j60 device !! Function ended !!
```
Library version 2.0.6
IDE: PlatformIO with local library installation

I can confirm that my ENC28J60 modules are functional, when using an Arduino Nano (LOGLEVEL_DEBUG):

```
ENC28J60::init DEBUG:csPin = 10
ENC28J60::init DEBUG:miso = 12
ENC28J60::init DEBUG:mosi = 11
ENC28J60::init DEBUG:sck = 13
ENC28J60::init DEBUG:Use Native hardware SPI
ENC28J60::init INFO: Chip erevid=6 initialization completed.
Enc28J60Network::receivePacket(void) DEBUG:receivePacket [6-42], next: 46, stat: C0, Packet count: 1 -> OK
Enc28J60Network::receivePacket(void) DEBUG: rxstat OK. receivePkt.size=60
UIPEthernetClass::tick() DEBUG:receivePacket: 255
UIPEthernetClass::tick() DEBUG:readPacket type IP, uip_len: 60
UIPEthernetClass::tick() DEBUG:freeing packet: 255
Enc28J60Network::setERXRDPT(void) DEBUG:Set actnextPacketPtr:44
```

cassy73, 2018-04-10:
Hi!

This is a SPI communication error.
Do You have this board?
https://www.adafruit.com/product/3010
See page 11 of board manual:
https://cdn-learn.adafruit.com/downloads/pdf/adafruit-feather-m0-wifi-atwinc1500.pdf
"
#8 - used as the Chip Select pin for the WiFi module, used to select it for SPI data transfer
MOSI / MISO /SCK - the SPI pins are also used for WiFi module communication
"

The ATWINC1500 board connected to ATSAMD on SPI bus, and use same CS pin,
then communication will not be good.

Please try another CS pin.
You can modify this in utility/Enc28J60Network.h file.
   #elif defined(ARDUINO_ARCH_SAMD)
      #define ENC28J60_CONTROL_CS     SS
Change "#define ENC28J60_CONTROL_CS     SS" to "#define ENC28J60_CONTROL_CS 9"

9 is A3 pin.

Best regards