UIPEthernet/UIPEthernet #31: DHCP program will not work, Static IP works OK

soccou, 2018-03-12:
I am hoping someone might have some suggestions for me. Running a very simple test Webserver on port 80 works fine if I use a static IP. As soon as I switch over to trying to use a DHCP address the same code no longer accepts any connections. If I run a nmap port scan it says nothing is open. 

Here is the only change made to the code between working and not
> Ethernet.begin(mac, ip);
  server.begin();
  Serial.print("server is at ");
  Serial.println(Ethernet.localIP());`

> if (Ethernet.begin(mac)) {
    Serial.println(Ethernet.localIP());
  } else{
    Serial.println("Ethernet failed");
  }

I tried turning on logging and doing a simple curl GET to that IP and this is the relevant output at that time. Not sure it helps at all:

> Enc28J60Network::receivePacket(void) DEBUG:receivePacket [212-25C], next: 260, stat: C0, Packet count: 1 -> OK
Enc28J60Network::receivePacket(void) DEBUG: rxstat OK. receivePkt.size=74
UIPEthernetClass::tick() DEBUG:receivePacket: 255
UIPEthernetClass::tick() DEBUG:readPacket type IP, uip_len: 74
UIPEthernetClass::upper_layer_chksum(uint8_t proto) DEBUG:uip_buf[34-74]: FFFF
UIPEthernetClass::upper_layer_chksum(uint8_t proto) DEBUG:uip_buf[34-54]: E6F1
UIPEthernetClass::network_send() DEBUG:uip_buf (uip_len): 54, packet: 1
Enc28J60Network::sendPacket(memhandle handle) DEBUG:sendPacket(1) [800-836]: 0 78 AC C0 AC C4 BC DE AD BE EF FE AB 8 0 45 0 0 28 0 B 0 0 40 6 F6 E6 C0 A8 1 8D C0 A8 1 1 0 50 AF 4C 0 0 0 0
UIPEthernetClass::tick() DEBUG:freeing packet: 255

Is this a library issue?

closed 2018-03-12 by soccou

baaridunnasr, 2019-02-18:
@soccou  Was this issue resolved? I am not able to get dynamically assigned IPs in NodeMCU ESP8266 V1. I have to use
```
byte mac[] = { 0xDE, 0xAD, 0xBE, 0xEF, 0xFE, 0xED };
EthernetClient client;
IPAddress IP(10,90,1,40);

void setup()
{
  Serial.begin(9600);
  Serial.println("Start...");

  Ethernet.begin(mac,IP);  // <-------Explicit IP declaration
 // rest of the code
```

luzrain, 2021-07-17:
Is there any solution? I can't get ip with dhcp too.