UIPEthernet/UIPEthernet #34: EN28J60 and MQTT pubsub

TLS1000, 2018-06-15:
Hi, I've been using the lib from truchess before. Every time after a few days my server didn't respond anymore. I updated to this latest version and it works, hopefully for more than a few days., fingers crossed.

I see you made EN28J60 compatible with ESP32, so basically future-proofing it, great work! Can the EN28J60 chip be used for the IOT, this MQTT protocol? The pubsub library: [](https://github.com/knolleary/pubsubclient/) states it cannot work with EN28J60 chip. There is reference to a 7-year old library: [](https://github.com/njh/NanodeMQTT). This nanode is specific hardware with the 28J60 chip. 

Could it work on let's say a mega with 28J60?

TLS1000, 2018-06-18:
Dear Cassy, the library is stable. While the original truchess version hangs after one or two days, with your updated version it keeps responding after 5 days now. Great work!

I read that the pubsub library would work on the uip library, contrary to what's indicated in the readme. I'll post my findings after test.

baaridunnasr, 2019-02-18:
Hi, do you mind sharing the wiring diagram for ESP32? Is it the same as the NodeMCU ESP8266 V1?

TLS1000, 2019-02-18:
hi, I would not recommend using the 28J60 for new projects. It is cheap bus has always been a problemchild.

baaridunnasr, 2019-02-18:
But the problem is that LAN8720 modules are costly and are not readily available in my place.

TLS1000, 2019-02-18:
you can assign any ESP32 pins you like to the 28J60. take the pins starting with 'd' on your board and you'll be fine.

jandrassy, 2019-02-18:
@TLS1000, can I close this issue?

TLS1000, 2019-02-18:
yes please

closed 2019-02-18 by TLS1000