UIPEthernet/UIPEthernet #41: UIPEthernet Light

zasquash, 2018-09-08:
Hello,

Is it possible to reduce the number of functionality to ICMP, static IP, hostname TCP UDP because the library occupies 60% of memory of a nano 3.0 with the Ethernet shield ENC28J60.

Zaq.

jandrassy, 2019-01-23:
yo can turn off UDP  in uipethernet-conf.h and save 5 kB of flash memory

davidfaj, 2019-02-02:
Thanks, great suggestion! It reduced 7.5 kb of program storage, and 400 bytes of dynamic memory.
In case anyone is having a compiling issue when turning off UDP, which I did, I also needed to begin the Ethernet assigning a static IP (`Ethernet.begin(mac, ip, dns)`), instead of using DHCP (`Ethernet.begin(mac)`). Further I learned that DHCP uses UDP, making sense this additional step.

davidmak2709, 2019-05-28:
Hi!
I'm also trying to save some space on my AtMega (with the ENC28j60) by setting to 0 the UIP_CONF_UDP   flag and assigning a static IP as @davidfaj  mentioned: 

`Ethernet.begin(mac,IPAddress(192,168,5,100),IPAddress(192,168,5,1),IPAddress(192,168,5,1));`

But it seems that it does not work in my case or I'm doing something wrong. 

When the UIP_CONF_UDP flag is set to 1 and I run  `nmap -sn 192.168.5.0/24`  to check if the AtMega is connected to my router the AtMega is listed in the results, but on the other hand when the flag is 0 its not.

jandrassy, 2020-08-28:
so nmap can't reach it if it doesn't respond to UDP. that is all

closed 2020-10-03 by jandrassy