UIPEthernet/UIPEthernet #42: Bluepill wiring diagram

vmasdani, 2018-09-11:
Hi.
I've been trying to find out how this library works with Bluepill and ENC28J60 for days and found no solution, it just kept disconnecting for about 30 secs after it starts. Just this day I tried inverting the position of the pins from the documentation in the "hardware" folder and it worked. I at least tried the simple chat server and it's working quite flawlessly.

I inverted the position 
from this:
CS->PA7
SCK->PA6
MISO->PA5
MOSI->PA4

to this:
CS->PA4
SCK->PA5
MISO->PA6
MOSI->PA7

Is anyone experiencing the same thing or is it my module that's just defective?

jandrassy, 2019-01-23:
duplicate of https://github.com/UIPEthernet/UIPEthernet/issues/35

jarwil, 2019-09-02:
Yes, I had the same problem.

CS->PA4
SCK->PA5
MISO->PA6
MOSI->PA7

This is correct wiring.

jandrassy, 2020-02-16:
yes, the picture is wrong.
@cassy73 do you have the picture's source?

jandrassy, 2020-02-23:
I corrected the picture

closed 2020-02-23 by jandrassy