UIPEthernet/UIPEthernet #43: Sketch stop working all 12 Days

Ravenbs, 2018-09-12:
Hello,

I have simple sketch. 
Reading analog value and posting it t a server once a Minute.
This runs for ~12 Days, then stop working.
Repeated 3 times, always with this 12 Days in between.

I do not use counters or times in my own code.
I use Version 2.0.6 on Arduino Nano with shield.

```
void loop() 
{
  int current1 = analogRead( currentIN );
  delay( 50 );
  int current2 = analogRead( currentIN );
  delay( 50 );
  int current3 = analogRead( currentIN );
  delay( 50 );
  int current4 = analogRead( currentIN );
  delay( 50 );
  int current5 = analogRead( currentIN );
  
  Serial.println( current1 );
  Serial.println( current2 );
  Serial.println( current3 );
  Serial.println( current4 );
  Serial.println( current5 );

  int current = (current1 + current2 + current3 +current4 + current5 ) / 5;
  
  sendValueToServer( 1, current );

  // ms
  delay( 10 * 1000 );
  delay( 10 * 1000 );
  delay( 10 * 1000 );
  delay( 10 * 1000 );
  delay( 10 * 1000 );
  delay( 10 * 1000 );
  
}

void sendValueToServer(int channel, int value)
{
  Serial.print("Send Data C: ");
  Serial.print( channel );
  Serial.print( " v= " );
  Serial.println( value );

  //client.maintain();

    // if you get a connection, report back via serial:
  if (client.connect(server, 80)) {
    Serial.println("connected");
    // Make a HTTP request:

    //client.println("GET /current.php HTTP/1.1");
    //client.println("GET /current.php?q=arduino HTTP/1.1");
    
    client.print("GET /current.php?s=");
    client.print( channel );
    client.print("&v=");
    client.print(value);
    client.println(" HTTP/1.1");
    
    client.println("Host: 192.168.178.46");
    client.println("Connection: close");
    client.println("User-Agent: Nano1");
    client.println();
  } else {
    // if you didn't get a connection to the server:
    Serial.println("connection failed");
  }

  // Wait for the server to finish
  while( client.connected() )
  {
    // if there are incoming bytes available
    // from the server, read them and print them:
    while (client.available()) 
    {
      char c = client.read();
      Serial.print(c);
    }
  }
  
  // if the server's disconnected, stop the client:
  if (!client.connected()) 
  {
    Serial.println();
    Serial.println("disconnecting.");
    client.stop();
  }
    
}

```

Any ideas?

jandrassy, 2019-02-11:
DHCP IP lease time? you have `maintain()` commented

Ravenbs, 2019-02-11:
Thank you for the tip.
Had the same idea a few month ago, but this has not solved the problem.
Also it does not help to use the watchdog, because the ethernet shield can not be re-initialised. 

Its still 2 to 4 weeks, then fails.
Does this work for others for longer times?

jandrassy, 2019-02-12:
I use a watchdog and it resets fine. Do you have the reset pins wired together?

Ravenbs, 2019-02-12:
It is this shield:
https://www.ebay.de/itm/USB-Nano-V4-0-V3-0-Atmega328P-Board-ENC28J60-Ethernet-Shield-Module-For-Arduino/272512183829?hash=item3f72fdee15:m:mDu21D9-CcenUBtCW4U1hzA:rk:1:pf:1&frcectupt=true

I do not think the arduino is triggering the reset pin when he does a watchdog reset?
Think this is only an internal software thing jumping the program counter to 0x0.

Would be very interest in any other long-rung experience.
Because for short (a few weeks) this is fine, only long run it does not work :(

jandrassy, 2019-02-12:
watchdog does a full reset, not a jump to 0x0. it is a separate circuit. but I am not sure if it sets reset pin low. I am on M0

jandrassy, 2019-02-12:
the shield has reset on pin 7 of Nano. set it LOW for a moment in setup() and then change it back to INPUT.

watchdog does a full reset, not only a jump to 0x0. it is a separate circuit in the MCU

Ravenbs, 2019-02-12:
Where you found the scematic that it is pin 7? 
The shield is labeled as the arduino itself.

Will try it and report if watchdog is then working - but can do it in a few hours.

jandrassy, 2019-02-12:
https://robotdyn.com/catalog/arduino/shields/nano-v3-ethernet-shield-enc28j60.html

it is common to use pin 7 as reset pin for Ethernet

Ravenbs, 2019-02-12:
Watchdog does not work, because I think I am affected by this "Arduino Nano bootloader watchdog glitch", so the unit is not booting up after watchdog reset. I try to install optiboot as replacement.

Ravenbs, 2019-02-12:
Using the new Bootloader "Optiboot" the watchdog is now working - also with ethernet after reboot.
Think this solves not the problem with ethernet issues all 12 Days, but now i can reboot the system once a day to avoid this. 

Thank you Jandrassy to point me to this old project and finish it :).

BlueAndi, 2019-02-12:
Does the sketch freezes or do you still see the serial output, but only miss the value sent to the server?

jandrassy, 2019-02-12:
right, I forgot to ask you to test the current version.

Ravenbs, 2019-02-12:
Can't have a look at serial when this happen, because when you reconnect serial, the Arduino resets. 
And the sensor is not near the pc where it runs for weeks.
So unfortunately I never was able to capture this by serial.

What I can say for shure - when this happen, the watchdog fires. 
So the main loop hangs.
And because it only do this analog read and sending data, it is very likely that the transmitting is hanging.

jandrassy, 2019-02-12:
I fixed two bugs so try the current repo version

underwoodblog, 2019-02-12:
When you have Pi running, you can install ser2net (https://github.com/cminyard/ser2net) and connect a ESP8266 (https://github.com/dparnell/esp8266-ser2net)  on your arduino (RX/TX) to log. Or another arduino with ethernet interface (https://github.com/stepansnigirev/ArduinoSerialToEthernet).

Or maby you can use this in you browser: https://github.com/espterm/espterm-firmware

Ravenbs, 2019-02-12:
I will try it Jandrassy, but test will need at least 2 weeks to see if this happen again.

jandrassy, 2019-04-16:
> I will try it Jandrassy, but test will need at least 2 weeks to see if this happen again.

@Ravenbs, can we close this?

Ravenbs, 2019-04-16:
The problem with UIPEthernet still exists, but I installed a new Bootloeader and therefore use the watchdog to reset the devices once a week.

KmanOz, 2019-09-21:
I also have this issue but takes less that 1 week. I'm doing a fair bit with Ethernet though. The same code with the standard Ethernet library for a Wizen chip runs indefinitely. I hope someone works this out otherwise these cheap replacements can't be used in any serious application.

Ravenbs, 2019-09-21:
I dont know about a fix, still here as well. But because this is so rare in my application I have no chance to capture the moment of occurance. Can you share your code where it is happening more often?

jandrassy, 2020-10-03:
try 2.0.9

closed 2020-10-03 by jandrassy