UIPEthernet/UIPEthernet #44: Initialization fails when SD Card module is connected

phd, 2018-09-16:
Environment
----------

UIPEthernet version: 2.0.6

Enc28J60Network.h: `#define ENC28J60_CONTROL_CS 49`

SD Card module simply connected to CS 53, not even programmed (53 just set to OUTPUT HIGH).

Sketch: example `UdpClient.ino`

Code added to Sketch:
```
pinMode(53, OUTPUT);
digitalWrite(53, HIGH);
pinMode(49, OUTPUT);
digitalWrite(49, HIGH);
```

Result
----------

"**Not found ENC28j60 device**" error as below.

EtherCard library works fine in this configuration.

UIPEthernet works in this configuration, but without SD Card module connected.

Debug
----------

V2:
```
ENC28J60::init DEBUG:csPin = 49
ENC28J60::init DEBUG:miso = 50
ENC28J60::init DEBUG:mosi = 51
ENC28J60::init DEBUG:sck = 52
ENC28J60::init DEBUG:Use Native hardware SPI
Enc28J60Network::phyWrite ERROR:TIMEOUT !!
Enc28J60Network::phyWrite ERROR:TIMEOUT !!
ENC28J60::init INFO: Chip erevid=0 initialization completed.
UIPEthernetClass::tick() ERROR:EREVID=0 -> Not found ENC28j60 device !! Function ended !!

```

V3:
```
UIPEthernetClass::begin(const uint8_t* mac, IPAddress ip) DEBUG_V3:Function started
UIPEthernetClass::begin(const uint8_t* mac, IPAddress ip, IPAddress dns) DEBUG_V3:Function started
UIPEthernetClass::begin(const uint8_t* mac, IPAddress ip, IPAddress dns, IPAddress gateway) DEBUG_V3:Function started
UIPEthernetClass::begin(const uint8_t* mac, IPAddress ip, IPAddress dns, IPAddress gateway, IPAddress subnet) DEBUG_V3:Function started
UIPEthernetClass::init(const uint8_t* mac) DEBUG_V3:Function started
Enc28J60Network::init(uint8_t* macaddr) DEBUG_V3:Function started
MemoryPool::init() DEBUG_V3:Function started
ENC28J60::init DEBUG:csPin = 49
ENC28J60::init DEBUG:miso = 50
ENC28J60::init DEBUG:mosi = 51
ENC28J60::init DEBUG:sck = 52
ENC28J60::init DEBUG:Use Native hardware SPI
Enc28J60Network::writeOp(uint8_t op, uint8_t address, uint8_t data) DEBUG_V3:Function started
ENC28J60::init DEBUG_V3:Before readOp(ENC28J60_READ_CTRL_REG, ESTAT)
Enc28J60Network::readOp(uint8_t op, uint8_t address) DEBUG_V3:Function started
ENC28J60::init DEBUG_V3:After readOp(ENC28J60_READ_CTRL_REG, ESTAT)
Enc28J60Network::writeRegPair(uint8_t address, uint16_t data) DEBUG_V3:Function started
Enc28J60Network::setBank(uint8_t address) DEBUG_V3:Function started
Enc28J60Network::writeOp(uint8_t op, uint8_t address, uint8_t data) DEBUG_V3:Function started
Enc28J60Network::writeOp(uint8_t op, uint8_t address, uint8_t data) DEBUG_V3:Function started
Enc28J60Network::writeOp(uint8_t op, uint8_t address, uint8_t data) DEBUG_V3:Function started
Enc28J60Network::writeOp(uint8_t op, uint8_t address, uint8_t data) DEBUG_V3:Function started
ENC28J60::init DEBUG_V3:After writeRegPair(ERXSTL, RXSTART_INIT)
[... 171 lines cut ...]
Enc28J60Network::readOp(uint8_t op, uint8_t address) DEBUG_V3:Function started
Enc28J60Network::readReg(uint8_t address) DEBUG_V3:Function started
Enc28J60Network::setBank(uint8_t address) DEBUG_V3:Function started
Enc28J60Network::readOp(uint8_t op, uint8_t address) DEBUG_V3:Function started
Enc28J60Network::readReg(uint8_t address) DEBUG_V3:Function started
Enc28J60Network::setBank(uint8_t address) DEBUG_V3:Function started
Enc28J60Network::readOp(uint8_t op, uint8_t address) DEBUG_V3:Function started
Enc28J60Network::readReg(uint8_t address) DEBUG_V3:Function started
Enc28J60Network::setBank(uint8_t address) DEBUG_V3:Function started
Enc28J60Network::readOp(uint8_t op, uint8_t address) DEBUG_V3:Function started
Enc28J60Network::readReg(uint8_t address) DEBUG_V3:Function started
Enc28J60Network::setBank(uint8_t address) DEBUG_V3:Function started
Enc28J60Network::readOp(uint8_t op, uint8_t address) DEBUG_V3:Function started
Enc28J60Network::phyWrite ERROR:TIMEOUT !!
ENC28J60::init DEBUG_V3:Before readReg(EREVID);
Enc28J60Network::readReg(uint8_t address) DEBUG_V3:Function started
Enc28J60Network::setBank(uint8_t address) DEBUG_V3:Function started
Enc28J60Network::readOp(uint8_t op, uint8_t address) DEBUG_V3:Function started
ENC28J60::init INFO: Chip erevid=0 initialization completed.
UIPEthernetClass::configure(IPAddress ip, IPAddress dns, IPAddress gateway, IPAddress subnet) DEBUG_V3:Function started
UIPUDP::beginPacket(IPAddress ip, uint16_t port) DEBUG_V3:Function started
UIPEthernetClass::tick() DEBUG_V3:Function started
Enc28J60Network::geterevid(void) DEBUG_V3:Function started and return:0
UIPEthernetClass::tick() ERROR:EREVID=0 -> Not found ENC28j60 device !! Function ended !!
```

jandrassy, 2019-01-16:
and on which pin is the ENC28j60  SS if SD is on 53?

jandrassy, 2019-01-16:
this is a problem with most cheap SD adapters. they disturb other SPI devices with level converter on MISO line. this is not an UIPEthernet issue and can be closed.

jandrassy, 2019-01-23:
@phd, can I close this?

closed 2019-02-09 by jandrassy