UIPEthernet/UIPEthernet #48: Stops working after some hours

BlueAndi, 2018-11-07:
I built a http server with the UIPEthernet lib. It works so far, but after some hours it is not possible to access any page on the server. Can someone help to examine the log output?

Every kind of log output is stopped as well. I already verified that the sketch loop is still running, there is also no kind of infinite loop, otherwise the watchdog would have triggered a reset. 

> 21:48:14.012 -> Enc28J60Network::receivePacket(void) DEBUG:receivePacket [20A-249], next: 24E, stat: C0, Packet count: 4 -> OK
> 21:48:14.050 -> Enc28J60Network::receivePacket(void) DEBUG: rxstat OK. receivePkt.size=63
> 21:48:14.050 -> UIPEthernetClass::tick() DEBUG:receivePacket: 255
> 21:48:14.050 -> Enc28J60Network::readPacket(memhandle handle, memaddress position, uint8_t* buffer, uint16_t len) DEBUG_V2: Read bytes:63 save to block(255) [0]: FF FF FF FF FF FF 0 8 9B E8 E4 1D 8 0 45 0 0 31 1D 89 40 0 40 11 98 AE C0 A8 1 35 C0 A8 1 FF 89 CC 7E 9C 0 1D FC 87 4D 2D 53 45 41 52 43 48 20 2A 20 48 54 54 50 2F 31 2E 31 D A 
> 21:48:14.097 -> UIPEthernetClass::tick() DEBUG:readPacket type IP, uip_len: 63
> 21:48:14.097 -> UIPEthernetClass::upper_layer_chksum(uint8_t proto) DEBUG:uip_buf[34-42]: C189
> 21:48:14.097 -> UIPEthernetClass::upper_layer_chksum(uint8_t proto) DEBUG:uip_packet(255)[42-63]: FFFF
> 21:48:14.097 -> UIPEthernetClass::tick() DEBUG:freeing packet: 255
> 21:48:14.097 -> Enc28J60Network::setERXRDPT(void) DEBUG:Set actnextPacketPtr:24C
> 21:48:14.097 -> Enc28J60Network::receivePacket(void) DEBUG:receivePacket [254-293], next: 298, stat: C0, Packet count: 4 -> OK
> 21:48:14.150 -> Enc28J60Network::receivePacket(void) DEBUG: rxstat OK. receivePkt.size=63
> 21:48:14.150 -> UIPEthernetClass::tick() DEBUG:receivePacket: 255
> 21:48:14.150 -> Enc28J60Network::readPacket(memhandle handle, memaddress position, uint8_t* buffer, uint16_t len) DEBUG_V2: Read bytes:63 save to block(255) [0]: FF FF FF FF FF FF 0 8 9B E8 E4 1D 8 0 45 0 0 31 6F C2 40 0 40 11 46 6D C0 A8 1 3D C0 A8 1 FF C1 49 7E 9E 0 1D C5 0 4D 2D 53 45 41 52 43 48 20 2A 20 48 54 54 50 2F 31 2E 31 D A 
> 21:48:14.150 -> UIPEthernetClass::tick() DEBUG:readPacket type IP, uip_len: 63
> 21:48:14.150 -> UIPEthernetClass::upper_layer_chksum(uint8_t proto) DEBUG:uip_buf[34-42]: C189
> 21:48:14.150 -> UIPEthernetClass::upper_layer_chksum(uint8_t proto) DEBUG:uip_packet(255)[42-63]: FFFF
> 21:48:14.197 -> UIPEthernetClass::tick() DEBUG:freeing packet: 255
> 21:48:14.197 -> Enc28J60Network::setERXRDPT(void) DEBUG:Set actnextPacketPtr:296
> 21:48:14.197 -> Enc28J60Network::receivePacket(void) DEBUG:receivePacket [29E-2DD], next: 2E2, stat: C0, Packet count: 3 -> OK
> 21:48:14.197 -> Enc28J60Network::receivePacket(void) DEBUG: rxstat OK. receivePkt.size=63
> 21:48:14.197 -> UIPEthernetClass::tick() DEBUG:receivePacket: 255
> 21:48:14.197 -> Enc28J60Network::readPacket(memhandle handle, memaddress position, uint8_t* buffer, uint16_t len) DEBUG_V2: Read bytes:63 save to block(255) [0]: FF FF FF FF FF FF 0 8 9B E8 E4 1D 8 0 45 0 0 31 1D 8A 40 0 40 11 98 AD C0 A8 1 35 C0 A8 1 FF 8C 22 7E 9E 0 1D FA 2F 4D 2D 53 45 41 52 43 48 20 2A 20 48 54 54 50 2F 31 2E 31 D A 
> 21:48:14.251 -> UIPEthernetClass::tick() DEBUG:readPacket type IP, uip_len: 63
> 21:48:14.251 -> UIPEthernetClass::upper_layer_chksum(uint8_t proto) DEBUG:uip_buf[34-42]: C189
> 21:48:14.251 -> UIPEthernetClass::upper_layer_chksum(uint8_t proto) DEBUG:uip_packet(255)[42-63]: FFFF
> 21:48:14.251 -> UIPEthernetClass::tick() DEBUG:freeing packet: 255
> 21:48:14.251 -> Enc28J60Network::setERXRDPT(void) DEBUG:Set actnextPacketPtr:2E0
> 21:48:14.251 -> Enc28J60Network::receivePacket(void) DEBUG:receivePacket [2E8-324], next: 328, stat: C0, Packet count: 2 -> OK
> 21:48:14.298 -> Enc28J60Network::receivePacket(void) DEBUG: rxstat OK. receivePkt.size=60
> 21:48:14.298 -> UIPEthernetClass::tick() DEBUG:receivePacket: 255
> 21:48:14.298 -> Enc28J60Network::readPacket(memhandle handle, memaddress position, uint8_t* buffer, uint16_t len) DEBUG_V2: Read bytes:60 save to block(255) [0]: 0 22 F9 1 27 EB B8 27 EB 2D 82 22 8 0 45 0 0 28 2B 29 40 0 40 6 8C 38 C0 A8 1 1C C0 A8 1 2 C9 AC 0 50 22 EF 17 20 0 12 C4 B2 50 10 76 0 ED 94 0 0 0 0 0 0 0 0 
> 21:48:14.298 -> UIPEthernetClass::tick() DEBUG:readPacket type IP, uip_len: 60
> 21:48:14.298 -> UIPEthernetClass::upper_layer_chksum(uint8_t proto) DEBUG:uip_buf[34-54]: FFFF
> 21:48:14.351 -> uipclient_appcall(void) DEBUG:UIPClient uip_acked
> 21:48:14.351 -> UIPClient::_eatBlock DEBUG:eatblock(1): 1 0 0 0 0 -> 0 0 0 0 0 
> 21:48:14.351 -> uipclient_appcall(void) DEBUG_V2:UIPClient state UIP_CLIENT_CLOSE
> 21:48:14.351 -> UIPClient::_dumpAllData() DEBUG_V2:UIPClient::all_data[0], state:110000
> 21:48:14.351 -> packets_in: 0 0 0 0 0 
> 21:48:14.351 -> packets_out: 0 0 0 0 0 
> 21:48:14.351 -> out_pos: 305
> 21:48:14.351 -> UIPClient::_dumpAllData() DEBUG_V2:UIPClient::all_data[1], state:0
> 21:48:14.351 -> packets_in: 0 0 0 0 0 
> 21:48:14.351 -> packets_out: 0 0 0 0 0 
> 21:48:14.351 -> out_pos: 0
> 21:48:14.351 -> UIPClient::_dumpAllData() DEBUG_V2:UIPClient::all_data[2], state:0
> 21:48:14.351 -> packets_in: 0 0 0 0 0 
> 21:48:14.351 -> packets_out: 0 0 0 0 0 
> 21:48:14.351 -> out_pos: 0
> 21:48:14.398 -> UIPClient::_dumpAllData() DEBUG_V2:UIPClient::all_data[3], state:0
> 21:48:14.398 -> packets_in: 0 0 0 0 0 
> 21:48:14.398 -> packets_out: 0 0 0 0 0 
> 21:48:14.398 -> out_pos: 0
> 21:48:14.398 -> uipclient_appcall(void) DEBUG_V2:no blocks out -> free userdata
> 21:48:14.398 -> UIPClient::_dumpAllData() DEBUG_V2:UIPClient::all_data[0], state:0
> 21:48:14.398 -> packets_in: 0 0 0 0 0 
> 21:48:14.398 -> packets_out: 0 0 0 0 0 
> 21:48:14.398 -> out_pos: 305
> 21:48:14.398 -> UIPClient::_dumpAllData() DEBUG_V2:UIPClient::all_data[1], state:0
> 21:48:14.398 -> packets_in: 0 0 0 0 0 
> 21:48:14.398 -> packets_out: 0 0 0 0 0 
> 21:48:14.398 -> out_pos: 0
> 21:48:14.398 -> UIPClient::_dumpAllData() DEBUG_V2:UIPClient::all_data[2], state:0
> 21:48:14.398 -> packets_in: 0 0 0 0 0 
> 21:48:14.451 -> packets_out: 0 0 0 0 0 
> 21:48:14.451 -> out_pos: 0
> 21:48:14.451 -> UIPClient::_dumpAllData() DEBUG_V2:UIPClient::all_data[3], state:0
[... 1137 lines cut ...]
> 21:50:11.422 -> Enc28J60Network::receivePacket(void) DEBUG:receivePacket [F3AD-6D2], next: D988, stat: 2B, Packet count: 8 -> failed
> 21:50:11.422 -> Enc28J60Network::setERXRDPT(void) DEBUG:Set actnextPacketPtr:0
> 21:50:11.422 -> Enc28J60Network::receivePacket(void) DEBUG:receivePacket [D18F-9E], next: FD16, stat: 94, Packet count: 7 -> OK
> 21:50:11.422 -> Enc28J60Network::setERXRDPT(void) DEBUG:Set actnextPacketPtr:0
> 21:50:11.422 -> Enc28J60Network::receivePacket(void) DEBUG:receivePacket [F51D-26A], next: 4FBA, stat: B7, Packet count: 6 -> OK
> 21:50:11.463 -> Enc28J60Network::setERXRDPT(void) DEBUG:Set actnextPacketPtr:0
> 21:50:11.463 -> Enc28J60Network::receivePacket(void) DEBUG:receivePacket [47C1-646], next: 87E5, stat: 44, Packet count: 5 -> failed
> 21:50:11.463 -> Enc28J60Network::setERXRDPT(void) DEBUG:Set actnextPacketPtr:0
> 21:50:11.463 -> Enc28J60Network::receivePacket(void) DEBUG:receivePacket [7FEC-12B], next: 4143, stat: 43, Packet count: 4 -> failed
> 21:50:11.463 -> Enc28J60Network::setERXRDPT(void) DEBUG:Set actnextPacketPtr:0
> 21:50:11.522 -> Enc28J60Network::receivePacket(void) DEBUG:receivePacket [394A-294], next: 0, stat: 35, Packet count: 3 -> failed
> 21:50:11.522 -> Enc28J60Network::setERXRDPT(void) DEBUG:Set actnextPacketPtr:7FE
> 21:50:11.522 -> Enc28J60Network::receivePacket(void) DEBUG:receivePacket [6-1], next: 3C0, stat: FF, Packet count: 2 -> OK
> 21:50:11.522 -> Enc28J60Network::receivePacket(void) DEBUG: rxstat OK. receivePkt.size=65531
> 21:50:11.522 -> UIPEthernetClass::tick() DEBUG:receivePacket: 255
> 21:50:11.522 -> Enc28J60Network::readPacket(memhandle handle, memaddress position, uint8_t* buffer, uint16_t len) DEBUG_V2: Read bytes:98 save to block(255) [0]: FF FF 0 8 9B E8 E4 1D 8 0 45 0 0 4E 71 32 40 0 40 11 44 E0 C0 A8 1 3D C0 A8 1 FF 0 89 0 89 0 3A 7D 13 7A 4B 1 10 0 1 0 0 0 0 0 0 20 46 48 45 50 46 43 45 4C 45 48 46 43 45 50 46 46 46 41 43 41 43 41 43 41 43 41 43 41 43 41 42 4E 0 0 20 0 1 BF 18 C9 6A C8 0 60 0 C0 3 
> 21:50:11.569 -> UIPEthernetClass::tick() DEBUG:freeing packet: 255
> 21:50:11.569 -> Enc28J60Network::setERXRDPT(void) DEBUG:Set actnextPacketPtr:3BE
> 21:50:11.569 -> Enc28J60Network::receivePacket(void) DEBUG:receivePacket [3C6-3C], next: A7D, stat: 1, Packet count: 1 -> failed
> 21:50:11.569 -> Enc28J60Network::setERXRDPT(void) DEBUG:Set actnextPacketPtr:0
> Enc28J60Network::receivePacket(void) DEBUG:receivePacket [284-5F9], next: 6772, stat: 6F, Packet count: 1 -> failed
> 21:50:12.905 -> Enc28J60Network::setERXRDPT(void) DEBUG:Set actnextPacketPtr:0
> Enc28J60Network::receivePacket(void) DEBUG:receivePacket [5F79-276], next: 0, stat: 3, Packet count: 1 -> failed
> 21:50:13.441 -> Enc28J60Network::setERXRDPT(void) DEBUG:Set actnextPacketPtr:7FE
>

jandrassy, 2019-01-23:
is this solved by https://github.com/UIPEthernet/UIPEthernet/pull/51 ?

BlueAndi, 2019-02-20:
Sry, but I didn't get it managed right now, because lack of time.
Lets close it and open it again on demand.

closed 2019-02-20 by BlueAndi