UIPEthernet/UIPEthernet #49: Can't compile for NodeMCU board in Arduino 1.8.8

Edward-CNRG-NTU, 2019-01-01:
I can't compile the basic example with ESP8266 and NodeMCU v1.0?
I don't understand the error stated by the compiler.
Does anyone know what problem I encountered? 

Thanks in advance.

```
In file included from C:\Users\edwar\OneDrive\文件\Arduino\libraries\UIPEthernet/UIPEthernet.h:37:0,

                 from C:\Users\edwar\OneDrive\??辣\ArduinoData\packages\esp8266\hardware\esp8266\2.5.0-beta2\libraries\Ethernet\examples\WebServer\WebServer.ino:19:

C:\Users\edwar\OneDrive\文件\Arduino\libraries\UIPEthernet/Dhcp.h:166:10: error: cannot declare field 'DhcpClass::_dhcpUdpSocket' to be of abstract type 'UIPUDP'

   UIPUDP _dhcpUdpSocket;

          ^

In file included from C:\Users\edwar\OneDrive\文件\Arduino\libraries\UIPEthernet/Dhcp.h:59:0,

                 from C:\Users\edwar\OneDrive\文件\Arduino\libraries\UIPEthernet/UIPEthernet.h:37,

                 from C:\Users\edwar\OneDrive\??辣\ArduinoData\packages\esp8266\hardware\esp8266\2.5.0-beta2\libraries\Ethernet\examples\WebServer\WebServer.ino:19:

C:\Users\edwar\OneDrive\文件\Arduino\libraries\UIPEthernet/UIPUdp.h:60:9: note:   because the following virtual functions are pure within 'UIPUDP':

   class UIPUDP : public UDP {

         ^

In file included from C:\Users\edwar\OneDrive\文件\Arduino\libraries\UIPEthernet/UIPUdp.h:34:0,

                 from C:\Users\edwar\OneDrive\文件\Arduino\libraries\UIPEthernet/Dhcp.h:59,

                 from C:\Users\edwar\OneDrive\文件\Arduino\libraries\UIPEthernet/UIPEthernet.h:37,

                 from C:\Users\edwar\OneDrive\??辣\ArduinoData\packages\esp8266\hardware\esp8266\2.5.0-beta2\libraries\Ethernet\examples\WebServer\WebServer.ino:19:

C:\Users\edwar\OneDrive\文件\ArduinoData\packages\esp8266\hardware\esp8266\2.5.0-beta2\cores\esp8266/Udp.h:81:27: note: 	virtual IPAddress UDP::remoteIP() const

         virtual IPAddress remoteIP() const =0;

                           ^

C:\Users\edwar\OneDrive\文件\ArduinoData\packages\esp8266\hardware\esp8266\2.5.0-beta2\cores\esp8266/Udp.h:83:26: note: 	virtual uint16_t UDP::remotePort() const

         virtual uint16_t remotePort() const =0;

                          ^

In file included from C:\Users\edwar\OneDrive\文件\Arduino\libraries\UIPEthernet/UIPEthernet.h:42:0,

                 from C:\Users\edwar\OneDrive\??辣\ArduinoData\packages\esp8266\hardware\esp8266\2.5.0-beta2\libraries\Ethernet\examples\WebServer\WebServer.ino:19:

C:\Users\edwar\OneDrive\文件\Arduino\libraries\UIPEthernet/UIPServer.h:49:13: error: invalid abstract return type for member function 'UIPClient UIPServer::available()'

   UIPClient available();

             ^

In file included from C:\Users\edwar\OneDrive\文件\Arduino\libraries\UIPEthernet/UIPEthernet.h:41:0,

                 from C:\Users\edwar\OneDrive\??辣\ArduinoData\packages\esp8266\hardware\esp8266\2.5.0-beta2\libraries\Ethernet\examples\WebServer\WebServer.ino:19:

C:\Users\edwar\OneDrive\文件\Arduino\libraries\UIPEthernet/UIPClient.h:75:9: note:   because the following virtual functions are pure within 'UIPClient':

   class UIPClient : public Client {

         ^

In file included from C:\Users\edwar\OneDrive\文件\Arduino\libraries\UIPEthernet/UIPClient.h:29:0,

                 from C:\Users\edwar\OneDrive\文件\Arduino\libraries\UIPEthernet/UIPEthernet.h:41,

                 from C:\Users\edwar\OneDrive\??辣\ArduinoData\packages\esp8266\hardware\esp8266\2.5.0-beta2\libraries\Ethernet\examples\WebServer\WebServer.ino:19:

C:\Users\edwar\OneDrive\文件\ArduinoData\packages\esp8266\hardware\esp8266\2.5.0-beta2\cores\esp8266/Client.h:29:21: note: 	virtual int Client::connect(const IPAddress&, uint16_t)

         virtual int connect(CONST IPAddress& ip, uint16_t port) =0;

                     ^

C:\Users\edwar\OneDrive\文件\ArduinoData\packages\esp8266\hardware\esp8266\2.5.0-beta2\cores\esp8266/Client.h:37:22: note: 	virtual bool Client::flush(unsigned int)

         virtual bool flush(unsigned int maxWaitMs = 0) = 0;

                      ^

C:\Users\edwar\OneDrive\文件\ArduinoData\packages\esp8266\hardware\esp8266\2.5.0-beta2\cores\esp8266/Client.h:38:22: note: 	virtual bool Client::stop(unsigned int)

         virtual bool stop(unsigned int maxWaitMs = 0) = 0;

                      ^

C:\Users\edwar\OneDrive\??辣\ArduinoData\packages\esp8266\hardware\esp8266\2.5.0-beta2\libraries\Ethernet\examples\WebServer\WebServer.ino: In function 'void loop()':

WebServer:53:44: error: cannot allocate an object of abstract type 'UIPClient'

In file included from C:\Users\edwar\OneDrive\文件\Arduino\libraries\UIPEthernet/UIPEthernet.h:41:0,

                 from C:\Users\edwar\OneDrive\??辣\ArduinoData\packages\esp8266\hardware\esp8266\2.5.0-beta2\libraries\Ethernet\examples\WebServer\WebServer.ino:19:

C:\Users\edwar\OneDrive\文件\Arduino\libraries\UIPEthernet/UIPClient.h:75:9: note:   since type 'UIPClient' has pure virtual functions

   class UIPClient : public Client {

         ^

WebServer:53:18: error: cannot declare variable 'client' to be of abstract type 'UIPClient'

In file included from C:\Users\edwar\OneDrive\文件\Arduino\libraries\UIPEthernet/UIPEthernet.h:41:0,

                 from C:\Users\edwar\OneDrive\??辣\ArduinoData\packages\esp8266\hardware\esp8266\2.5.0-beta2\libraries\Ethernet\examples\WebServer\WebServer.ino:19:

C:\Users\edwar\OneDrive\文件\Arduino\libraries\UIPEthernet/UIPClient.h:75:9: note:   since type 'UIPClient' has pure virtual functions

   class UIPClient : public Client {

         ^

exit status 1
cannot allocate an object of abstract type 'UIPClient'
```

jandrassy, 2019-02-09:
it is a problem of the esp8266 core. they changed the common Arduino base class Client in their core

closed 2019-02-09 by jandrassy

jandrassy, 2019-02-20:
use esp8266 core 2.4.2.