UIPEthernet/UIPEthernet #56: PROBLEM WITH SERVER IN ESP32

vallesolucoes, 2019-01-23:
```
GOOD AFTERNOON EVERYONE
I TRIED TO ROLL THE SIMPLE SERVER CODE
AND IT DOES NOT WORK
```

CODE:

#define MACADDRESS 0x00,0x01,0x02,0x03,0x04,0x05
#define MYIPADDR 192,168,1,6
#define MYIPMASK 255,255,255,0
#define MYDNS 192,168,1,1
#define MYGW 192,168,1,1
#define LISTENPORT 1000
#define UARTBAUD 9600

#include <UIPEthernet.h>
#include "utility/logging.h"

EthernetServer server = EthernetServer(LISTENPORT);

void setup() {

  uint8_t mac[6] = {MACADDRESS};
  uint8_t myIP[4] = {MYIPADDR};
  uint8_t myMASK[4] = {MYIPMASK};
  uint8_t myDNS[4] = {MYDNS};
  uint8_t myGW[4] = {MYGW};

//  Ethernet.begin(mac,myIP);
  Ethernet.begin(mac,myIP,myDNS,myGW,myMASK);

  server.begin();

}

void loop() {

  size_t size;

  if (EthernetClient client = server.available())
    {
      while((size = client.available()) > 0)
        {
          uint8_t* msg = (uint8_t*)malloc(size+1);
          memset(msg, 0, size+1);
          size = client.read(msg,size);
          free(msg);
        }
      client.stop();
    }
}

--------------------------------------------------------------------------------

ERROR

Arduino: 1.8.7 (Windows 10), Placa:"ESP32 Dev Module, Disabled, Default, 160MHz (WiFi/BT), QIO, 80MHz, 4MB (32Mb), 921600, None"

TcpServerUIPETHERNET:16:50: error: invalid cast to abstract class type 'UIPServer'

 EthernetServer server = EthernetServer(LISTENPORT);

                                                  ^

In file included from D:\arduino-1.8.7-windows\arduino-1.8.7\portable\sketchbook\libraries\UIPEthernet/UIPEthernet.h:42:0,

                 from D:\arduino-1.8.7-windows\arduino-1.8.7\Portable\sketchbook\TcpServerUIPETHERNET\TcpServerUIPETHERNET.ino:13:

D:\arduino-1.8.7-windows\arduino-1.8.7\portable\sketchbook\libraries\UIPEthernet/UIPServer.h:42:9: note:   because the following virtual functions are pure within 'UIPServer':

   class UIPServer : public Server {

         ^

In file included from D:\arduino-1.8.7-windows\arduino-1.8.7\portable\packages\esp32\hardware\esp32\1.0.1\cores\esp32/Arduino.h:158:0,

                 from sketch\TcpServerUIPETHERNET.ino.cpp:1:

D:\arduino-1.8.7-windows\arduino-1.8.7\portable\packages\esp32\hardware\esp32\1.0.1\cores\esp32/Server.h:28:18: note: 	virtual void Server::begin(uint16_t)

     virtual void begin(uint16_t port=0) =0;

                  ^

TcpServerUIPETHERNET:16:16: error: cannot declare variable 'server' to be of abstract type 'UIPServer'

 EthernetServer server = EthernetServer(LISTENPORT);

                ^

Foram encontradas múltiplas bibliotecas para "UIPEthernet.h"
Usado: D:\arduino-1.8.7-windows\arduino-1.8.7\portable\sketchbook\libraries\UIPEthernet
Não usado: D:\arduino-1.8.7-windows\arduino-1.8.7\portable\sketchbook\libraries\UIPEthernet-2.0.6
exit status 1
invalid cast to abstract class type 'UIPServer'

Este relatório teria mais informações com
"Mostrar a saida detalhada durante a compilação"
opção pode ser ativada em "Arquivo -> Preferências"

jandrassy, 2019-02-09:
and you can compile a native WiFi example or an Ethernet library example without errors? because this looks like a common esp8266 core installation problem

vallesolucoes, 2019-02-09:
> o núcleo esp32 mudou as classes base de rede do Arduino estabelecidas
thank you!
but how can I solve this?

jandrassy, 2019-02-09:
you could modify your copy of the library. make the changes necessary for the esp32 core

jandrassy, 2019-02-09:
wait, I fix it

jandrassy, 2019-02-09:
I fixed it in my fork https://github.com/jandrassy/UIPEthernet
try it

vallesolucoes, 2019-02-18:
Thank you [andrassy] (/ andrassy)
I tried your version but it still gives error
I can not use the server example on the ESP32 platform

vallesolucoes, 2019-02-18:
If anyone can help me solve this, I thank you

vallesolucoes, 2019-02-18:
I followed the instructions of the link below and solved my problem
thank you very much!
https://github.com/jandrassy/UIPEthernet/commit/1590f98ae956a8435dd93008b02f207d4bfd46a3

jandrassy, 2019-02-18:
you first tested the master branch version not the esp32 branch?
is it working or only compiles for now?

vallesolucoes, 2019-02-18:
compiles and works perfect!

jandrassy, 2019-02-18:
fix https://github.com/UIPEthernet/UIPEthernet/commit/1590f98ae956a8435dd93008b02f207d4bfd46a3 merged

closed 2019-02-18 by jandrassy