UIPEthernet/UIPEthernet #61: Local IP always returns as 0.0.0.0 in DHCP but setting IP always returns some default values in ESP8266

baaridunnasr, 2019-02-20:
Board : Node MCU ESP8266
Connections have been made as per /hardware/ folder in this repo.

## DHCP
When I run the following code : 
```
Ethernet.begin(mac);
}

void loop()
{
  Serial.print("localIP: ");
  Serial.println(Ethernet.localIP());
  Serial.print("subnetMask: ");
  Serial.println(Ethernet.subnetMask());
  Serial.print("gatewayIP: ");
  Serial.println(Ethernet.gatewayIP());
  Serial.print("dnsServerIP: ");
  Serial.println(Ethernet.dnsServerIP());
  delay(5000);
}
```

My output is : 
```
localIP: 0.0.0.0
subnetMask: 0.0.0.0
gatewayIP: 0.0.0.0
dnsServerIP: 0.0.0.0
```

##STATIC IP

The code : 
```
IPAddress IP(192, 168, 1, 101);
void setup()
{
  pinMode(15, OUTPUT);
  Serial.begin(9600);
  Serial.println("Start...");

  Ethernet.begin(mac,IP);
}

void loop()
{
  Serial.print("localIP: ");
  Serial.println(Ethernet.localIP());
  Serial.print("subnetMask: ");
  Serial.println(Ethernet.subnetMask());
  Serial.print("gatewayIP: ");
  Serial.println(Ethernet.gatewayIP());
  Serial.print("dnsServerIP: ");
  Serial.println(Ethernet.dnsServerIP());
  delay(5000);
```

My output is : 
```
localIP: 192.168.1.101
subnetMask: 255.255.255.0 //<------- Actual router settings is 255.255.255.128 
gatewayIP: 192.168.1.1
dnsServerIP: 192.168.1.1
```

And after this static IP assignment the device does not appear in the routing table?

How to solve this? and are there some router settings that need to be done. The Router in use is WRT120N.

baaridunnasr, 2019-02-20:
The same issue is observed in ESP32 as well

jandrassy, 2019-02-20:
to static IP: you must set all settings. otherwise they are set to defaults

does your network have DHCP? Ethernet.begin(mac) returns true, if DHCP was sucessful

baaridunnasr, 2019-02-20:
Will check and report in an hour about the static configs.

Network does have DHCP enabled on the router.

[image] is the DHCP config I see on my router.

Thank you.

baaridunnasr, 2019-02-20:
>   Ethernet.begin(mac, ip, gw, gw, sn);

Must the Gateway be declared twice in function call?

baaridunnasr, 2019-02-20:
If say DHCP is enabled on a router, and I try to connect using a static IP config from the ESP32/8266 will there be any clashes or issues?

jandrassy, 2019-02-20:
second IP is DNS server. I edited the snippet.
you can use static IP, only choose a unique IP address outside the DHCP range.

in my experience, if DHCP doesn't work on a DHCP enabled network and the LEDs on the module blink, the problem is communication with the module. check wiring, cs pin selection, other SPI devices on same bus

baaridunnasr, 2019-02-20:
@jandrassy Can confirm with what you said, the DHCP worked. There must've been a improper connection. However trying it a third time, suddenly gave the same error. I will get back to you with within a few hours when I have access to better cabling and a decent breadboard.

closed 2019-02-22 by baaridunnasr

baaridunnasr, 2019-02-26:
Hello, @jandrassy. I have tried all things that you said and completely solidified the circuit. The Ethernet.localIP does return a particular IP Address but it does not show up in any network scanner. Why is that to?

Also, if I try to run some NTP code, the ESP32 crashes and reboots. NTP works fine with Wifi though...

Any thoughts and ideas?

jandrassy, 2019-02-26:
NTP code? some library?

baaridunnasr, 2019-02-26:
```
 configTime(other_comms_offset * 3600, 3600, ntp_server);
 delay(5000);
```

This is code I use to get the NTP Time and set the ESP32...

Also, any ideas why IP Scan doesn't show up any device against the reported IP?

baaridunnasr, 2019-02-26:
```
localIP: 192.168.1.84
subnetMask: 255.255.255.0
gatewayIP: 192.168.1.2
dnsServerIP: 125.22.47.125
``` 

This is what I get from the DHCP assignment...

jandrassy, 2019-02-26:
what is `configTime`? it is a esp32 function which uses the build in WiFi

baaridunnasr, 2019-02-26:
> `configTime` is a esp32 SDK function which uses the build in WiFi
Well, damn! How do you know this?
Anyway to fix this? Any ideas? How should I go about implementing NTP?

jandrassy, 2019-02-26:
use NtpClientLib

baaridunnasr, 2019-02-26:
I'll look onto NTP Client. Any ideas about the IP Scanner though?

jandrassy, 2019-02-26:
maybe first the device must be active on network

baaridunnasr, 2019-02-26:
Didn't quite understand what you meant. Could you please clarify? If not please tell what I can read up to understand all this. 

Also, I tried NTP Client lib and it works perfectly fine with ethernet. Only have to attach the UDP Object from UIPEthernet to the NTP Object.

jandrassy, 2019-02-26:
> 
> 
> Didn't quite understand what you meant. Could you please clarify? If not please tell what I can read up to understand all this.
> 
> Also, I tried NTP Client lib and it works perfectly fine with ethernet. Only have to attach the UDP Object from UIPEthernet to the NTP Object.

and the IP scanner still doesn't show the device?

baaridunnasr, 2019-02-26:
> and the IP scanner still doesn't show the device?

Yes. Tried using different IP Scanners. They show other devices but not mine. I will try connecting to some other Ethernet network and see if I have any issues... I'll let you know if I face the same.