UIPEthernet/UIPEthernet #62: EthernetClient socket loses connection Wireshark loses connection [TCP Spurious Retransmission]

JoaoPagotto, 2019-02-27:
Hello dear developers, after analyzing Wireshark I realized that UIPEthernet 2.0.7 and ENC28J60 is generating TCP Retransmissions, I am using it with an Arduino Nano. I open a connection with a TCP server and send and receive packets in a period of 20 packets per second, but with UIPEthernet happens from time to time also lose connection to the server, I have other hardware running in this same way PIC24FJ256GB106 and ENC28J60 with the Microchip TCPIP Stack v5.42.08 stack, and the Arduino Shield W5100, and these TCP Restransmissions problems do not occur. What could it be? If you can guide me in something so I can try to discover this bug in the library and be able to send it to you, any help is welcome. 

Attached is the Wireshark log, reminding me that my connection connects to the server and must stay on RX / TX constantly.

[http://simcore.com.br/dist/cap01.pcapng](url)

Thank you all.

jandrassy, 2019-02-27:
UIPEthernet library is made for small MCUs. To store the TCP packets before the application reads them, the library uses the memory of the ENC28J60. To have more for stored packets the part of memory reserved for receive is small. You can try give it more space. In file utility/enc28j60.h change the `RXSTOP_INIT' for example to `(0x1FFF-0x1000)`.

JoaoPagotto, 2019-02-27:
Hello dear friend, good afternoon Thanks a lot for the help.

The problem still persists, a series of problematic packets generating
TCP Retransmission and TCP Spurious Retransmission when it definitely
loses the connection.

Do you have any other tips for solving this problem?

I still do a lot of tests to try to solve this, but still without
success, what can help me and guide I will be here to make UIPEthernet
very stable.

Hugs.

jandrassy, 2019-02-27:
sorry,  I edited the comment. you read the first version. soon I realized  that changing the parameter will not help.

JoaoPagotto, 2019-02-27:
I think the problem is in receiving the data in ENC28J60, as you see the Wireshark log, the server generates packet retransmissions, follow image.

ENC28J60 IP : 192.168.2.58
SERVER IP: 192.168.2.253

[image]

jandrassy, 2019-02-27:
but do you read the received data as fast as possible? because the memory of the ENC28J60 is small and if there is no space, packets are dropped. But TCP must try to retransmit them.

JoaoPagotto, 2019-02-27:
Yes, the packets are very small The worst case is 64bytes, and only one socket is open in the Arduino,
the Arduino connects to the server and when connected they are sending packets

> Arduino TX
  <Server RX
  > Server TX
<Ardino RX

and stays in that script constantly.

The program in Arduino is attached.
I'm using Arduino NANO Old Bootloader

[http://simcore.com.br/dist/Custom_v1_00_eth.zip](http://simcore.com.br/dist/Custom_v1_00_eth.zip)

jandrassy, 2019-02-27:
If I see it right it is a 'ping/pong' between the server and Nano. Does the server send a next packHeader only if previous FUNC calls were finished by the Nano? 328p on Nano is small and slow and UIPEthernet library must handle much more then the Ethernet library because the ENC28J60 does less of TCP/IP handling then Wiznet chips.

JoaoPagotto, 2019-02-27:
Exactly the process is a ping pong and where one process waits the other, always in the form read, write, read, write so on constantly. I even put a Delay in the server application of 10 msecs between cycles. But even so the sporadic disconnection occurs as well as retransmissions in the ratio of 100 to 1 packets with problem and requires retransmission.

Do you have any other tips or guidance or are you subject to Nano's limitations? Do you think I should test all this on a Mega Arduino to make sure?

Hugs.

jandrassy, 2019-02-27:
Mega is not faster. It has more memory, but the library has no option to use it instead of the small ENC28J60 memory.
It is strange with this retransmissions, if only so small packets are exchanged. I would like to investigate this problem, but now I don't have time for it.

jandrassy, 2019-02-27:
what ENC28J60 module/shield do you have and how is it wired? could it be powering issue or overheating or resets from relays EMI?

JoaoPagotto, 2019-02-27:
This Arduino Nano + ENC28J60 (Energized via USB Cable, on computer)

[image]
[image]
[image]
[image]

This PIC24FJ256GB106 + ENC28J60

[image]

JoaoPagotto, 2019-02-27:
now that an idea came up .. did it lack the 3v3 and 5v shift level on the spi and cs pins ??

JoaoPagotto, 2019-02-28:
I will now test with the Arduino DUE as they have the 3v3 compatible pins.

JoaoPagotto, 2019-02-28:
I ended up doing several tests using the Arduino DUE with the UIPEthernet library, but the problem still persists, retransmission of lost packets from the server to the Arduino, and sporadic disconnection problems. Any tips on where to go? Configuration Parameters etc?
Hugs.

JoaoPagotto, 2019-03-13:
I do not know if it helps the team of developers, but I make available to Microchip TCP / IP Stack v5.42.08 that the stack I use in my projects with PIC18 and PIC24, is quite tested and does not generate retransmission of packages and disconnections sporadically, esotu committed to power contribute to solving this issue. Hugs.

http://simcore.com.br/dist/Microchip%20TCPIP%20Stack/TCPIPStack%205.42.zip

jandrassy, 2020-08-18:
I fixed a bug. could you please try the Git version to see if it solved your problem?

closed 2020-09-14 by jandrassy

MonirA1990, 2020-12-17:
The problem still exist, what should i do to solve it?