UIPEthernet/UIPEthernet #64: No ACK is send and forcing a retransmission

elsucht, 2019-03-13:
We are using a teensy with Enc28J60 as TCPClient (based on UIPEthernet) to connect to a PC based TCP-Server to exchange small data packets. We randomly (roughly 1 of 25times) detect some missing ACK which schould be sent from the client to the server.
The log file shows that there is no writePacket-call following immediatly to the readPacket-call, so the server does not get an ACK and retransmits the packet. However the client log shows the the original packet and the retransmitted packet are identical. What can cause such a behaviour?
[teensylog_correct.txt](https://github.com/UIPEthernet/UIPEthernet/files/2961533/teensylog_correct.txt)
[teensylog _retransmit.txt](https://github.com/UIPEthernet/UIPEthernet/files/2961534/teensylog._retransmit.txt)
[tcp_retransmission.pcapng.zip](https://github.com/UIPEthernet/UIPEthernet/files/2961543/tcp_retransmission.pcapng.zip)

jandrassy, 2019-03-13:
the IP stack is handled by the MCU. if it is busy, maybe it misses to do something in time. do you call maintain() in lloop?

JoaoPagotto, 2019-03-13:
I had same problem.

https://github.com/UIPEthernet/UIPEthernet/issues/62

elsucht, 2019-03-13:
> the IP stack is handled by the MCU. if it is busy, maybe it misses to do something in time. do you call maintain() in lloop?

Yes. I do.

jandrassy, 2019-03-13:
Teensy model? MCU?

elsucht, 2019-03-13:
Teensy 3.2

elsucht, 2019-03-13:
created a log with increased ACTLOGLEVEL.
Same packet is received twice, but the ACK is written only after second packet (as far as I understand).
Maybe thats helpfull for you?
[teensylog_debuglvl.txt](https://github.com/UIPEthernet/UIPEthernet/files/2962296/teensylog_debuglvl.txt)

jandrassy, 2019-03-14:
UIPEthernet library is made for small MCUs. To store the TCP packets before the application reads them, the library uses the 2kB memory of the ENC28J60. To have more for stored packets the part of memory reserved for receive ring buffer is small.

But the enc28j80 puts in the RX ring buffer almost every packet seen on network and the library must analyse the packet. Some packets are ignored, some are processed for support protocols and the packets for the host are moved to other part of the enc28j60 internal memory to have place for the next packet from the network. The processing of the RX ring buffer is done in maintain() and in every UIPEthernet library function called by host application. 

In a network with large traffic sometimes the RX ring buffer doesn't have place for a packet.

elsucht, 2019-03-15:
Thanks for your support and the explanation. However in our scenario we just send one packet with a payload of max 20byte once in a second to one teensy. So I would not expect problems related to high traffic.
A little more debugging revealed that the tcp-checksum of the not-ACKed packet is bad. Even though the retransmitted packet is identical (excluding some counters) the tcp-checksum is good and is ACKed.
Comparing the caclculation of the checksum of both packets the difference occurs in the function "UIPEthernetClass::upper_layer_chksum" line 565ff
` if (upper_layer_memlen < upper_layer_len)
    {
      sum = Enc28J60Network::chksum(
          sum,
          UIPEthernetClass::uip_packet,
          UIP_IPH_LEN + UIP_LLH_LEN + upper_layer_memlen,
          upper_layer_len - upper_layer_memlen
      );}`
As far as I understand  the "Enc28J60Network::chksum"-function reads the payload from the enc28J60 for the computation of the sum again. But somehow the payload-data on the ENC28J60 changed compared to the last time it was read (and printed to the log). As a result the checksum is bad.
Changing this part of re-read from ENC28J60 for incoming packets to work with the data in uip_buffer seems to solve our issue. However we have to increase the uip_buffer size (1500 bytes) to make sure it can hold the data of larger packets and thus can calculate the checksum correctly.
What is still open for us is if we are able to interpret the payload correctly.

jandrassy, 2019-03-15:
do you use the latest version 2.07?

elsucht, 2019-03-15:
Yes

elsucht, 2019-03-15:
looks like there could be an off by one somewhere when reading from the ENC.
The "first" readPacket() by tick() gives a (correct) payload of : "24 24 24 24 24 24 20 6A 00 6A 00 6A 00".
A "second" explicit UIPClient.read() (or the internal read int checksum function) gives "24 24 24 24 24 20 6A 00 6A 00 6A 00 44". which is not correct.

jandrassy, 2019-03-15:
try the github version please. I just merged a pull request

elsucht, 2019-03-18:
tried it with no luck. Issue persists.

ChangeDan, 2020-08-07:
> Thanks @jandrassy for your support and the explanation. However in our scenario we just send one packet with a payload of max 20byte once in a second to one teensy. So I would not expect problems related to high traffic.
> 
> A little more debugging revealed that the tcp-checksum of the not-ACKed packet is bad. Even though the retransmitted packet is identical (excluding some counters) the tcp-checksum is good and is ACKed.
> 
> Comparing the caclculation of the checksum of both packets the difference occurs in the function "UIPEthernetClass::upper_layer_chksum" line 565ff
> 
> ` if (upper_layer_memlen < upper_layer_len)
> 
>     {
> 
>       sum = Enc28J60Network::chksum(
> 
>           sum,
> 
>           UIPEthernetClass::uip_packet,
> 
>           UIP_IPH_LEN + UIP_LLH_LEN + upper_layer_memlen,
> 
>           upper_layer_len - upper_layer_memlen
> 
>       );}`
> 
> As far as I understand  the "Enc28J60Network::chksum"-function reads the payload from the enc28J60 for the computation of the sum again. But somehow the payload-data on the ENC28J60 changed compared to the last time it was copied to the uip_buffer(and printed to the log). As a result the checksum is bad.
> 
> Changing this part of re-read from ENC28J60 for incoming packets to work with the data in uip_buffer seems to solve our issue. However we have to increase the uip_buffer size (1500 bytes) to make sure it can hold the data of larger packets and thus can calculate the checksum correctly.
> 
> What is still open for us is if we are able to interpret the payload correctly.

Could you please explain to me more in detail how to fix this issue with the workaround you mentioned? I mean where I can find the function I have to change and what exactly to write instead. Increasing the buffer is not necessary in my case (I think) because the packets I send via Telnet only contain strings of 3 characters... or could I be wrong?

Many thanks in advance!

jandrassy, 2020-08-18:
I fixed a bug. could you please try the Git version to see if it solved your problem?

ChangeDan, 2020-08-18:
> I fixed a bug. could you please try the Git version to see if it solved your problem?

Thank you very much! In my case (after a quick check with no external hardware to the nano) it seems to work fine now! I haven't checked it with wireshark, but my application is running well after the bugfix (I've just changed all library data with the ones from git hub, right?). No 3 secs struggles all 10 secs anymore. Now I can start testing my application more advanced 😉

closed 2020-08-25 by jandrassy