UIPEthernet/UIPEthernet #66: ENC28J60 TCP/IP UIPETHERNET

ErencanErkarakas, 2019-03-20:
enc28j60 ile sadece 4 kere bağlantıyı kurabiliyorum sonra bir süre bouynca tekrar bağlanmaya çalışsamda tekrar bağlanmıyor.

jandrassy, 2019-03-20:
do you close the connection?

ErencanErkarakas, 2019-04-01:
Serial.println("OKUDU");
  while (true)
  {
    if (!client.connect(IPAddress(192, 168, 1, 111), 123)) {
      delay(100);
      Serial.println("Baglanmadi");
 }
    else{
      break;
    }

  }
  /*if () {
    Serial.println("Baglanmadi");

    }*/
  Serial.println("Baglanti Kuruldu");

  client.println(rfid);
  Serial.println(rfid);
  getmessage();
  ServerDisconnect();
}

ErencanErkarakas, 2019-04-01:
maximum connection count =4

ErencanErkarakas, 2019-04-01:
@jandrassy

jandrassy, 2019-04-01:
you don't call client.stop()

ErencanErkarakas, 2019-04-01:
void ServerDisconnect() {
  Serial.println("Serverdan Cikis Yapildi");
  client.stop();
}

ErencanErkarakas, 2019-04-01:
@jandrassy

jandrassy, 2019-04-01:
you don't need to '@jandrassy` me. I get a notification on every change in the repository.
the TCP connection is closed if booth sides close the connection (or timeout). does the server close the connection? do you need to open the connection every time? can't you leave the connection open?

In [my project](https://github.com/jandrassy/Regulator) I have 3 servers (telnet, web, sketch upload), modbus client, upnp client, http client, Blynk on 6 connections

ErencanErkarakas, 2019-04-01:
I just want to connect to a single server with this server can only connect 4 times

jandrassy, 2019-04-01:
the connection is not free until the other side closes the socket too. the library doesn't abort the connection.

ErencanErkarakas, 2019-04-01:
I closed the connection by the server and the result is not changed Are you using the sample code, or would you suggest another library?

jandrassy, 2019-04-01:
do you use version 2.0.7?

ErencanErkarakas, 2019-04-01:
latest version.

closed 2020-09-14 by jandrassy