UIPEthernet/UIPEthernet #67: ESP32 support. Wiring?

VanKurt, 2019-03-20:
Does this library work with the ESP32? How should it be wired?

jandrassy, 2019-03-20:
yes it does. wire it to SPI

VanKurt, 2019-03-20:
But how can I control e.g. which pin shall be used for chip select?

jandrassy, 2019-03-20:
Ethernet.init(CS_PIN)

VanKurt, 2019-03-20:
Awesome. Thanks a lot!

closed 2019-03-20 by VanKurt

frederic-bonjour, 2019-03-28:
I have a ESP32-WROOM-32 NodeMCU. Are the transistors needed, as mentioned on the schematics for the ESP8266 NodeMCU?

jandrassy, 2019-03-28:
no. the transistors handle the state of the esp8266 boot pin io 15 to be LOW at boot, because it is the SPI SS pin which should be HIGH.
on esp32 this is not a problem.

frederic-bonjour, 2019-03-28:
Thank you very much, I'll try that.

frederic-bonjour, 2019-03-28:
It tells me:
> Failed to configure Ethernet using DHCP
Ethernet shield was not found.  Sorry, can't run without hardware. :(

frederic-bonjour, 2019-03-28:
My mistake: it works! Thank you so much! :)

frederic-bonjour, 2019-03-28:
And is there a way to make OTA updates working via Ethernet?

jandrassy, 2019-03-28:
copy one of the OTA libraries of the esp32 core and in the copy replace "WiFi" with "Ethernet".
Only the mDns part will not work, so the Arduino IDE will not show a network port. because UDP.beginMulticast() is not implemented in UIPEthernet yet

frederic-bonjour, 2019-03-28:
OK, thanks!
But how do I update the sketch if the Arduino IDE does not allow me to select the network port?

jandrassy, 2019-03-28:
the esp8266 has a HTTPUpdateServer library, which (now I see) is missing in esp32 package. 
the HTTPUpdate library checks a web server for new bin file.
for ArduinoOTA upload from command line is possible

frederic-bonjour, 2019-03-29:
I'm trying to make HTTP updates working. Still have a checksum error, but I keep on searching this way ;) Thank you again for your answers and your time!