UIPEthernet/UIPEthernet #69: UIPEthernetClass::tick() is corrupting outgoing packages

gokcer, 2019-04-05:
First of all, thanks for this great library. It contains your huge efforts and I appreciate you for this. I have detected a strange behavior and tracked the bug down to tick() method. Problem is ethernet module was seldom able to receive dhcp response although everything was normal. I noticed that if `ACTLOGLEVEL` set to `LOG_DEBUG_V2`, program starts to receive ip from dhcp successfully everytime but if `ACTLOGLEVEL` set to `LOG_INFO` or less, it stops working. Same was happening to tcp connections. If log level is high which generates lots of output then everything work flawlessly but without log generation receiving packages starts not working. So I started to suspect from timing issues and started to put`delay(100)` lines for dhcp lease parts of the program. By trial and error, I narrowed the issue to `UIPEthernetClass::tick()` Let me explain more:
`parseDHCPResponse` is called just after `send_DHCP_MESSAGE`. If any delay happens before
```
while (_dhcpUdpSocket.parsePacket() <= 0) {
        if ((millis() - startTime) > DHCP_RESPONSE_TIMEOUT) {
            return 255;
        }
        delay(50);
    }
```
Code works flawlessly. So I stepped into `dhcpUdpSocket.parsePacket()`. In the method there is 
```
int UIPUDP::parsePacket(void) {
#if ACTLOGLEVEL >= LOG_DEBUG_V3
    LogObject.uart_send_strln(F("UIPUDP::parsePacket(void) DEBUG_V3:Function started"));
#endif

    UIPEthernetClass::tick();

    if (appdata.packet_in != NOBLOCK) {
```
So if I put delay(100) just before UIPEthernetClass::tick() everything is ok (a bit slow normally)

```
int UIPUDP::parsePacket(void) {
#if ACTLOGLEVEL >= LOG_DEBUG_V3
    LogObject.uart_send_strln(F("UIPUDP::parsePacket(void) DEBUG_V3:Function started"));
#endif
    delay(100);
    UIPEthernetClass::tick();

    if (appdata.packet_in != NOBLOCK) {
``` 
but if code is like this it never works
```
int UIPUDP::parsePacket(void) {
#if ACTLOGLEVEL >= LOG_DEBUG_V3
    LogObject.uart_send_strln(F("UIPUDP::parsePacket(void) DEBUG_V3:Function started"));
#endif

    UIPEthernetClass::tick();
    delay(100);
    UIPEthernetClass::tick(); // added to verify issue is not related with network delay

    if (appdata.packet_in != NOBLOCK) {
```
Probably before module was able to process outgoing packages, `UIPEthernetClass::tick()` corrupts them. This happens for both UDP and TCP traffic. I am sure if this issue resolved, many strange behaviors will also gone and this library will be much more strong. I can provide any information you need (except log output since everything start to work perfect if logs are enabled :)). My compute board is STM32F103C and there is no other device on the SPI port of the ENC28J60. 

Thanks a lot for.

Gokcer Gokdal

jandrassy, 2019-04-06:
with which version did you test?

gokcer, 2019-04-06:
Sorry. Forgot to mention. 2.0.7.

jandrassy, 2019-04-06:
try the HEAD version from GitHub repository, please

gokcer, 2019-04-06:
I switched to HEAD version. Same behavior exists. When I enable LOG_DEBUG, DHCP and TCP communication everything works, else unable to get DHCP address. Without enabling LOG_DEBUG and putting delay before tick(), card is able to receive DHCP address but then TCP communication fails.

gokcer, 2019-04-06:
Can I do anything to help for you?

jandrassy, 2019-04-06:
tick() is the main function of the library. it reads incoming packets in the enc28j60 RX memory, and lets the uip part of the library process TCP/IP related things. Packets with user data are moved to the "TX and storage" part of the enc28j60 memory to be processed by the application.
why do you think data is corrupted?

gokcer, 2019-04-06:
As you know of course dhcp retriewal is two step process. Client broadcast discovery and receive ip offer. Then send registration request and get acknowledgment. What I tried is inserting small delays like 100ms just after module send  discovery and registration requests then everything worked as expected. If I remove one if the two delays, module was unable to receive dhcp answers. Since tick() is the main method which works correct for receiving data, problem is probably corruption of the outgoing package. I guess that before module send packages to ethernet layer, tick() disturbs this process. It smells like a memory corruption before module process outgoing data. What can be the problem do you think?

gokcer, 2019-04-06:
Forgot to mention, I iterated debugging one step further. I moved the delays from "after sending broadcast message calls" to "receive dhcp response calls", a small delay just before tick() makes dhcp success. So it looks like tick() corrupts package sending process of the module. There are some open issues as I see which also mention tick() method generate incostistent behaviours like incorrect checksum and retransmit packages so I strongly suspect there is a problem at tick() function.

gokcer, 2019-04-07:
Hi jandrassy. It is working without any log or delay flawlessly for 12 hours. I am going to try reproducing the error to capture some traffic for inspection. My question is while bootp progress, DHCP Discover and DHCP Request packages are sent twice one after another in 8 milliseconds. Is it normal? Here is the wireshark log: https://drive.google.com/open?id=1XOYL_SssrLhao41JA5ow7i7VVLoZZ8Np

gokcer, 2019-04-08:
Hi jandrassy. After inspecting with wireshark under different conditions, it looks like problem is at receiving incoming packages, not outgoing packages. Module is sending DHCP DISCOVER two times in a very short time (1 ms) and receive DHCP OFFER but ignore it then after 4 seconds send again two DHCP DISCOVER packages one another. This routine repeats a few times. Then at some point module this time sends DHCP REQUEST again two times in a very short time and receive DHCP ACK. So strangely it seems module is not processing incoming packages correctly. This incorrect behavior increases if I remove delays by decreasing DEBUG level. I have prepared another debug log for an tcp http request which shows similar misbehaving characteristics.

```
Enc28J60Network::receivePacket(void) DEBUG:receivePacket [380-3BC], next: 3C0, stat: C0, Packet count: 3 -> OK
Enc28J60Network::receivePacket(void) DEBUG: rxstat OK. receivePkt.size=60
UIPEthernetClass::tick() DEBUG:receivePacket: 255
UIPEthernetClass::tick() DEBUG:readPacket type ARP, uip_len: 60
UIPEthernetClass::network_send() DEBUG:uip_buf (uip_len): 42, packet: 1
Enc28J60Network::sendPacket(memhandle handle) DEBUG:sendPacket(1) [800-82A]: 0 E8 37 7A 6B 72 42 74 69 69 2D 30 33 8 6 0 1 8 0 6 4 0 2 74 69 69 2D 30 33 C0 A8 1 2F E8 37 7A 6B 72 42 C0 A8 1 1 
UIPEthernetClass::tick() DEBUG:freeing packet: 255
Enc28J60Network::setERXRDPT(void) DEBUG:Set actnextPacketPtr:3BF
Enc28J60Network::receivePacket(void) DEBUG:receivePacket [3C6-402], next: 406, stat: C0, Packet count: 2 -> OK
Enc28J60Network::receivePacket(void) DEBUG: rxstat OK. receivePkt.size=60
UIPEthernetClass::tick() DEBUG:receivePacket: 255
UIPEthernetClass::tick() DEBUG:readPacket type ARP, uip_len: 60
UIPEthernetClass::network_send() DEBUG:uip_buf (uip_len): 42, packet: 1
Enc28J60Network::sendPacket(memhandle handle) DEBUG:sendPacket(1) [800-82A]: 0 E8 37 7A 6B 72 42 74 69 69 2D 30 33 8 6 0 1 8 0 6 4 0 2 74 69 69 2D 30 33 C0 A8 1 2F E8 37 7A 6B 72 42 C0 A8 1 1 
UIPEthernetClass::tick() DEBUG:freeing packet: 255
Enc28J60Network::setERXRDPT(void) DEBUG:Set actnextPacketPtr:405
UIPUDP::beginPacket DEBUG:udp beginPacket, rip: 192.168.1.1, port: 53
uipudp_appcall(void) DEBUG:udp, uip_poll preparing packet to send: 1, size: 89
UIPEthernetClass::upper_layer_chksum(uint8_t proto) DEBUG:uip_buf[34-42]: 3C88
UIPEthernetClass::upper_layer_chksum(uint8_t proto) DEBUG:uip_packet(1)[42-89]: AFBF
UIPUDP::_send() DEBUG:udp, uip_packet to send: 1
UIPEthernetClass::network_send() DEBUG:uip_packet: 1, hdrlen: 42
Enc28J60Network::sendPacket(memhandle handle) DEBUG:sendPacket(1) [800-859]: 0 E8 37 7A 6B 72 42 74 69 69 2D 30 33 8 0 45 0 0 4B 0 D 0 0 40 11 F7 14 C0 A8 1 2F C0 A8 1 1 4 7 0 35 0 37 40 50 A3 4F 1 0 0 1 0 0 0 0 0 0 8 73 75 6E 75 6D 73 72 76 10 62 69 64 75 6E 79 61 74 65 6B 6E 6F 6C 6F 6A 69 3 63 6F 6D 0 0 1 0 1 
Enc28J60Network::receivePacket(void) DEBUG:receivePacket [40C-448], next: 44C, stat: C0, Packet count: 2 -> OK
Enc28J60Network::receivePacket(void) DEBUG: rxstat OK. receivePkt.size=60
UIPEthernetClass::tick() DEBUG:receivePacket: 255
UIPEthernetClass::tick() DEBUG:readPacket type ARP, uip_len: 60
UIPEthernetClass::network_send() DEBUG:uip_buf (uip_len): 42, packet: 1
Enc28J60Network::sendPacket(memhandle handle) DEBUG:sendPacket(1) [800-82A]: 0 E8 37 7A 6B 72 42 74 69 69 2D 30 33 8 6 0 1 8 0 6 4 0 2 74 69 69 2D 30 33 C0 A8 1 2F E8 37 7A 6B 72 42 C0 A8 1 1 
UIPEthernetClass::tick() DEBUG:freeing packet: 255
Enc28J60Network::setERXRDPT(void) DEBUG:Set actnextPacketPtr:44B
Enc28J60Network::receivePacket(void) DEBUG:receivePacket [452-4BB], next: 4C0, stat: C0, Packet count: 1 -> OK
Enc28J60Network::receivePacket(void) DEBUG: rxstat OK. receivePkt.size=105
UIPEthernetClass::tick() DEBUG:receivePacket: 255
UIPEthernetClass::tick() DEBUG:readPacket type IP, uip_len: 105
UIPEthernetClass::upper_layer_chksum(uint8_t proto) DEBUG:uip_buf[34-42]: 60D6
UIPEthernetClass::upper_layer_chksum(uint8_t proto) DEBUG:uip_packet(255)[42-105]: FFFF
uipudp_appcall(void) DEBUG:udp, uip_newdata received packet: 1, size: 63
UIPEthernetClass::tick() DEBUG:freeing packet: 255
Enc28J60Network::setERXRDPT(void) DEBUG:Set actnextPacketPtr:4BF
UIPUDP::parsePacket(void) DEBUG:udp parsePacket received packet: 1, size: 63
MemoryPool::freeBlock(memhandle handle) WARNING: Don't free NOBLOCK handle
MemoryPool::freeBlock(memhandle handle) WARNING: Don't free NOBLOCK handle
MemoryPool::freeBlock(memhandle handle) WARNING: Don't free NOBLOCK handle
UIPEthernetClass::upper_layer_chksum(uint8_t proto) DEBUG:uip_buf[34-58]: 54C9
UIPEthernetClass::network_send() DEBUG:uip_buf (uip_len): 58, packet: 1
Enc28J60Network::sendPacket(memhandle handle) DEBUG:sendPacket(1) [800-83A]: 0 E8 37 7A 6B 72 42 74 69 69 2D 30 33 8 0 45 0 0 2C 0 E 0 0 40 6 FF 35 C0 A8 1 2F B9 5F 0 52 4 5 0 50 0 0 0 38 F6 20 EC F8 60 2 2 0 36 AB 0 0 2 4 2 0 
Enc28J60Network::receivePacket(void) DEBUG:receivePacket [4C6-5CF], next: 5D4, stat: C0, Packet count: 1 -> OK
Enc28J60Network::receivePacket(void) DEBUG: rxstat OK. receivePkt.size=265
UIPEthernetClass::tick() DEBUG:receivePacket: 255
UIPEthernetClass::tick() DEBUG:readPacket type IP, uip_len: 265
UIPEthernetClass::upper_layer_chksum(uint8_t proto) DEBUG:uip_buf[34-42]: 4BC3
UIPEthernetClass::upper_layer_chksum(uint8_t proto) DEBUG:uip_packet(255)[42-265]: FFFF
UIPEthernetClass::tick() DEBUG:freeing packet: 255
Enc28J60Network::setERXRDPT(void) DEBUG:Set actnextPacketPtr:5D3
[... cut ...]
Enc28J60Network::sendPacket(memhandle handle) DEBUG:sendPacket(1) [800-836]: 0 E8 37 7A 6B 72 42 74 69 69 2D 30 33 8 0 45 0 0 28 0 1C 0 0 40 6 FF 2B C0 A8 1 2F B9 5F 0 52 4 7 0 50 0 0 2 8C A8 1D 91 EE 50 14 0 0 F3 58 0 0 
UIPEthernetClass::tick() DEBUG:freeing packet: 255
Enc28J60Network::setERXRDPT(void) DEBUG:Set actnextPacketPtr:401
Got status code: 200
Content length is: 188
```

So why do you thing module is sending DHCP packages two times and why there is a correlation between incostistent incoming package processing and delays between calls (with delay() or enabling log outputs)?

jandrassy, 2019-04-08:
what is
```
Connect failed: -1
etrying communication. Error code: 
Sending request..
/api/Sunum?serial=3e3b95e9-b9e8-4484-8b31-9b96f8e40f37&cardnumber=09FFD656&status=0&rezid=
```
In log I see only one udp packet sent and one udp packet received before that

dkorobkov, 2019-04-17:
Excuse me for interrogation into your dialog. I am now fiddling with some ARP issues around tick() function, too. Don't you think the behaviour described above is caused by not having enough time to receive DHCP response? While the library is printing logs it has time to receive DHCP response (which speed depends on router, cheaper routers are obviously slow - this explains why in other environment the module works good). If you remove logging then tick() runs when DHCP response did not arrive yet.
All above is my suspicion, not yet proved.

jandrassy, 2019-04-18:
> Excuse me for interrogation into your dialog. I am now fiddling with some ARP issues around tick() function, too. Don't you think the behaviour described above is caused by not having enough time to [...]

you could try to change UIP_PERIODIC_TIMER in uipethernet-conf.h

jandrassy, 2020-08-25:
please try the current GitHub version. it has important fixes

closed 2020-09-14 by jandrassy