UIPEthernet/UIPEthernet #71: UIP Interface Layer

ccvelandres, 2019-05-10:
I need some help porting the library to use ENC424J600 instead of ENC28j60. I'm currently updating a project and I need to port this library. Could I just ask for assistance on what to change? 

Also, does this library has a built-in socket implementation?

On a side note, it would probably good to insert an interface layer between the UIPEthernet and the PHY Layer to make it flexible and easier to port to another Ethernet PHY Chip. Say for example in my case, I'm currently updating a project using 2 ENC424J600 Ethernet ports. I can help with this since I also need to port this library to my application.

jandrassy, 2019-05-10:
this library uses UIP framework. https://en.wikipedia.org/wiki/UIP_(micro_IP)

there are many Arduino Ethernet library ports. this is only one of them. 
the original Arduino Ethernet library uses IP stack build-in in the Wiznet chips.
some of the ports use LwIP framework https://en.wikipedia.org/wiki/LwIP

jandrassy, 2019-05-10:
UIPEthernet library is made for small MCUs. To store the TCP packets before the application reads them, the library uses the 2kB memory of the ENC28J60. To have more for stored packets the part of memory reserved for receive ring buffer is small.

But the enc28j80 puts in the RX ring buffer almost every packet seen on network and the library must analyse the packet. Some packets are ignored, some are processed for support protocols and the packets for the host are moved to other part of the enc28j60 internal memory to have place for the next packet from the network. The processing of the RX ring buffer is done in maintain() and in every UIPEthernet library function called by host application.

ccvelandres, 2019-05-22:
I see. Thanks for the info. I'll try to look for other ports of the LwIP framework then.

Closing this issue now.

closed 2019-05-22 by ccvelandres