UIPEthernet/UIPEthernet #74: Cannot use with ST-CORE

nopnop2002, 2019-07-27:
Hello.

I use Blue Pill + ST-Core Ver1.61 + ENC28J60.

Wireing is:
PA4:CS
PA5:SCK
PA6:SO
PA7:SI

I tested this sketch:
But Internet.begin become fail.

[image]

Do you have any help?

	#include <UIPEthernet.h> // For ENC28J60

	EthernetClient client;

	void setup() {
	  byte mac[] = { 0xDE, 0xAD, 0xBE, 0xEF, 0xFE, 0xED }; 
	   
	  Serial.begin(115200);
	  while (!Serial) {
	    ; // wait for serial port to connect. Needed for native USB port only
	  }
	  Serial.println("Start Ethernet.begin");
	  if (!Ethernet.begin(mac) ) {
	    Serial.println("Ethetnet begin fail");
	    while(1) {}
	  }
	  Serial.print("localIP: ");
	  Serial.println(Ethernet.localIP());
	  Serial.print("subnetMask: ");
	  Serial.println(Ethernet.subnetMask());
	  Serial.print("gatewayIP: ");
	  Serial.println(Ethernet.gatewayIP());
	  Serial.print("dnsServerIP: ");
	  Serial.println(Ethernet.dnsServerIP());

	  for (byte thisByte = 0; thisByte < 4; thisByte++) {
	    // print the value of each byte of the IP address:
	    Serial.print(Ethernet.localIP()[thisByte], DEC);
	    if (thisByte != 3) Serial.print("-");
	  }

	}

	void loop() {
	}

jandrassy, 2020-02-07:
use Ethernet.init(pin) to set the CS pin

nopnop2002, 2020-02-07:
I changed wire.

```
ENC28J60    STM32F103
----------------------
GND         GND
3V3         3V3
RESET       PullUp
SCK         PA5(SCK)
SO          PA6(MISO)
SI          PA7(MOSI)
INT         PA4(CS)
CS          PullDown
WCL         N/C
CLKOUT      N/C
```

But Ethetnet begin fail.

jandrassy, 2020-02-07:
I didn't write anything about wiring.
in code use `Ethernet.init(PA4);` to use PA4 as the CS pin. without it the library doesn't know what pin did you wire as CS

nopnop2002, 2020-02-07:
Thank you for your reply.
I changed the sketch:

```
#include <UIPEthernet.h> // For ENC28J60

void setup() {
  byte mac[] = { 0xDE, 0xAD, 0xBE, 0xEF, 0xFE, 0xED }; 
   
  Serial.begin(115200);
  while (!Serial) {
    ; // wait for serial port to connect. Needed for native USB port only
  }
  Serial.println("Start Ethernet.begin");
  Ethernet.init(PA4);
  if (!Ethernet.begin(mac) ) {
    Serial.println("Ethetnet begin fail");
    while(1) {}
  }
  Serial.print("localIP: ");
  Serial.println(Ethernet.localIP());
  Serial.print("subnetMask: ");
  Serial.println(Ethernet.subnetMask());
  Serial.print("gatewayIP: ");
  Serial.println(Ethernet.gatewayIP());
  Serial.print("dnsServerIP: ");
  Serial.println(Ethernet.dnsServerIP());

  for (byte thisByte = 0; thisByte < 4; thisByte++) {
    // print the value of each byte of the IP address:
    Serial.print(Ethernet.localIP()[thisByte], DEC);
    if (thisByte != 3) Serial.print("-");
  }
}

void loop() {
}
```

Conection:

```
ENC28J60    STM32F103
----------------------
GND         GND
3V3         3V3
RESET       PullUp
SCK         PA5(SCK)
SO          PA6(MISO)
SI          PA7(MOSI)
INT         N/C
CS          PA4(CS)
WCL         N/C
CLKOUT      N/C
```

But Ethetnet begin fail.
The Ethernet module works fine with UNO.

jandrassy, 2020-02-08:
try some other pin as CS

jandrassy, 2020-02-16:
I retested it with STM32 core 1.8.0. No problems.

closed 2020-02-16 by jandrassy