UIPEthernet/UIPEthernet #76: Compile issue using Arduino nano every

LeoBu, 2019-08-26:
Hi all, 
I cannot compile any sketch that uses UIPEthernet with ENC28J60 and the Arduino mega avr libraries.
Below the errors I have.
Could you please help me to understand what's wrong?

Thank you!

`:\Users\leona\OneDrive\Documenti\Arduino\libraries\EtherCard-1.1.0\src\enc28j60.cpp: In static member function 'static void ENC28J60::initSPI()':

C:\Users\leona\OneDrive\Documenti\Arduino\libraries\EtherCard-1.1.0\src\enc28j60.cpp:245:5: error: 'SPCR' was not declared in this scope

     SPCR = bit(SPE) | bit(MSTR); // 8 MHz @ 16

     ^~~~

C:\Users\leona\OneDrive\Documenti\Arduino\libraries\EtherCard-1.1.0\src\enc28j60.cpp:245:5: note: suggested alternative: 'SCL'

     SPCR = bit(SPE) | bit(MSTR); // 8 MHz @ 16

     ^~~~

     SCL

In file included from C:\Users\leona\AppData\Local\Arduino15\packages\arduino\hardware\megaavr\1.8.1\cores\arduino/api/ArduinoAPI.h:52:0,

                 from C:\Users\leona\AppData\Local\Arduino15\packages\arduino\hardware\megaavr\1.8.1\cores\arduino/Arduino.h:23,

                 from C:\Users\leona\OneDrive\Documenti\Arduino\libraries\EtherCard-1.1.0\src\enc28j60.cpp:12:

C:\Users\leona\OneDrive\Documenti\Arduino\libraries\EtherCard-1.1.0\src\enc28j60.cpp:245:16: error: 'SPE' was not declared in this scope

     SPCR = bit(SPE) | bit(MSTR); // 8 MHz @ 16

                ^

C:\Users\leona\AppData\Local\Arduino15\packages\arduino\hardware\megaavr\1.8.1\cores\arduino/api/Common.h:82:25: note: in definition of macro 'bit'

 #define bit(b) (1UL << (b))

                         ^

C:\Users\leona\OneDrive\Documenti\Arduino\libraries\EtherCard-1.1.0\src\enc28j60.cpp:245:16: note: suggested alternative: 'PE'

     SPCR = bit(SPE) | bit(MSTR); // 8 MHz @ 16

                ^

C:\Users\leona\AppData\Local\Arduino15\packages\arduino\hardware\megaavr\1.8.1\cores\arduino/api/Common.h:82:25: note: in definition of macro 'bit'

 #define bit(b) (1UL << (b))

                         ^

C:\Users\leona\OneDrive\Documenti\Arduino\libraries\EtherCard-1.1.0\src\enc28j60.cpp:245:27: error: 'MSTR' was not declared in this scope

     SPCR = bit(SPE) | bit(MSTR); // 8 MHz @ 16

                           ^

C:\Users\leona\AppData\Local\Arduino15\packages\arduino\hardware\megaavr\1.8.1\cores\arduino/api/Common.h:82:25: note: in definition of macro 'bit'

 #define bit(b) (1UL << (b))

                         ^

C:\Users\leona\OneDrive\Documenti\Arduino\libraries\EtherCard-1.1.0\src\enc28j60.cpp:245:27: note: suggested alternative: 'PSTR'

     SPCR = bit(SPE) | bit(MSTR); // 8 MHz @ 16

                           ^
[... 355 lines cut ...]
     if (bitRead(SPCR, SPE) == 0)

                       ^

C:\Users\leona\AppData\Local\Arduino15\packages\arduino\hardware\megaavr\1.8.1\cores\arduino/api/Common.h:76:43: note: in definition of macro 'bitRead'

 #define bitRead(value, bit) (((value) >> (bit)) & 0x01)

                                           ^~~

C:\Users\leona\OneDrive\Documenti\Arduino\libraries\EtherCard-1.1.0\src\bufferfiller.cpp: In member function 'void BufferFiller::emit_p(const char*, ...)':

C:\Users\leona\OneDrive\Documenti\Arduino\libraries\EtherCard-1.1.0\src\bufferfiller.cpp:61:25: error: 'eeprom_read_byte' was not declared in this scope

             while ((d = eeprom_read_byte(s++)) != 0)

                         ^~~~~~~~~~~~~~~~

C:\Users\leona\OneDrive\Documenti\Arduino\libraries\EtherCard-1.1.0\src\bufferfiller.cpp:61:25: note: suggested alternative: 'pgm_read_byte'

             while ((d = eeprom_read_byte(s++)) != 0)

                         ^~~~~~~~~~~~~~~~

                         pgm_read_byte`

jandrassy, 2019-08-26:
you use a different library, not UIPEthernet

LeoBu, 2019-08-26:
I'm sorry, you are right. I posted the wrong output. I was trying with a different library too. 
Below the output from UIPEthernet sketch example:

`Arduino: 1.8.10 Hourly Build 2019/08/22 10:33 (Windows 10), Board: "Arduino Nano Every, ATMEGA328"

C:\Users\leona\OneDrive\Desktop\arduino-nightly\arduino-builder -dump-prefs -logger=machine -hardware C:\Users\leona\OneDrive\Desktop\arduino-nightly\hardware -hardware C:\Users\leona\AppData\Local\Arduino15\packages -tools C:\Users\leona\OneDrive\Desktop\arduino-nightly\tools-builder -tools C:\Users\leona\OneDrive\Desktop\arduino-nightly\hardware\tools\avr -tools C:\Users\leona\AppData\Local\Arduino15\packages -built-in-libraries C:\Users\leona\OneDrive\Desktop\arduino-nightly\libraries -libraries C:\Users\leona\OneDrive\Documenti\Arduino\libraries -fqbn=arduino:megaavr:nona4809:mode=on -ide-version=10810 -build-path C:\Users\leona\AppData\Local\Temp\arduino_build_455278 -warnings=none -build-cache C:\Users\leona\AppData\Local\Temp\arduino_cache_782515 -prefs=build.warn_data_percentage=75 -prefs=runtime.tools.avr-gcc.path=C:\Users\leona\AppData\Local\Arduino15\packages\arduino\tools\avr-gcc\7.3.0-atmel3.6.1-arduino5 -prefs=runtime.tools.avr-gcc-7.3.0-atmel3.6.1-arduino5.path=C:\Users\leona\AppData\Local\Arduino15\packages\arduino\tools\avr-gcc\7.3.0-atmel3.6.1-arduino5 -prefs=runtime.tools.arduinoOTA.path=C:\Users\leona\AppData\Local\Arduino15\packages\arduino\tools\arduinoOTA\1.3.0 -prefs=runtime.tools.arduinoOTA-1.3.0.path=C:\Users\leona\AppData\Local\Arduino15\packages\arduino\tools\arduinoOTA\1.3.0 -prefs=runtime.tools.avrdude.path=C:\Users\leona\AppData\Local\Arduino15\packages\arduino\tools\avrdude\6.3.0-arduino16 -prefs=runtime.tools.avrdude-6.3.0-arduino16.path=C:\Users\leona\AppData\Local\Arduino15\packages\arduino\tools\avrdude\6.3.0-arduino16 -verbose C:\Users\leona\OneDrive\Documenti\Arduino\libraries\UIPEthernet\examples\TcpServer\TcpServer.ino
C:\Users\leona\OneDrive\Desktop\arduino-nightly\arduino-builder -compile -logger=machine -hardware C:\Users\leona\OneDrive\Desktop\arduino-nightly\hardware -hardware C:\Users\leona\AppData\Local\Arduino15\packages -tools C:\Users\leona\OneDrive\Desktop\arduino-nightly\tools-builder -tools C:\Users\leona\OneDrive\Desktop\arduino-nightly\hardware\tools\avr -tools C:\Users\leona\AppData\Local\Arduino15\packages -built-in-libraries C:\Users\leona\OneDrive\Desktop\arduino-nightly\libraries -libraries C:\Users\leona\OneDrive\Documenti\Arduino\libraries -fqbn=arduino:megaavr:nona4809:mode=on -ide-version=10810 -build-path C:\Users\leona\AppData\Local\Temp\arduino_build_455278 -warnings=none -build-cache C:\Users\leona\AppData\Local\Temp\arduino_cache_782515 -prefs=build.warn_data_percentage=75 -prefs=runtime.tools.avr-gcc.path=C:\Users\leona\AppData\Local\Arduino15\packages\arduino\tools\avr-gcc\7.3.0-atmel3.6.1-arduino5 -prefs=runtime.tools.avr-gcc-7.3.0-atmel3.6.1-arduino5.path=C:\Users\leona\AppData\Local\Arduino15\packages\arduino\tools\avr-gcc\7.3.0-atmel3.6.1-arduino5 -prefs=runtime.tools.arduinoOTA.path=C:\Users\leona\AppData\Local\Arduino15\packages\arduino\tools\arduinoOTA\1.3.0 -prefs=runtime.tools.arduinoOTA-1.3.0.path=C:\Users\leona\AppData\Local\Arduino15\packages\arduino\tools\arduinoOTA\1.3.0 -prefs=runtime.tools.avrdude.path=C:\Users\leona\AppData\Local\Arduino15\packages\arduino\tools\avrdude\6.3.0-arduino16 -prefs=runtime.tools.avrdude-6.3.0-arduino16.path=C:\Users\leona\AppData\Local\Arduino15\packages\arduino\tools\avrdude\6.3.0-arduino16 -verbose C:\Users\leona\OneDrive\Documenti\Arduino\libraries\UIPEthernet\examples\TcpServer\TcpServer.ino
Using board 'nona4809' from platform in folder: C:\Users\leona\AppData\Local\Arduino15\packages\arduino\hardware\megaavr\1.8.4
Using core 'arduino' from platform in folder: C:\Users\leona\AppData\Local\Arduino15\packages\arduino\hardware\megaavr\1.8.4
Detecting libraries used...
"C:\\Users\\leona\\AppData\\Local\\Arduino15\\packages\\arduino\\tools\\avr-gcc\\7.3.0-atmel3.6.1-arduino5/bin/avr-g++" -c -g -Os -w -std=gnu++11 -fpermissive -fno-exceptions -ffunction-sections -fdata-sections -fno-threadsafe-statics -Wno-error=narrowing -flto -w -x c++ -E -CC -mmcu=atmega4809 -DF_CPU=16000000L -DARDUINO=10810 -DARDUINO_AVR_NANO_EVERY -DARDUINO_ARCH_MEGAAVR -DAVR_NANO_4809_328MODE -DMILLIS_USE_TIMERB3 -DNO_EXTERNAL_I2C_PULLUP "-IC:\\Users\\leona\\AppData\\Local\\Arduino15\\packages\\arduino\\hardware\\megaavr\\1.8.4\\cores\\arduino/api/deprecated" "-IC:\\Users\\leona\\AppData\\Local\\Arduino15\\packages\\arduino\\hardware\\megaavr\\1.8.4\\cores\\arduino" "-IC:\\Users\\leona\\AppData\\Local\\Arduino15\\packages\\arduino\\hardware\\megaavr\\1.8.4\\variants\\nona4809" "C:\\Users\\leona\\AppData\\Local\\Temp\\arduino_build_455278\\sketch\\TcpServer.ino.cpp" -o nul
"C:\\Users\\leona\\AppData\\Local\\Arduino15\\packages\\arduino\\tools\\avr-gcc\\7.3.0-atmel3.6.1-arduino5/bin/avr-g++" -c -g -Os -w -std=gnu++11 -fpermissive -fno-exceptions -ffunction-sections -fdata-sections -fno-threadsafe-statics -Wno-error=narrowing -flto -w -x c++ -E -CC -mmcu=atmega4809 -DF_CPU=16000000L -DARDUINO=10810 -DARDUINO_AVR_NANO_EVERY -DARDUINO_ARCH_MEGAAVR -DAVR_NANO_4809_328MODE -DMILLIS_USE_TIMERB3 -DNO_EXTERNAL_I2C_PULLUP "-IC:\\Users\\leona\\AppData\\Local\\Arduino15\\packages\\arduino\\hardware\\megaavr\\1.8.4\\cores\\arduino/api/deprecated" "-IC:\\Users\\leona\\AppData\\Local\\Arduino15\\packages\\arduino\\hardware\\megaavr\\1.8.4\\cores\\arduino" "-IC:\\Users\\leona\\AppData [... cut ...]
C:\Users\leona\OneDrive\Documenti\Arduino\libraries\UIPEthernet\utility\Enc28J60Network.cpp:1110:5: note: suggested alternative: 'SDA'

     SPDR = ENC28J60_READ_BUF_MEM;

     ^~~~

     SDA

C:\Users\leona\OneDrive\Documenti\Arduino\libraries\UIPEthernet\utility\Enc28J60Network.cpp:1111:5: error: 'waitspi' was not declared in this scope

     waitspi();

     ^~~~~~~

Using library UIPEthernet at version 2.0.7 in folder: C:\Users\leona\OneDrive\Documenti\Arduino\libraries\UIPEthernet 
exit status 1
Error compiling for board Arduino Nano Every.`

jandrassy, 2019-08-26:
in Enc28J60Network.h add  `|| defined(ARDUINO_ARCH_MEGAAVR)` to line 207

LeoBu, 2019-08-26:
Thank you very much!
Now I'm able to compile correctly the code. I still have an issue to make the shield working with the Arduino Nano Every but I think it is something on hardware side. 
Thank you again!

closed 2019-08-26 by LeoBu

jandrassy, 2019-08-31:
default SS pin is 8 for Every