UIPEthernet/UIPEthernet #77: Example not compiling

VanKurt, 2019-09-16:
I'm using Arduino IDE 1.8.9 and the example code provided with the library (UDPClient Example). When compiling (for ESP32 target) I get plenty of errors like:

> In file included from C:\Users\visbbr\Documents\Arduino\libraries\UIPEthernet-2.0.7/UIPEthernet.h:42:0,
> 
>                  from C:\Users\rbbtooli\Desktop\ArduinoECU\MyProject\MyProject.ino:27:
> 
> C:\Users\visbbr\Documents\Arduino\libraries\UIPEthernet-2.0.7/UIPServer.h:49:3: error: invalid abstract return type 'UIPClient'
> 
>    UIPClient available();

Shouldn't this code work out of the box?

jandrassy, 2019-09-16:
and do you use the latest esp32 boards package 1.0.3?

VanKurt, 2019-09-16:
I just realized my Arduino IDE is broken. It doesn't seem to load any more updates...
So I switched to PlatformIO. Here I get exactly the same error using the latest Espressif 32 platform (v1.9.0, released ~3 month ago).

If I understood you correctly, PlatformIO should offer an update to the Espressiv 32 platform soon, containing a fix for the problem? That would be great :-)

jandrassy, 2019-09-16:
1.9.0?
https://github.com/espressif/arduino-esp32/releases

VanKurt, 2019-09-16:
I'm talking about this thing: https://github.com/platformio/platform-espressif32/releases
(it's what PlatformIO uses)

Seems to be based on ESP-IDF framework 3.2.2, not the latest 3.2.3.

jandrassy, 2019-09-16:
it is not about IDF. the error is in esp32 Arduino core

VanKurt, 2019-09-17:
I guess that the ESP32 Platform for PlatformIO is based on the ESP32 Arduino Core. So the fix should arrive in PlatformIO sooner or later...

closed 2019-09-17 by jandrassy