UIPEthernet/UIPEthernet #78: First few bytes of packet gets overwritten with 2 TCP ports listening

ppescher, 2019-09-23:
I have two listening TCP ports:
```
EthernetServer server1 = EthernetServer(1234);
EthernetServer server2 = EthernetServer(1235);
```

When some packet is retransmitted from my PC to the device, the initial few bytes of the packets are corrupted. You can see it from the packet index which is printed inside the payload:
```
DATA from Client: 'Hello 1235 - Pkt 2 + Rnd PLWZKPCTCROMSDMBKHOYGJWAAHFDDRQM' (57)<\r><\n>
Reply len = 54<\r><\n>
DATA from Client: 'Hello 1234 - Pkt 3 + Rnd USEWIGAJBEPVAPJMEJYFQASMMQQFOXSF' (57)<\r><\n>
Reply len = 54<\r><\n>
DATA from Client: 'Hello 1235 - Pkt 4 + Rnd GPHMDKWFQXATIETCKLVVDDUHZRVEIHAH' (57)<\r><\n>
Reply len = 54<\r><\n>
DATA from Client: 'Hello 1234 - Pkt 5 + Rnd IIOIUMTBXRUCKFYATJVUHPVHNVJULTBS' (57)<\r><\n>
Reply len = 54<\r><\n>
DATA from Client: 'Hello 1235 - Pkt 6 + Rnd OMCDFBNUYEPNQFLHQRNPPQWJCIASZCJT' (57)<\r><\n>
Reply len = 54<\r><\n>
DATA from Client: 'Hello 1234 - Pkt 1 + Rnd ZDCFKUAPKYXYFUHGMNXSTISYALYULDAL' (57)<\r><\n>
Reply len = 54<\r><\n>
DATA from Client: 'Hello 1235 - Pkt 8 + Rnd PGOQDPJVCKDXQMSULJQUWOELLHSRLDBP' (57)<\r><\n>
Reply len = 54<\r><\n>
DATA from Client: '<\0>@<\0><\0>o 1234 - Pkt 7 + Rnd EDWOZKVCTVVBAHQFMJTQYCXZZNEZFJMV' (57)<\r><\n>
Reply len = 0<\r><\n>
DATA from Client: 'Hello 1234 - Pkt 9 + Rnd SZQOMCPLFJQVSFNFTPMLRUXZXXZAOMMT' (57)<\r><\n>
Reply len = 54<\r><\n>
DATA from Client: 'Hello 1235 - Pkt 10 + Rnd DXACLCWKKHIRWGNTHHZQKCVJRZHSOMDW' (58)<\r><\n>
Reply len = 55<\r><\n>
[...]
DATA from Client: 'Hello 1235 - Pkt 166 + Rnd NHIVXPQUEXKRESPNYRQNRKMAIXPEVAAD' (59)<\r><\n>
Reply len = 56<\r><\n>
DATA from Client: '<\0><\0>llo 1234 - Pkt 165 + Rnd ORKSVKEOSATPHRJEZRPSLXLJGGOXUORH' (59)<\r><\n>
Reply len = 0<\r><\n>
DATA from Client: 'Hello 1235 - Pkt 172 + Rnd DJHXCZNIBLGXPSRNGGYYKDCAKAIGHCMO' (59)<\r><\n>
Reply len = 56<\r><\n>
DATA from Client: 'Hello 1234 - Pkt 171 + Rnd DMUVTEWBFEMPEKEVHWAYGHOEWWGBXLMX' (59)<\r><\n>
Reply len = 56<\r><\n>
DATA from Client: 'Hello 1235 - Pkt 168 + Rnd NFEVTKZRLLJGMHGHNXJWPSOCABRYODLL' (59)<\r><\n>
Reply len = 56<\r><\n>
DATA from Client: '<\0><\0>llo 1234 - Pkt 167 + Rnd YTWKHIMSRMFPMVDPJAZDEDHFMGRYTASH' (59)<\r><\n>
Reply len = 0<\r><\n>
DATA from Client: 'Hello 1235 - Pkt 174 + Rnd ZHVQMBXLCMAXZVDBHGGKEZTFGKAJWYMJ' (59)<\r><\n>
Reply len = 56<\r><\n>
DATA from Client: 'Hello 1234 - Pkt 173 + Rnd FTDQMMLKRBZSBPDAPINZSIWNUKFEAZHR' (59)<\r><\n>
Reply len = 56<\r><\n>
DATA from Client: 'Hello 1234 - Pkt 169 + Rnd OWCYKTHPKBKMJERLIMFICDVRMPPWCMQT' (59)<\r><\n>
Reply len = 56<\r><\n>
DATA from Client: 'Hello 1235 - Pkt 176 + Rnd YNXCVDSSYECQVXNUAKSLZHFUUIHYUYZT' (59)<\r><\n>
Reply len = 56<\r><\n>
DATA from Client: '<\0><\0>llo 1234 - Pkt 175 + Rnd YCMPWPPOALFOAMKYKSCKZUZDVKPLTACU' (59)<\r><\n>
Reply len = 0<\r><\n>
DATA from Client: 'Hello 1235 - Pkt 170 + Rnd QEXNLPCOZRCCTOLPRZJGDQDJPRZYLERN' (59)<\r><\n>
Reply len = 56<\r><\n>
DATA from Client: 'Hello 1234 - Pkt 177 + Rnd EWZPIRUWYCJTGSVRYWJMGGSVYOJRNXGB' (59)<\r><\n>
Reply len = 56<\r><\n>
DATA from Client: 'Hello 1235 - Pkt 178 + Rnd QFIQJAYNBKBFGWFPAUTJWQDRURNJDAOD' (59)<\r><\n>
Reply len = 56<\r><\n>
```

jandrassy, 2019-09-23:
I suspected this problem and had a fix for it. But the fix was complicated and I was thinking it can't happen that the data are used again after the chip rewrote the first few bytes. I didn't think about retransmission.

The fix is complicated, because it requires to reserve more memory and write the packet not from first byte of it and the existing code is not accommodated for this change.

ppescher, 2019-09-23:
Why sometimes 2 bytes and sometimes 4 bytes get overwritten?

Could a simple fix be to reserve these extra bytes in "alloc" function?

jandrassy, 2019-09-23:
this is in the working memory of the ENC28j60 managed by mempool.cpp

ppescher, 2019-09-23:
How many extra bytes to reserve: 2 or 4?

jandrassy, 2019-09-23:
I mixed two problems in my memory. I was fixing the library for my project in January.

The chip requires one byte before the packet. The library handles it in Enc28J60Network.cpp in function sendPacket by decrementing the start position of the memory reserved for the packet, storing the value and restoring it after the transmission.  

After the TX packet the chip writes 7 bytes status vector.  The space is not reserved by library. Usually there are no important data, but maybe it could be the problem. I saw the overwrite of RX packets stored in the 'transmit + store' part of the memory. But overwritten was only the already processed header of the RX packet. The RX packets are stored here until the application reads the data.

jandrassy, 2020-08-18:
I fixed a bug. could you please try the Git version to see if it solved your problem?

ppescher, 2020-08-19:
I had a look at your last commit and it seems your calculation was off by 1 byte sometimes. My problem involved more bytes and always at the beginning of the payload. It doesn't seem related, but you never know...

Unfortunately it's not easy for me to try it out as the project I was working on has ended almost one year ago.

jandrassy, 2020-08-25:
try the GitHub version please. there is a fix for this

jandrassy, 2020-08-25:
I think a fixed this one too.

closed 2020-08-25 by jandrassy