UIPEthernet/UIPEthernet #80: Esp8266 + Enc28j60 (wifi + ethernet)

SurionAndrew, 2019-11-27:
Good Day,

Can you advise if it is possible to use the Esp8266 in its wifi mode with the Enc28j60 with ethernet:
Eg
Send a char to ESP via wifi over udp 192.168.1.1 port 1337
Send a char to ESP via ethernet udp 192.168.0.1 port 2337?

Or will the networking libraries conflict ( I am trying both at the same time but only wifi seems to work, i cannot ping the ENC28J60 end point)
when i just do ESP+ENC without using the esp wifi i can ping and udp send via ENC endpoint
Regards Andrew

jandrassy, 2019-11-27:
it should work. do you use this library with ESP8266WiFi of esp8266 Arduino?
(there is an option to use the enc28j60 with the LWIP of the esp8266 Arduino)

SurionAndrew, 2019-11-27:
I use ESP8266WiFi ,

I literally posted my message and got it working 5min later :)
So yes it does work, great job

closed 2019-11-27 by SurionAndrew

antonmayko, 2022-05-12:
> Good Day,
>
> Can you advise if it is possible to use the Esp8266 in its wifi mode with the Enc28j60 with ethernet: Eg Send a char to ESP via wifi over udp 192.168.1.1 port 1337 Send a char to ESP via [...]

Hello Andrew.
Help me please. How You make ping with ESP+ENC via ENC? Which library do you use? Post some example, please.