UIPEthernet/UIPEthernet #82: Poor + inconsistent performance on TCP server

globalcitizen, 2020-01-11:
## Environment

 * STM32F103C8T6 blue pill style board
 * OSX build environment

## Code

```
#define MACADDRESS 0x00,0x01,0x02,0x03,0x04,0x05
#define MYIPADDR 192,168,1,6
#define MYIPMASK 255,255,255,0
#define MYDNS 192,168,1,1
#define MYGW 192,168,1,1
#define LISTENPORT 80

#include <UIPEthernet.h>

EthernetServer server = EthernetServer(LISTENPORT);

void setup() {
  Serial.begin(57600);
  uint8_t mac[6] = {MACADDRESS};
  uint8_t myIP[4] = {MYIPADDR};
  uint8_t myMASK[4] = {MYIPMASK};
  uint8_t myDNS[4] = {MYDNS};
  uint8_t myGW[4] = {MYGW};
  Ethernet.begin(mac,myIP,myDNS,myGW,myMASK);
  server.begin();
}

void loop() {
        // get client
        EthernetClient client = server.available();
        if(client.connected()) {
              // do stuff (snipped but basically parses HTTP and responds with no other processing)
              client.stop();
        }
}
```

## Expected behavior
 * Rapid and reliable HTTP responses (certainly <5ms) with no spiking / latency buildup

## Observed behavior
 * Very poor performance
    * High deltas
    * Occasional >1 second spikes.

Here is a screenshot of httping output. Command line was `httping -K -L 2048 -G http://192.168.1.6/asd -i 0.1 -s -Z`.

[image]

## Debugging attempted

After analyzing traffic it seems there are some fundamental issues with the TCP open or close behavior, as for very simple output with almost no processing and rapid close three TCP related issues are observed:
- Unexpected TCP keep-alives (all HTTP connections are immediately closed after a request is received and data is returned)
- TCP 'out of order'
- TCP duplicate ACK

[image]

When I removed the `// do stuff` section and just had immediate close, high latency per request (~40ms) and spiking (>~1s) remain, but the three TCP related issues disappear.

## Working theories
 * Library issue
 * Power related (underclocked or ~brownout mode MCU?) as I lack a bench supply in current environment. I will test with an alternate power environment when eventually possible and post an update.

jandrassy, 2020-01-12:
try to use buffering in `// do stuff`, for example with BufferedPrint from my [StreamLib](https://github.com/jandrassy/StreamLib/blob/master/README.md).

globalcitizen, 2020-01-12:
OK, I can do that but and it may assist with the TCP issues, but with nothing in there at all 10x expected latency and occasionally >1 second latency spikes it seems power or something else is at play. Any chance you could clarify what level of HTTP latency is normal on this MCU class with this library and chipset?

jandrassy, 2020-01-12:
UIPEthernet library is made for small MCUs. To store the TCP packets before the application reads them, the library uses the 2kB memory of the ENC28J60. To have more for stored packets the part of memory reserved for receive ring buffer is small.

But the enc28j80 puts in the RX ring buffer almost every packet seen on network and the library must analyse the packet. Some packets are ignored, some are processed for support protocols and the packets for the host are moved to other part of the enc28j60 internal memory to have place for the next packet from the network. The processing of the RX ring buffer is done in maintain() and in every UIPEthernet library function called by host application.

In a network with large traffic sometimes the RX ring buffer doesn't have place for a packet.

globalcitizen, 2020-01-15:
Thanks. I was really hoping for a millisecond figure representing typical IP level latency, eg. ping or HTTP response time on same subnet. My assumption is single digit ms should be possible, based on previous experience with less powerful MCUs (ATmega328P + ENC28J60).

jandrassy, 2020-02-16:
I tested my SD card content browser web server sketch with this library on the Blue Pill. it works great.

closed 2020-02-16 by jandrassy

jandrassy, 2020-08-18:
I fixed a bug. could you please try the Git version to see if it solved your problem?

globalcitizen, 2020-12-22:
Sorry I have changed countries and no longer have the test hardware available that I was using. However, I am purchasing some identical hardware and will try to make a similar evaluation. I know as an open source maintainer it is frustrating when people ghost you. Happy holidays...

globalcitizen, 2021-01-04:
Had a look at notes but found its a different module to what I now have, so not sure I can replicate the setup to re-test. Migrating to STM32 so not much motivation to dwell further on this. Thanks for your work.