UIPEthernet/UIPEthernet #83: Fast way to know if module connected and connect / monitor in a non-blocking way

esp13, 2020-01-27:
Hi,

I am adding an ENC28J60 to ESP8266 and ESP32 board, I have some basic questions to use properly this library:

**1. Check if module is there and responding**

Actually i connect like this:
if(Ethernet.begin(mac) == 0)
    {
    // Dosn't work, go WiFi
    ConnectWiFi();
    }
  else
    {
    // Works, continue to MQTT
    ClientMQTT.setClient(ClientEthernet);
    ClientMQTT.setServer("192.168.xxx.xxx",xxxx);
    ConnectMQTT();
    }

But when the module isn't connected, it will take time (about 30 seconds) to fail.

=> It is maybe more a SPI question, but how to check if the Ethernet module is connected and responding before trying to connect?

**2. Non blocking connection / monitoring**

I recently change the way I connect to WiFi, to do it without waiting/blocking in the program, I want do the same with Ethernet.
For the WiFi I have a function that is called on each WiFi event:
```

wifieventfct(WiFiEvent_t event, WiFiEventInfo_t info)
    switch(event)
          {
          case SYSTEM_EVENT_STA_DISCONNECTED:
                ...
                break;
          case SYSTEM_EVENT_STA_GOT_IP:
                ...
                break;
          ...
          }
```

=> Is there a way to do the same with Ethernet client ?

Sorry for my terrible English level, if I shouldn’t post this in "issues" tell me, I will remove.
Thank you.

jandrassy, 2020-01-27:
Ethernet.begin(mac) uses DHCP to get the IP address and co.. There is not an asynchronous way to wait for it in this library. You could use a static IP with Ethernet.begin(mac, ip) if the default gateway and dns addresses are OK  or Ethernet.begin(mac, ip, dns, gateway, subnet) if you want to specify all static parameters.

btw: there is a work in progress in the Arduino esp8266 repository to integrate W5500 and enc28j60 as additional networking interfaces in the LWIP TCP/IP stack used for WiFi STA and SoftAP interfaces. https://github.com/esp8266/Arduino/pull/6680

esp13, 2020-01-27:
Thank you for your answer,

1. For the first point I will edit, by "module connected" I meant "module wired/responding". I want to check if the ENC28J60 is present, wired, powered and working.
I tried your example: Ethernet.begin(mac, ip), the problem is that in this case, there is no way to check if it worked. I tried if (Ethernet.connected()) but it always return true
=> Is there a simple way to check if the module is present, wired, powered and working? It is maybe more a SPI question?

2. For the second point, it seems to be a good news but I hope it will soon be integrated (no news about 1 month ago?).  Hope it will be added for the esp32 too.
=>Until it is integrated, How do you do to check if the Ethernet link is working ?

jandrassy, 2020-01-27:
Ethernet.linkStatus() can return 
enum EthernetLinkStatus {
  Unknown,
  LinkON,
  LinkOFF
};
it returns Unknown if the communication with the module failed

closed 2020-02-16 by jandrassy