UIPEthernet/UIPEthernet #84: For a streaming tcp server how to implement server.accept()

vk3-hca, 2020-01-27:
RE: Using UIPEthernet v2.0.7 on a arduino nano with ENC28J60
I am modifying the Advanced Chat Server example to work as a streaming network 2 serial server to send the output stream as a serial port over the network from a tcp port.  

This is code is straight out of the  Advanced Chat Server example which works fine but it requires the client to send some data but not all clients send data they just open the port and accept the incoming stream. eg Realterm. Normally I use USR-VCOM Virtual Port Server which is free in tcp client mode.  I also have used their serial/tcp modules in a few places. It happens to send a 8 bytes on connection so "EthernetClient client = server.available(); " works with it.

The normal Arduino library allows server.accept(); for this purpose which means UIPE is not really a plug in replacement. Possibly this another limitation of the ENC28J60 chip or its on the to do list?
https://www.arduino.cc/en/Reference/EthernetServerAccept

So this small modification to the example code I would like to try fails as below.

  // wait for a new client:
 // EthernetClient client = server.available();
EthernetClient client = server.accept();
  if (client)
    {
    //check whether this client refers to the same socket as one of the existing instances:

Error on compile-------

 Used: C:\Users\h\Documents\Arduino\libraries\UIPEthernet
Using library UIPEthernet at version 2.0.7 in folder: C:\Users\hca_2\Documents\Arduino\libraries\UIPEthernet 
exit status 1
'class UIPServer' has no member named 'accept'

So is there a way around this?

jandrassy, 2020-01-27:
accept() was added to Ethernet library in 2018. right now I don't know how is available() implemented and if it is possible to implement accept()

jandrassy, 2020-02-05:
https://github.com/UIPEthernet/UIPEthernet/pull/86

jandrassy, 2020-02-06:
the pull request was merged. @vk3-hca, can you try the git version?

vk3-hca, 2020-02-06:
Yes certainly, that's great.
I will probably be able do it over the weekend, I will get back in a few days.

closed 2020-02-16 by jandrassy