UIPEthernet/UIPEthernet #87: Device freezes - with many connections and disconnections

janioanselmo, 2020-02-24:
Hi,
Here at the company we have some microcontrolled electronic projects that use this library. We have a "chronic problem" (we have never been able to find a definitive solution). Through a socket TCP connection (client) we send our data to a remote server. What happens is, when we have a lot of connections and disconnections, at the next connection (number at random) the equipment hangs completely during the reconnection attempt! Sometimes with 5, 10 and even 22 disconnections. Since previous versions we have this problem. For example: 10 successful disconnections and reconnections occur. When there is a new "drop of the link" (disconnection), in the 11th attempt (as I said, this number of times is random) it BRAKES general. Our current "solution" is to redo the entire ethernet part:

1. Ethernet.begin (mac) - request the IP (DHCP);
2. "startConnection" function - start the TCP socket connection;
3. Reconnection successfully established!

However, this time is time consuming! And it has hindered the use of the equipment. Because it is a "vending machine" - with coins. Because, sometimes the user can put a coin and the equipment is in the process of "redoing the entire connection" and losing the inserted credit.

Packages:
Arduino IDE (latest version): 1.8.12
UIPEthernet Library (latest version): 2.0.7

jandrassy, 2020-02-24:
try 2.0.8

janioanselmo, 2020-02-24:
Wow!! Excellent, Mr. @jandrassy I will try and return to inform you of the news!

closed 2020-08-18 by jandrassy