UIPEthernet/UIPEthernet #88: TCP Client Sample - Time Interval - Send Info in Order - Without Delay -

mahirgul, 2020-03-04:
This example; It connects to the other side with the TCP client at certain time intervals and send the information in order.
#include <UIPEthernet.h>
byte mac[6] = {0x01, 0x02, 0x03, 0x04, 0x05, 0x06};
IPAddress ip(192, 168, 0, 55);
IPAddress server(192, 168, 0, 10);
EthernetClient client;
//Device reset duration
unsigned long duration = 110;

//Device resef function
void(* resetFunc) (void) = 0;//declare reset function at address 0

void setup()
{
  Serial.begin(9600);
  Ethernet.begin(mac, ip);
  Serial.println("Eth.Start.");
  delay(1000);
  Serial.println(Ethernet.localIP());
}

void loop()
{
  Serial.println(millis());
  Serial.println("Dev.Start.");
  unsigned long i = 0;
  if (client.connect(server, 5964))
  {
    Serial.println("Conn.Start.");
    if (client.connected())
    {
      Serial.println("Conn.OK.");
      Serial.println("Data.Send.");
      while (i < duration)
      {
        unsigned long previousMillis = millis();
        unsigned long interval = 1000;
        while (true)
        {
          client.flush();
          unsigned long currentMillis = millis();
          if (currentMillis - previousMillis > interval)
          {
            previousMillis = currentMillis;
            break;
          }
        }
        sendData(i);
        i++;
      }
      i = 0;
    }
  }
  else
  {
    Serial.println("Conn.Problem.");
  }
  Serial.println("Dev.Reset.");
  Serial.println(millis());
  resetFunc();  
}

void sendData(int i)
{
  //Serial.println(i);
  client.flush();
  if (client.connected())
  {
    //if (i == 100)         client.connect(server, 5964);
    if (i == 2)    client.println(F("SV8100"));
    else if (i == 3)    client.println(F("Data Number 1"));
    else if (i == 4)    client.println(F("Data Number 2"));
    else if (i == 5)    client.println(F("Data Number 3"));
    else if (i == 6)    client.println(F("Data Number 4"));
    else if (i == 7)    client.println(F("Data Number 5"));
    else if (i == 8)    client.println(F("Data - Data"));
    else if (i == 9)    client.println(F("Bla Bla Bla"));
    else if (i == 10)   client.println(F("Lorem ipsum dolor sit amet"));
    else if (i > 12)
    {
      client.stop();
      Serial.println("Conn.Stop.");
    }
  }
}

jandrassy, 2020-03-04:
what do you thin client.flush() does?

closed 2020-03-04 by jandrassy

mahirgul, 2020-03-04:
> what do you think client.flush() does?

If I don't use client.flush(), I cannot ping the device during the wait.

jandrassy, 2020-03-04:
it is not an issue. I was lazy to create an [not an issue] label.
you have client.flush() twice in the sketch.

mahirgul, 2020-03-04:
I also thank you for this great job.
uipethernet is very useful and simple.

jandrassy, 2020-03-04:
not verified:

```
EthernetClient client;

void loop() {
  Ethernet.maintain(); // handles the network traffic while no other library function is called
  if (millis() - previousMillis > INTERVAL) {
    previousMillis = millis();
    if (!client.connected()) { 
      if (!client.connect(server, port)) {
        Serial.println("connect failed");
      }
    }
    if (client.connected()) {
      sendData(client);
    }
  }
}
```

mahirgul, 2020-03-04:
Thank you very much, this was much better.
```
#include <UIPEthernet.h>
IPAddress ip(192, 168, 0, 55);
IPAddress server(192, 168, 0, 10);
EthernetClient client;

byte mac[6] = {0x01, 0x02, 0x03, 0x04, 0x05, 0x06};
int port = 5964;
unsigned long INTERVAL = 1000;
unsigned long i = 0;
unsigned long PERIOD = 20;
unsigned long previousMillis;

void(* resetFunc) (void) = 0;//declare reset function at address 0

void setup()
{
  Serial.begin(9600);
  Ethernet.begin(mac, ip);
  Serial.println("Eth.Start.");
  Serial.println(Ethernet.localIP());
  previousMillis = millis();
}

void loop()
{
  Ethernet.maintain();
  if (client.connected())
  {
    if (millis() - previousMillis > INTERVAL)
    {
      sendData(i);
      previousMillis = millis();
      i++;
    }
  }
  else
  {
    i = 0;
    client.connect(server, port);
  }
  if (i > PERIOD)
  {
    resetFunc();
  }
}

void sendData(int i)
{
if (client.connected())
  {
    if (i == 2)    client.println(F("SV8100"));
    else if (i == 3)    client.println(F("Data Number 1"));
    else if (i == 4)    client.println(F("Data Number 2"));
    else if (i == 5)    client.println(F("Data Number 3"));
    else if (i == 6)    client.println(F("Data Number 4"));
    else if (i == 7)    client.println(F("Data Number 5"));
    else if (i == 8)    client.println(F("Data - Data"));
    else if (i == 9)    client.println(F("Bla Bla Bla"));
    else if (i == 10)   client.println(F("Lorem ipsum dolor sit amet"));
    else if (i > 12)
    {
      client.stop();
      Serial.println("Conn.Stop.");
    }
}
```