UIPEthernet/UIPEthernet #89: Unable to use UIPEthernet on STM32F303 because of missed "HardwareSPI.h" header file

N0rbert, 2020-03-15:
Steps to reproduce:
1. Have STM32F303 Nucleo-32 board
1. Have Arduino 1.8.12 installed with latest STM32Duino 1.8.0
1. Install UIPEthernet 2.0.8 to Arduino
1. Select Nucleo F303K8 board
1. Open UIPEtehernet → TCP Client example and compile it

Expected result:
* the TCP Client example compiled succesfully

Actual results:
* the TCP Client example is not compiled because of the error:

```
ResolveLibrary(HardwareSPI.h)
  -> candidates: []
In file included from /home/norbert/Arduino/libraries/UIPEthernet/UIPEthernet.h:35,
                 from /home/norbert/Arduino/libraries/UIPEthernet/examples/TcpClient/TcpClient.ino:27:
/home/norbert/Arduino/libraries/UIPEthernet/utility/Enc28J60Network.h:209:16: fatal error: HardwareSPI.h: No such file or directory
  209 |       #include "HardwareSPI.h"
      |                ^~~~~~~~~~~~~~~
compilation terminated.
Using library UIPEthernet at version 2.0.8 in folder: /home/norbert/Arduino/libraries/UIPEthernet 
exit status 1
Error compiling for board Nucleo-32.
```

jandrassy, 2020-03-16:
in Enc28J60Network.h replace on line 208
```
   #if defined(ARDUINO) && defined(STM32F3)
      #include "HardwareSPI.h"
   #else
      #include <SPI.h>
   #endif
```
with
```
   #include <SPI.h>
```

N0rbert, 2020-03-16:
> in Enc28J60Network.h replace on line 208
> 
> ```
>    #if defined(ARDUINO) && defined(STM32F3)
>       #include "HardwareSPI.h"
>    #else
>       #include <SPI.h>
>    #endif
> ```
> 
> with
> 
> ```
>    #include <SPI.h>
> ```

This is correct, but not enough. After this change I have the following:

```
/home/norbert/Arduino/libraries/UIPEthernet/utility/Enc28J60Network.cpp:49:17: fatal error: HardwareSPI.h: No such file or directoryAlternatives for HardwareSPI.h: []

ResolveLibrary(HardwareSPI.h)
   49 |        #include "HardwareSPI.h"  -> candidates: []

      |                 ^~~~~~~~~~~~~~~
compilation terminated.
```

So we need somehow official solution. 
I fear to break something by manual patching of UIPEthernet project code with which I'm not familiar.

jandrassy, 2020-03-16:
I am the current maintainer of this library.
test this changes https://github.com/UIPEthernet/UIPEthernet/pull/90/commits/c6d6519a260166b722b9cee5dd1f0fb2682e6782
it compiles, but I don't have a STM32 F3 or F4 board

N0rbert, 2020-03-16:
Thanks, this fixed the issue.

jandrassy, 2020-03-16:
> Thanks, this fixed the issue.

and does it work? can the library communicate with the chip?

N0rbert, 2020-03-16:
Yes, of course. I have compiled the TCPClient example and my own program with UIPEthernet and HTTPClient - both can communicate with the chip, get IP and DNS. 
For my program I needed to change just one other line from [`UIPEthernet/UIPClient.h:26`](https://github.com/UIPEthernet/UIPEthernet/blob/master/UIPClient.h#L26):

```
  #if defined(__STM32F3__) || defined(STM32F3) || defined(__RFduino__)
    #include "mbed/Client.h"

```

to 

```
  #if defined(__RFduino__)
    #include "mbed/Client.h"
```

Without this change it stalled after `http.responseStatusCode()` where `http` is a `HttpClient` object over `EthernetClient`.

I suppose that other problems may exist with STM32F3 `define`s. I'll test further and report them as separate bugs if any.

jandrassy, 2020-03-17:
please don't create additional issues. add the problems here.

jandrassy, 2020-03-17:
I found 'the pattern' an reworked the PR

N0rbert, 2020-03-17:
I have tested the whole PR, it works now as expected.
But for correct logging on this platform the `Serial` should be used instead of `Serial1` in [STM32F303K8 variant](https://github.com/stm32duino/Arduino_Core_STM32/blob/master/variants/NUCLEO_F303K8/variant.h#L108). So I changed 

https://github.com/UIPEthernet/UIPEthernet/blob/master/utility/logging.h#L32

```
     #if defined(__STM32F1__) || defined(__STM32F3__) || defined(STM32F3) || defined(__STM32F4__) || defined(ARDUINO_ARCH_SAM)
```

to

```
     #if defined(__STM32F1__) || defined(__STM32F4__) || defined(ARDUINO_ARCH_SAM)
```

on my local installation.

jandrassy, 2020-03-17:
the `__STM32Fx__` defines are from Roger Clark's STM32 cores. the `STM32F3` was used for some Adafruit STM32 board. 
this library was not tested with the new STM32 core by STM, but the core is so standard that everything works, only it has STM32F3 defined too a that caused to use the special code for the Adafruit board.
I missed the occurrence of STM32F3 in logging.h. I will later today make the right modification there too.

jandrassy, 2020-03-17:
> I missed the occurrence of STM32F3 in logging.h. I will later today make the right modification there too.

done

N0rbert, 2020-03-17:
Small note: you have had `#define ACTLOGLEVEL     LOG_NONE` before, now you have `#define ACTLOGLEVEL LOG_WARNING` by default.
Logging works, thank you!

jandrassy, 2020-03-17:
> Small note: you have had `#define ACTLOGLEVEL LOG_NONE` before, now you have `#define ACTLOGLEVEL LOG_WARNING` by default.
> Logging works, thank you!

thank you

closed 2020-03-17 by jandrassy