UIPEthernet/UIPEthernet #92: DHCP address assignment not working

t-dan, 2020-04-10:
Hello,
I am trying the DhcpAddressPrinter from the default Ethernet package on Arduino Mega, only with change of include to `#include <UIPEthernet.h>` (since I have Ethernet Shield ENC28J60).

However, the DHCP address is never assigned. I have set the debug output and later is what  I got. I have tried it with the latest GIT sources. 

Where can be a problem? Any other computer on the net gets its address assigned.

Thank you very much,
DT

Here is the log output:

---
Initialize Ethernet with DHCP:

ENC28J60::init DEBUG:csPin = 53
Initialize Ethernet with DHCP:

ENC28J60::init DEBUG:csPin = 53

ENC28J60::init DEBUG:miso = 50

ENC28J60::init DEBUG:mosi = 51

ENC28J60::init DEBUG:sck = 52

ENC28J60::init DEBUG:Use SPI lib SPI.begin()

ENC28J60::init INFO: Chip erevid=6 initialization completed.

DhcpClass::beginWithDHCP(uint8_t *mac) DEBUG_V1:Function started

DhcpClass::reset_DHCP_lease(void) DEBUG_V1:Function started

DhcpClass::request_DHCP_lease(void) DEBUG_V1:Function started

DhcpClass::presend_DHCP(void) DEBUG_V1:Function started (Empty function)

DhcpClass::request_DHCP_lease(void) DEBUG_V1:dhcp_state=STATE_DHCP_START -> send_DHCP_MESSAGE DHCP_DISCOVER

DhcpClass::send_DHCP_MESSAGE(uint8_t messageType, uint16_t secondsElapsed) DEBUG_V1:Function started

Enc28J60Network::receivePacket(void) DEBUG:receivePacket [6-42], next: 46, stat: C0, Packet count: 1 -> OK

Enc28J60Network::receivePacket(void) DEBUG: rxstat OK. receivePkt.size=60

UIPEthernetClass::tick() DEBUG:receivePacket: 255

Enc28J60Network::readPacket(memhandle handle, memaddress position, uint8_t* buffer, uint16_t len) DEBUG_V2: Read bytes:60 save to block(255) [0]: FF FF FF FF FF FF 28 39 26 8 99 EF 8 6 0 1 8 0 6 4 0 1 28 39 26 8 99 EF C0 A8 C8 94 0 0 0 0 0 0 C0 A8 C8 19 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 

UIPEthernetClass::tick() DEBUG:readPacket type ARP, uip_len: 60

UIPEthernetClass::tick() DEBUG:freeing packet: 255

Enc28J60Network::setERXRDPT(void) DEBUG:Set actnextPacketPtr:45

UIPUDP::beginPacket DEBUG:udp beginPacket, rip: 255.255.255.255, port: 67

Enc28J60Network::writePacket(memhandle handle, memaddress position, uint8_t* buffer, uint16_t len) DEBUG_V2: Write bytes:28 save to block(1) [82B]: 1 1 6 0 0 0 3 31 0 0 80 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 

Enc28J60Network::writePacket(memhandle handle, memaddress position, uint8_t* buffer, uint16_t len) DEBUG_V2: Write bytes:16 save to block(1) [847]: 77 99 22 55 66 66 0 0 0 0 0 0 0 0 0 0 

Enc28J60Network::writePacket(memhandle handle, memaddress position, uint8_t* buffer, uint16_t len) DEBUG_V2: Write bytes:32 save to block(1) [857]: 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 

Enc28J60Network::writePacket(memhandle handle, memaddress position, uint8_t* buffer, uint16_t len) DEBUG_V2: Write bytes:32 save to block(1) [877]: 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 

Enc28J60Network::writePacket(memhandle handle, memaddress position, uint8_t* buffer, uint16_t len) DEBUG_V2: Write bytes:32 save to block(1) [897]: 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 

Enc28J60Network::writePacket(memhandle handle, memaddress position, uint8_t* buffer, uint16_t len) DEBUG_V2: Write bytes:32 save to block(1) [8B7]: 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 

[... 887 lines cut ...]

Enc28J60Network::writePacket(memhandle handle, memaddress position, uint8_t* buffer, uint16_t len) DEBUG_V2: Write bytes:9 save to block(1) [935]: 37 6 1 3 6 F 3A 3B FF 

uipudp_appcall(void) DEBUG:udp, uip_poll preparing packet to send: 1, size: 317

UIPEthernetClass::upper_layer_chksum(uint8_t proto) DEBUG:uip_buf[34-42]: CE02

UIPEthernetClass::upper_layer_chksum(uint8_t proto) DEBUG:uip_packet(1)[42-317]: DA67

UIPUDP::_send() DEBUG:udp, uip_packet to send: 1

UIPEthernetClass::network_send() DEBUG:uip_packet: 1, hdrlen: 42

Enc28J60Network::writePacket(memhandle handle, memaddress position, uint8_t* buffer, uint16_t len) DEBUG_V2: Write bytes:42 save to block(1) [801]: FF FF FF FF FF FF 77 99 22 55 66 66 8 0 45 0 1 2F 0 7 0 0 40 11 79 B8 0 0 0 0 FF FF FF FF 0 44 0 43 1 1B 98 25 

Enc28J60Network::sendPacket(memhandle handle) DEBUG:sendPacket(1) [800-93D]: 0 FF FF FF FF FF FF 77 99 22 55 66 66 8 0 45 0 1 2F 0 7 0 0 40 11 79 B8 0 0 0 0 FF FF FF FF 0 44 0 43 1 1B 98 25 1 1 6 0 0 0 3 37 0 38 80 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 77 99 22 55 66 66 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 63 82 53 63 35 1 1 3D 7 1 77 99 22 55 66 66 C C 45 4E 43 32 38 4A 35 35 36 36 36 36 37 6 1 3 6 F 3A 3B FF 

MemoryPool::freeBlock(memhandle handle) WARNING: Don't free NOBLOCK handle

MemoryPool::freeBlock(memhandle handle) WARNING: Don't free NOBLOCK handle

MemoryPool::freeBlock(memhandle handle) WARNING: Don't free NOBLOCK handle

Failed to configure Ethernet using DHCP
`

jandrassy, 2020-04-11:
and has your network DHCP? is the module connected to router or switch?

t-dan, 2020-04-11:
Hello,
thank you for reply.
Sure there are (both). Actually to check it, I am sending this mail using the same cable I use to connect the Arduino.

From the log it also seems that there are data coming, but the DHCP response is probably not recognized somehow?.

Best regards,
DT

jandrassy, 2020-04-11:
what module do you use and how do you power it?

t-dan, 2020-04-11:
Hello,
it is Arduino Mega2560-16AU, CH340G with Ethernet Shield ENC28J60.

I have to "re-wire" it according to https://www.scribd.com/document/343865837/how-to-use-enc28j60-ethernet-shield-with-arduino-mega-2560:

- SO to Pin50
- SI to Pin51
- SCK to Pin52
- CS to Pin53

without it, instead of `ENC28J60::init INFO: Chip erevid=6 initialization completed`, there was an error message (now it works, I suppose).

The power goes through USB cable from my laptop.

Thank you,
DT

jandrassy, 2020-04-11:
exactly what shield? do you have a link to a product page?

t-dan, 2020-04-11:
Hmmm, sorry the e-shop where I bought it does not have a full specification. There is just a photo: https://arduino-shop.cz/arduino/1590-ethernet-shield-enc28j60-s-univerzalni-casti-dps.html

There is also HW-270 on it, so it might be this one: https://www.alibaba.com/product-detail/HW-270-ENC28J60-network-module-IoT_62331799311.html?spm=a2700.pc_countrysearch.main07.22.38a64057rUUHkG, or this

Is it heplfull?
DT

jandrassy, 2020-04-11:
did you wire 5 V to 5V for power?

on the photos it looks like the ICSP/SPI header is wired so I would stack the shield on Mega and use Ethernet.init(10); to set the CS pin.

t-dan, 2020-04-11:
OK. Do you mean without the "special" wining?

jandrassy, 2020-04-11:
> OK. Do you mean without the "special" wining?

sorry I deleted that, because I see only one trace to the ICSP header, but there are traces to pins 11 to 13, so the shield creator didn't know that SPI shields should use the ICSP header.

how did you wire powering? it should be 5 V to 5 V

t-dan, 2020-04-11:
> how did you wire powering? it should be 5 V to 5 V

Yes. Except the 4 wires to pins 50-53, the shield sits on the Mega board. The connections should match.

Dan

[image]

t-dan, 2020-04-11:
Just a side-note.

This shield (exactly this piece) did work with different Arduino clone (namely UNO R3, Atmega328P, CH340) with static IP set. Unfortunately, I cannot use the original UNO clone, since I have not it available any more.
With the Mega, the connection does not work even with static IP assigned (tested on older project with MQTT PubSubClilent module - it cannot connect the MQTT broker). 

DT

jandrassy, 2020-04-11:
did you try with fixed IP address?

you can use pin 10 as CS with `Ethernet.init(10);`
once the pins are bended, you can put the jumpers between the go-thru ICSP header of the shield and pins 11 to 13 of the shield

http://www.gammon.com.au/images/ArduinoUno_R3_Pinouts.png

t-dan, 2020-04-11:
OK, i will try it, but I'am not sure I copy.

So I put CS of shield to PIN10 of MEGA and set `Ethernet.init(10);`

But I am not sure I understand the rest. Do you mean to connect _ICSP pins MISO, SCK and MOSI_ of MEGA (or the shield, if they just go through when put on) to the corresponding pins MOSI, MISO and SCK of the shield instead of using pins 50,51 and 52 on MEGA? 

Thank you very much,
DT

jandrassy, 2020-04-11:
> So I put CS of shield to PIN10 of MEGA and set `Ethernet.init(10);`
>
> But I am not sure I understand the rest. Do you mean to connect _ICSP pins MISO, SCK and MOSI_ of MEGA (or the shield, if they [...]

yes. it will of course not solve the DHCP problem, but you will have a 'compact' shield. :-)
MISO and 50, MOSI and 51, SCK and 52 go to the same pin of the ATmega2560.

t-dan, 2020-04-11:
OK, works, but as you said, did not solve the DHCP.
The log looks a bit differently, but the overall behaviour is the same.

It seems to send `DHCP_DISCOVER` message:
 `DhcpClass::request_DHCP_lease(void) DEBUG_V1:dhcp_state=STATE_DHCP_START -> send_DHCP_MESSAGE DHCP_DISCOVER`
but the `messageType` in `request_DHCP_lease()` is 255 (I thing it is returned by `parseDHCPResponse()`) , resetting the attempt again ...

So, what is the next step?

Thank you,
DT

jandrassy, 2020-04-11:
did you try with static IP?

t-dan, 2020-04-11:
O have tried a slightly modified TcpServer example from this project (adjusting address to my network configuration and call `Ethernet.init(10);  Ethernet.begin(mac,myIP);`.

The static IP was used. However, I was unable to neither ping the Arduino, not to telnet to it (telnet IP 1000), as said in the example header.

Using `Ethernet.begin(mac,myIP,myDNS,myGW,myMASK);` did not improved the situation.

My other trials did not work as well.

DT

t-dan, 2020-04-11:
Neither WebServer example from the Ethernet library (modified to use UIPEthernet, of course) does not work.
DT

jandrassy, 2020-04-12:
on the pictures it looks like the output of the 5 V to 3.3 V converter on shield has a trace back to 3.3 V pin of Mega. I don't like it.

could you try to wire the shield without stacking it on the Mega. ground, 5 V, reset, the SPI pins and the CS pin. on the Mega you can use the ICSP header of all but pin 10.

t-dan, 2020-04-12:
Hello,
I did as you have said, but no change, unfortunately. I have also checked the description, and 5V pin on MEGA matches the 5V pin on the shiled (at least as printed on them).

There is some communication going, but the WebServer (the latest example I have described) cannot connected.

---
**Connection:**

Ethernet WebServer Example

ENC28J60::init DEBUG:csPin = 10
�Ethernet WebServer Example

ENC28J60::init DEBUG:csPin = 10

ENC28J60::init DEBUG:miso = 50
Connection:
ENC28J60::init DEBUG:mosi = 51

ENC28J60::init DEBUG:sck = 52

ENC28J60::init DEBUG:Use SPI lib SPI.begin()

ENC28J60::init INFO: Chip erevid=6 initialization completed.

server is at 192.168.200.240

Enc28J60Network::receivePacket(void) DEBUG:receivePacket [6-42], next: 46, stat: C0, Packet count: 1 -> OK

Enc28J60Network::receivePacket(void) DEBUG: rxstat OK. receivePkt.size=60

UIPEthernetClass::tick() DEBUG:receivePacket: 255

Enc28J60Network::readPacket(memhandle handle, memaddress position, uint8_t* buffer, uint16_t len) DEBUG_V2: Read bytes:60 save to block(255) [0]: FF FF FF FF FF FF 28 39 26 8 99 EF 8 6 0 1 8 0 6 4 0 1 28 39 26 8 99 EF C0 A8 C8 94 0 0 0 0 0 0 C0 A8 C8 19 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 

----
**The sample of packets:**

UIPEthernetClass::tick() DEBUG:readPacket type IP, uip_len: 75

UIPEthernetClass::upper_layer_chksum(uint8_t proto) DEBUG:uip_buf[34-42]: C4BA

UIPEthernetClass::upper_layer_chksum(uint8_t proto) DEBUG:uip_packet(255)[42-75]: FFFF

UIPEthernetClass::tick() DEBUG:freeing packet: 255

Enc28J60Network::setERXRDPT(void) DEBUG:Set actnextPacketPtr:679

Enc28J60Network::receivePacket(void) DEBUG:receivePacket [680-6BC], next: 6C0, stat: C0, Packet count: 21 -> OK

Enc28J60Network::receivePacket(void) DEBUG: rxstat OK. receivePkt.size=60

UIPEthernetClass::tick() DEBUG:receivePacket: 255

Enc28J60Network::readPacket(memhandle handle, memaddress position, uint8_t* buffer, uint16_t len) DEBUG_V2: Read bytes:60 save to block(255) [0]: FF FF FF FF FF FF 28 39 26 8 99 EF 8 6 0 1 8 0 6 4 0 1 28 39 26 8 99 EF C0 A8 C8 94 0 0 0 0 0 0 C0 A8 C8 19 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 

UIPEthernetClass::tick() DEBUG:readPacket type ARP, uip_len: 60

UIPEthernetClass::tick() DEBUG:freeing packet: 255

Enc28J60Network::setERXRDPT(void) DEBUG:Set actnextPacketPtr:6BF

Enc28J60Network::receivePacket(void) DEBUG:receivePacket [6C6-702], next: 706, stat: C0, Packet count: 21 -> OK

Enc28J60Network::receivePacket(void) DEBUG: rxstat OK. receivePkt.size=60

UIPEthernetClass::tick() DEBUG:receivePacket: 255

Enc28J60Network::readPacket(memhandle handle, memaddress position, uint8_t* buffer, uint16_t len) DEBUG_V2: Read bytes:60 save to block(255) [0]: FF FF FF FF FF FF 28 39 26 8 99 EF 8 6 0 1 8 0 6 4 0 1 28 39 26 8 99 EF C0 A8 C8 94 0 0 0 0 0 0 C0 A8 C8 19 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0

jandrassy, 2020-04-12:
did you try a different Ethernet cable?

t-dan, 2020-04-12:
No I didn't, but I had my laptop on exactly that cable yesterday, and it worked without issues. Do you think, that the data read by the shield (ant printed in debug mode) are incorrect?

Thank you,
DT

t-dan, 2020-04-12:
OK, I have double checked it and printer on the same cable works....
Do you think that the data printed by readPacket() are nonsense values?

DT

jandrassy, 2020-04-12:
I don't know if the hex codes are meaningful. I know that the library works so the problem must be some instability with the shield or the wiring.

closed 2020-04-12 by jandrassy

t-dan, 2020-04-12:
I would say so ...

t-dan, 2020-04-13:
OK, I have used tcpdump packets sniffer and here are my final observation:
- connecting the cable *directly* to my laptop - **I see** DHCP requests coming from the shield
- connecting the same cable to my home network - **I don't see** DHCP requests neither in the router, nor in my laptop (I do see requests from other computers/phones, though)
- connecting other device through the same cable - I see DHCP requests from that device.

I am absolutely confused since it does not make sense to me ...
DT

t-dan, 2020-04-13:
The last status update:

the problem has most likely be the MAC address (although I have used an automatically generated one by a web MAC address generator).

When I used MAC from a different Arduino sample, **all started to work magically**.

So sorry for your time and thank you very much for all the help and advice. I really appreciate it!

Best regards,
DT