UIPEthernet/UIPEthernet #94: ESP8266 TCP client not correctly works.

cyberpablis, 2020-04-21:
I'm trying to send and receive binary data packets using TCP client but is not efficiently working. A lot of packets lost and sometimes, the data sent not really sent. The size of the each data packet is 500 bytes.
My hardware is made with ESP8266 and ENC28J60 shield.

jandrassy, 2020-04-22:
is the WiFi off? they both need 150 mA while transmiting

cyberpablis, 2020-04-22:
WiFI is off
Transmission is through ENC28J60 using EthernetClient object

cyberpablis, 2020-04-24:
This is my code:

#include <ESP8266WiFi.h>
#include <UIPEthernet.h>

EthernetClient tETHClient;
uint8_t        vMAC[6] = { 0x40, 0x23, 0x33, 0x11, 0x10, 0x02 };
uint8_t        Buf[500];

void setup()
  {
  WiFi.mode(WIFI_OFF); 
  delay(1);       
  // Wait for DHCP request.
  while (Ethernet.begin(vMAC) == 0);
  }

void loop()
  {
  // Stay connected to remote server socket.
  if (!tETHClient.connected())
    if (tETHClient.connect("190.61.77.251", 3005);

  // When a packet must be sended ... 
  tETHClient.write(Buf, sizeof(Buf));
  tETHClient.flush();

  // To Check if a packet incoming...
  while (tETHClient.available() > 0) 
    if (tETHClient.read(Paq.tfBuf, sizeof(Paq.tfBuf)) == sizeof(Paq.tfBuf))
      {
      // Process packet...
     
       
      }
  }

cyberpablis, 2020-04-24:
#include <ESP8266WiFi.h>
#include <UIPEthernet.h>

EthernetClient tETHClient;
uint8_t        vMAC[6] = { 0x40, 0x23, 0x33, 0x11, 0x10, 0x02 };
uint8_t        Buf[500];

void setup()
  {
  WiFi.mode(WIFI_OFF); 
  delay(1);       
  // Wait for DHCP request.
  while (Ethernet.begin(vMAC) == 0);
  }

void loop()
  {
  // Stay connected to remote server socket.
  bool b = tETHClient.connected();
  if (!b) b = tETHClient.connect("190.61.77.251", 3005);
  
  if (b)
    {
    // When a packet must be sended ... 
    tETHClient.write(Buf, sizeof(Buf));
    tETHClient.flush();

    // To Check if a packet incoming...
    while (tETHClient.available() > 0) 
      if (tETHClient.read(Paq.tfBuf, sizeof(Paq.tfBuf)) == sizeof(Paq.tfBuf))
        {
        // Process packet...
     
       
        }
    }
  }

jandrassy, 2020-04-25:
I think you put more load on the library then it can handle. The library is written for small MCUs and uses only the 2kB RAM of the enc28j60 to store incoming packets. This takes the space normally to send packets.
you should try this https://github.com/esp8266/Arduino/pull/6680

cyberpablis, 2020-04-25:
Tks for your answer. 
I will try to implement the enc28j60 interrupt to intercept incoming packets

cyberpablis, 2020-04-25:
I resized the buffer to 50 bytes and nothing.
The library has a problem and do not properly works

jandrassy, 2020-04-25:
> I resized the buffer to 50 bytes and nothing.
> The library has a problem and do not properly works

500 for receive is good. but try to send only 250 bytes at once.

yes it has a problem, the concept of the library doesn't allow receive and send so much so fast. it was not made for fast MCU with so much memory.

jandrassy, 2020-08-18:
I fixed a bug. could you please try the Git version to see if it solved your problem?

cyberpablis, 2020-08-18:
Excellent! It works!!!! Thanks a lot

closed 2020-08-19 by jandrassy

cyberpablis, 2020-08-19:
Hi.

I use a UIPEthernet TCP client on a ESP8266 to connect a server socket in a PC, and send periodic packets.
If from Windows, I deactivate the network card on which it receives the packets and activate it again, the device freezes. I thought it would stay blocked until after the TCP connection timeout, but not, it freezes forever.

What I can be doing wrong?

Best regards

jandrassy, 2020-08-20:
> Hi. I use a UIPEthernet TCP client on a ESP8266 to connect a server socket in a PC, and send periodic packets. If from Windows, I deactivate the network card on which it receives the packets and [...]

did you try to set UIP_CONNECT_TIMEOUT in uipethernet-conf.h?

cyberpablis, 2020-08-20:
I've tried, but the ESP8266 stays frozen forever, like it's in an infinite loop. does not respond, does not refresh the display, nothing.