UIPEthernet/UIPEthernet #95: Can't change CS pin on ESP32

ringmybell, 2020-04-25:
Hi, thanks for the library. 

Having a problem where I can't change the CS pin on an ESP32 using ENC28J60 from the standard pin 5 to another IO pin ( i have picked 4 to test it out). 

I have tried the following snippets of code in various combinations, in all cases I can only get the sketch working if I keep all the SPI pins standard. 

> In main sketch using Ethernet.init(4) in void(setup). 
> In main sketch, making sure pin 5 is set to output to ensure SPI master mode is entered at start up pinMode(5, OUTPUT) in void(setup). 
> In Enc28J60Network.h I added #define ENC28J60_CONTROL_CS  4
> In Enc28J60Network.h I added #define SPI_SS 5

When operating in standard pin 5 configuration the webpage reloads every 5 seconds with ADC data, when I try to alter the CS with above combinations it either doesn't detect the IP address, connects then disconnects immediately or connects then disconnects after 10 seconds, but the webpage never loads. Often the server does not respond if I leave the webpage open then refresh, but if i close the webpage, open a new tab and reenter the IP address it will react. 

I have not found a really clear answer on what code needs to be altered and where. Can someone just spell it out clearly?

jandrassy, 2020-04-26:
`Ethernet.init(4)` should be enough

ringmybell, 2020-04-27:
It for sure does not work with my setup. Strange.  

See attached screen grab of my serial monitor response. With the working version it refreshes every 5 seconds. The non working version attempts one connection then stops. The line "Cache-Control: max-age=0" seems to be missing from the one that is not working. 

Using ESP32, current versions of ESP 1.04 and UIPEthernet library. 

Code is from standard Arduino examples Examples>Ethernet>Webserver, all I changed was commenting out SPI.h and Ethernet.h library and adding UIPEthernet.h and adding the line Ethernet.init(4)
 
[image]

jandrassy, 2020-04-27:
is the enc28j60 the only one SPI device on SPI bus?

ringmybell, 2020-04-27:
Nothing else is on that SPI bus, just the Ethernet chip.

jandrassy, 2020-10-03:
so I tested it finally. this happens to me again and again with WebServer example on different architectures.

the example reads first 7 analog pins. pin 4 is one of them on esp32. so analogRead reconfigures the pin.

closed 2020-10-03 by jandrassy