UIPEthernet/UIPEthernet #97: UIPEthernet 2.0.8 and STM32duino (Roger Core)

luca-stm32, 2020-05-19:
Hi Jandrassy.
I used your 2.0.4 version library for a project with STM32 micro and Roger core.
Today I upgraded to 2.0.8 version and my project doesn't work anymore.
I found the issue in ENC28j60Network.cpp file: I replaced line 26:
`uint8_t ENC28J60ControlCS = ENC28J60_CONTROL_CS;`
with:
`#define ENC28J60ControlCS ENC28J60_CONTROL_CS`

With this patch, 2.0.8 library works for me.
I use Eclipse (Sloeber) and VSCode, not Arduino IDE.

Thanks for your great work.
Hope this could help.

Regrads.
Luca

jandrassy, 2020-05-20:
more details please. for me it compiles without problems in Sloeber for BluePill.

luca-stm32, 2020-05-20:
Goodmorning Jandrassy.
Yes, it compiles without errors, but sketch doesn't work. I connected a scope to SPI CS PIN (PA4) and I noticed that it never goes down.
I don't have the board with me now, but, if I remember well, it seem like that 
'uint8_t ENC28J60ControlCS = ENC28J60_CONTROL_CS;'
assign to ENC28J60ControlCS the value 0 (ENC28J60_CONTROL_CS in Bluepill is PA4).

luca-stm32, 2020-05-28:
Hello Jandrassy.
I made some more test.
In Enc28J60Network.cpp I changed 
`#define CSACTIVE digitalWrite(ENC28J60ControlCS, LOW)` in
`#define CSACTIVE digitalWrite(ENC28J60ControlCS, LOW); while(1) Serial.println (ENC28J60ControlCS)` in order to see the number of pin ENC28J60ControlCS.
With the modification `#define ENC28J60ControlCS ENC28J60_CONTROL_CS
` in serial terminal I see 4.
With original `uint8_t ENC28J60ControlCS = ENC28J60_CONTROL_CS;` in serial terminal I see 0.

jandrassy, 2020-06-28:
so I analysed it. yes it is an error and only for Roger's core. the variable `ENC28J60ControlCS` was intruduced to support `Ethernet.init(cspin)` so it must be a variable. Only for Roger's core  the ENC28J60_CONTROL_CS macro contains a function call. At the time of `uint8_t ENC28J60ControlCS = ENC28J60_CONTROL_CS;` the function doesn't know the pin and returns 0. 

simply use `Ethernet.init(PA4);` in your sketch

closed 2020-06-28 by jandrassy