UIPEthernet/UIPEthernet #98: Check if Arduino web server is still listening...

matusiam34, 2020-05-22:
Hello,

I did some investigation and can't figure out how to check from the Arduino Nano perspective if the web server is still running or the status of the 4 sockets that are available. This way if I can detect that for example the sockets are locked up I can trigger a Nano reboot. There is apparently a way with the W5100 chipset to see the status of the sockets and even do if (!server) to check if the web server is running.

Excuse my ignorance.

Thank you in advance

Regards
Matt

jandrassy, 2020-05-23:
sorry, there is no way to check the server's state with this library for now

matusiam34, 2020-05-24:
> sorry, there is no way to check the server's state with this library for now

Thank you for the prompt response. It seems like I will have to be creative and figure out a different way to get the same result somehow.

Regards
Matt

jandrassy, 2020-08-25:
in W5x00 the server runs in the chip. here the server is run by the uip library. it decides if an incoming packet is for a port which is in `uip_listenports` array. so if the port is in the array then the server is 'running'

closed 2020-08-25 by jandrassy

jandrassy, 2020-08-31:
server now has bool operator so you can use

`if (server) ...`