UIPEthernet/UIPEthernet #101: Wait for CLKRDY to be set

rayshobby, 2020-07-10:
The following line is supposed to wait till CLKRDY is set (to 1) but I think something is wrong:
https://github.com/UIPEthernet/UIPEthernet/blob/7c16ed4fcb4ca31049f678878514208efd28a8b2/utility/Enc28J60Network.cpp#L210

Specifically, I think the condition should be
`if( !(readOp(...) & ESTAT_CLKRDY) && (timeout>0))`

currently the code is:
`if( (!readOp(...) & ESTAT_CLKRDY) && (timeout>0))`
because ! has higher precedence than &, this will result in ! applied to readOp first, then taking bitwise and with ESTAT_CLKRDY. This seems wrong.

jandrassy, 2020-07-10:
does it cause you some problems? as the comment says, the bit doesn't work so it doesn't matter how it is tested

rayshobby, 2020-07-10:
No obvious problem yet, but I have been monitoring CLKRDY through serial output and noticed that upon the end of Enc28j60::init function, CLKRDY is still 0, which led me to find this bug. After correcting the code, CLKRDY is indeed 1 after the wait loop.

If this while loop is useless and incorrect, why not just remove it?

jandrassy, 2020-07-10:
> 
> 
> No obvious problem yet. But out of curiosity: if this while loop is useless and incorrect, why not just remove it?

no one cares

rayshobby, 2020-07-10:
all right, fair enough. guess it's ok to leave useless garbage in the code from time to time.

closed 2020-07-10 by rayshobby