UIPEthernet/UIPEthernet #102: I could it be possible to have a simple way to acces ethernet raw packet and discriminate with ethertype

fblaisarduino, 2020-07-10:


fblaisarduino, 2020-07-10:
i am trying to do a lldp sniffer with an espduino-32 and a enc28j60 and it is difficult to have someting compatible to both

jandrassy, 2020-08-28:
this library has 2 layers over raw packets. the UIP stack and the Arduino API.
the low level is based on https://people.ece.cornell.edu/land/courses/ece4760/FinalProjects/s2009/dat38/Website/code/enc28j60.c

closed 2020-08-28 by jandrassy

fblaisarduino, 2020-08-29:
do you have the same for esp32/m5stack and and exemple ?

thanks

Francois Blais
fblais001@hotmail.com

jandrassy, 2020-08-29:
> do you have the same for esp32/m5stack and and exemple ? thanks Francois Blais fblais001@hotmail.com
> […](#)

examine the esp32 Arduino WiFi library. the esp32 uses LWIP TCP stack but WiFi handling code is closed source. esp8266 Arduino core replaced some parts of the closed source with open source so maybe there it is better to see how it handles raw packets